freifunk-gluon/gluon #1358: OOM in respondd after "topology change"

jplitza, 2018-04-07:
In Bremen, we are currently experiencing weird mesh-wide reboots due to OOM. They are most often triggered by rebooting one of the VPN gateways, which causes almost all the nodes to suddenly reboot approximately 3 minutes later (not only the direct neighbors of the gateway!).

We were able to capture the logs of multiple crashes, and they always look like this (which is also what @genofire pasted in #1296, but we didn't really know about the problem back then):
```
<27>Apr  7 12:18:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator da:9a:a7:0c:fd:1a (router 52:54:00:17:cb:b7)
<27>Apr  7 12:18:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator ae:ca:0b:5e:a2:57 (router 52:54:00:0f:3c:f7)
<27>Apr  7 12:18:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator 0e:6f:be:24:15:bf (router 52:54:00:6e:dd:43)
<27>Apr  7 12:18:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator d2:e4:b1:db:0e:3b (router 2e:a0:97:b0:c4:62)
<27>Apr  7 12:19:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator da:9a:a7:0c:fd:1a (router 52:54:00:17:cb:b7)
<27>Apr  7 12:19:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator ae:ca:0b:5e:a2:57 (router 52:54:00:0f:3c:f7)
<27>Apr  7 12:19:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator 0e:6f:be:24:15:bf (router 52:54:00:6e:dd:43)
<27>Apr  7 12:19:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator d2:e4:b1:db:0e:3b (router 2e:a0:97:b0:c4:62)
<27>Apr  7 12:20:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator da:9a:a7:0c:fd:1a (router 52:54:00:17:cb:b7)
<27>Apr  7 12:20:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator ae:ca:0b:5e:a2:57 (router 52:54:00:0f:3c:f7)
<27>Apr  7 12:20:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator 0e:6f:be:24:15:bf (router 52:54:00:6e:dd:43)
<27>Apr  7 12:20:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator d2:e4:b1:db:0e:3b (router 2e:a0:97:b0:c4:62)
<27>Apr  7 12:20:14 hhma-13-1 gluon-radv-filterd[2133]: Unable to find TQ for originator d2:e4:b1:db:0e:3b (router 2e:a0:97:b0:c4:62)
<1>Apr  7 12:21:00 hhma-13-1 kernel: [56425.075256] CPU 0 Unable to handle kernel paging request at virtual address 00000040, epc == 8742a56c, ra == 874200bc
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.086245] Oops[#1]:
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.088569] CPU: 0 PID: 2308 Comm: respondd Not tainted 4.4.93 #0
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.094751] task: 87c2f250 ti: 86c62000 task.ti: 86c62000
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.100228] $ 0   : 00000000 00000001 00000008 00000003
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.105569] $ 4   : 87105f00 875dcdf0 871fac20 ffff3fff
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.110911] $ 8   : fffffffc 80065090 803d0000 00400904
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.116252] $12   : 7fc334d0 ffffff80 00000000 777dc2c0
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.121585] $16   : 86e72240 87f36c80 5ac89b8d 8737679c
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.126917] $20   : 00000000 875dcdf0 00000001 777d6d8c
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.132251] $24   : 00000000 77741000                  
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.137592] $28   : 86c62000 86c63b48 871fac20 874200bc
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.142926] Hi    : 0000001f
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.145847] Lo    : 04924924
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.148848] epc   : 8742a56c batadv_orig_node_get_router+0x14/0x4c [batman_adv]
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.156276] ra    : 874200bc 0x874200bc
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.160164] Status: 1100dc03    KERNEL EXL IE 
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.164443] Cause : 00800008 (ExcCode 02)
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.168508] BadVA : 00000040
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.171431] PrId  : 0001974c (MIPS 74Kc)
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.175405] Modules linked in: ath9k ath9k_common iptable_nat ath9k_hw ath nf_nat_ipv4 nf_conntrack_ipv6 nf_conntrack_ipv4 mac80211 ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_filter ebtable_broute cfg80211 xt_time xt_tcpudp xt_state xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_LOG xt_CT nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_nat nf_log_ipv4 nf_defr
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.287410] Process respondd (pid: 2308, threadinfo=86c62000, task=87c2f250, tls=777ddd48)
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.295788] Stack : 00000000 804304d0 00000000 00000000 00000000 00000000 86e72240 020012c0
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.295788]       87436180 00400904 00000000 87437d3c 00000014 777d6d8c 00000000 800eb6e0
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.295788]       00000141 86c63c74 86e72240 87376780 871fa800 8737679c 875dcdf0 00000000
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.295788]       00000014 777d6d8c 00000000 87426fbc 020012c0 0000401f 87c02780 00000000
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.295788]       875dcdf0 86e72240 87435ca8 875dcdf0 00004000 8029da38 00000000 80430000
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.295788]       ...
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.332129] Call Trace:
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.334631] [<8742a56c>] batadv_orig_node_get_router+0x14/0x4c [batman_adv]
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.341732] [<874200bc>] 0x874200bc
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.345285] 
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.346794] 
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.346794] Code: 00000000  03e00008  00001021 <8c430038> 1060fffc  24650001  c0440038  14830005  00000000 
<4>Apr  7 12:21:00 hhma-13-1 kernel: [56425.357100] ---[ end trace fd3a62d9246c7803 ]---
```

The last dump of `top -b -d1` from right before the crash looks unsuspicious:
```
Mem: 30292K used, 94964K free, 104K shrd, 2240K buff, 7724K cached
CPU:   1% usr   3% sys   0% nic  88% idle   0% io   0% irq   5% sirq
Load average: 0.27 0.14 0.10 4/53 13646
  PID  PPID USER     STAT   VSZ %VSZ %CPU COMMAND
13562 13132 root     R     1184   1%   3% top -b -d 1
13131  1096 root     S     1124   1%   2% /usr/sbin/dropbear -F -P /var/run/dropbear.1.pid -p 22 -K 300
13312     2 root     RW       0   0%   2% [kworker/u2:2]
    3     2 root     SW       0   0%   1% [ksoftirqd/0]
 5093     2 root     RW       0   0%   1% [kworker/0:2]
 2308     1 root     D     2296   2%   0% /usr/bin/respondd -d /usr/lib/respondd -d /lib/gluon/respondd -p 1001 -g ff02::2:1001 -i ibss0 -i ibss1 -i br-client -t 10 -g ff05::2:1001 -i br-client -t 10
 1051     1 root     S     1744   1%   0% /sbin/netifd
  824     1 root     R     1716   1%   0% /usr/sbin/haveged -w 1024 -d 32 -i 32 -v 1
 1790     1 root     D     1532   1%   0% /usr/sbin/hostapd -s -P /var/run/wifi-phy0.pid -B /var/run/hostapd-phy0.conf
 1729     1 root     D     1532   1%   0% /usr/sbin/hostapd -s -P /var/run/wifi-phy1.pid -B /var/run/hostapd-phy1.conf
[... 25 lines cut ...]
  329     2 root     SW<      0   0%   0% [ipv6_addrconf]
  230     2 root     SW<      0   0%   0% [bioset]
  225     2 root     SW<      0   0%   0% [bioset]
  235     2 root     SW<      0   0%   0% [bioset]
  614     2 root     SW<      0   0%   0% [cfg80211]
  336     2 root     SW<      0   0%   0% [deferwq]
  210     2 root     SW<      0   0%   0% [bioset]
  165     2 root     SW       0   0%   0% [fsnotify_mark]
  112     2 root     SW       0   0%   0% [kswapd0]
  174     2 root     SW       0   0%   0% [spi0]
  215     2 root     SW<      0   0%   0% [bioset]
   75     2 root     SW<      0   0%   0% [bioset]
   73     2 root     SW<      0   0%   0% [crypto]
   76     2 root     SW<      0   0%   0% [kblockd]
    5     2 root     SW<      0   0%   0% [kworker/0:0H]
   72     2 root     SW<      0   0%   0% [writeback]
  220     2 root     SW<      0   0%   0% [bioset]
  527     2 root     SW<      0   0%   0% [bat_events]
```

I actually triggered this crash "manually" by shutting down the fastd daemon on the offloader the node was connected to while still asking its respondd for data (nodeinfo, statistics and neighbours). It is essential that respondd is actually queried during the disconnect: When doing the s [... cut ...] thing happened.

N.B.: We are running a backported version of gluon-radv-filterd (as is evident from the log and process list). While the previous paragraph indicates that it probably doesn't cause the crash on its own, it might very well have something to do with it. At least that would explain why other communities don't have this problem. And since it is merged by now, even if it is to blaim, we should find and fix it.

If anybody has any idea how to further debug this, I'd be glad for hints.

NeoRaider, 2018-04-07:
Hmm, `kernel: [56425.075256] CPU 0 Unable to handle kernel paging request at virtual address 00000040, epc == 8742a56c, ra == 874200bc` is a NULL pointer dereference in batman-adv kernel code, I don't see any indication of an OOM. Which Gluon version is this?

jplitza, 2018-04-07:
Ah, okay, I'm not very good at reading kernel OOPSes. ^^'
It occurs at least on 2017.1.4 and 2017.1.5 (the only two versions we massively use) and doesn't seem to occur on 2016.2.x.
Also, we're using batman-adv 2013.4.0

rotanid, 2018-04-07:
@jplitza you said that v2016.2.x isn't affected: is radv-filterd also backported to this one?

jplitza, 2018-04-07:
@rotanid Yep. In 2017.1.5 it's the current version with netlink interface, the versions before (since 2016.2) have the debugfs-based version.

@NeoRaider Thanks, will test.

jplitza, 2018-04-07:
@NeoRaider The patch seems to help! Just patched one of the two nodes at home and repeated the experiment. The unpatched node crashed after a while, the patched one still lives.

Is the patch so obvious that this is all the proof you need, or need I gather more evidence?

commit bf611a9a (2018-04-07): modules: update Gluon packages

878f83eb741f batman-adv-legacy: update to latest git

Fixes #1358
closed 2018-04-07 by NeoRaider