freifunk-gluon/gluon #1359: RFC: Adding IPv6 GW to site.conf

A-Kasper, 2018-04-07:
Hi

I would like to set an static IPv6 GW in Clientnetwork to site.conf

why:
I would like to make nodes reachable via Internet without offering routed ipv6 in clientlan.

Idea: If a gw is defined it is binded to br-client as default gw. I think this would be simple to realize.

A-Kasper, 2018-08-29:
maybe another way would be to block ra on client interface if "disable client ipv6" is activated.
since 2018.x releases ipv6 seem to work much better, but i still would like to be able to disable client ipv6

rotanid, 2018-08-30:
we decided to reject this request.
maybe you can fulfil your need by adding a patch to your  own gluon build, but this isn't acceptable upstream.

the present is dual-stack, the future is "IPv6 only" not "IPv4 only"

closed 2018-08-30 by rotanid

mweinelt, 2018-08-30:
https://www.usingenglish.com/reference/irregular-verbs/bind.html