freifunk-gluon/gluon #1364: VXLAN migration path

NeoRaider, 2018-04-11:
At the moment, newly installed nodes will have VXLAN enabled, while upgraded nodes will default to legacy mode. We must decide on a migration strategy.

Ideas:
* Fully disabling legacy mode in a domain config should be possible
* Switching domains should be handled like a new installation and switch to VXLAN
* Should it be possible to disable VXLAN for a specific domain?

lemoer, 2018-04-15:
For reference, manual switching:
```
uci set network.mesh_wan.legacy='0'
uci set network.mesh_lan.legacy='0'
uci commit network
reboot
```

rotanid, 2018-04-15:
> For reference, manual switching:

haven't tested this, but shouldn't `/etc/init.d/network restart` be enough? (instead of a reboot)

mweinelt, 2018-04-15:
If performance is not an issue anymore I'd very much like to have VXLAN enabled when the multidomain feature is used, to prevent bridging, which will most certainly be happening with out.

Non-VXLAN Mesh should be phased out asap.

NeoRaider, 2018-04-16:
Idea:
* Do not make legacy mode user-controllable, but make it a domain option
* Legacy mode will be phased out some time after after v2018.1 (possibly already in the next release)
* Migration could be handled by a scheduled domain switch

I like this scheduled domain switch idea a lot, as it can also be used for a number of other migration scenarios, like switching between IBSS and 11s, or to another wifi channel. Switching routing protocols, e.g. batadv 14 -> 15 or batadv -> babel are not possible though, as we can't install them in the same firmware. We could also do away with IBSS support when we drop VXLAN this way.

I think a scheduled migration is superior to parallel operation of mesh technologies (like IBSS+11s, or VXLAN+legacy), as weak nodes don't deal well with the additional traffic (especially in dense meshes where there are already multiple neighbours without seeing each twice).

One open question would be how to handle the scheduled switch when the time can't be determined because there is no mesh connection. Possible approaches include:
* Signalling between nodes (broadcasts on wired mesh, beacon stuffing for wireless)
* Alternating between both domains periodically (maybe switching every hour?) until an uplink connection is possible again

mweinelt, 2018-04-16:
For nodes without working time we decided to migrate them to the target domain, when they noticed they couldn't reach their gateway for a certain duration.

Adorfer, 2018-04-19:
I would preferr as well the "default config" and "fallback after an hour, if no gateway can be reached". (detecting on "no mesh neighbour" can be hard, if n+1 nodes with 'old config' boot next to each other and by consequence stay on their island forever. On the other hand we do not want to break 'island operation scenario' totally.)

NeoRaider, 2018-04-27:
I think we can split the solution into two parts:

1. Add a site.conf switch to disable VXLAN for a mesh domain; the manual legacy switch is removed. As for the defaults, I'm considering the following:

    * In multidomain setup, VXLAN is enabled by default; it can be disabled expicitly in a domain config ("legacy domain"). I think we should not allow disabling VXLAN for all domains with a single site.conf switch, as such setups will easily lead to bridging of domains.
    * In single domain setup, there is no default: VXLAN must be enabled or disabled in site.conf, the option is required. Enabling VXLAN by default would be problematic, as it would break meshing on updates unexpectedly. Disabling VXLAN by default would be the opposite of the multidomain default, which is unexpected as well.

2. Provide a means of migrating nodes between domains automatically, e.g. scheduled migration.

Only part 1 is a release blocker. Part 2 will need more testing and could be backported to a later v2018.1.x release when it is ready.

commit 1f7ed28b (2018-04-27): gluon-core: set VXLAN/legacy mode in site config

In multidomain setups, VXLAN is enabled by default, but can be disabled in
domain configs using the mesh/vxlan option. In single domain setups, the
mesh/vxlan option is mandatory.

The UCI option for legacy mode is removed.

Fixes #1364