freifunk-gluon/gluon #1367: Support for Litebeam AC Gen2?

bookking3000, 2018-04-14:
Will there be support for Litebeam AC-Gen2 Model? It has got an WA Board. 

WA.v8.5.1# cat /proc/cpuinfo
system type             : Atheros AR934x
processor               : 0
cpu model               : MIPS 74Kc V4.12
BogoMIPS                : 266.24
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 32
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0000, 0x0018, 0x08e0, 0x06a0]
ASEs implemented        : mips16 dsp
shadow register sets    : 1
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available
 

WA.v8.5.1# dmesg
[    0.000000] Linux version 2.6.32.68 (jenkins@db36a17c8935) (gcc version 4.8.3 (OpenWrt/Linaro GCC 4.8-2014.01 unknown) ) #1 Wed Mar 7 15:38:44 EET 2018
[    0.000000] flash_size passed from bootloader = 16
[    0.000000] CPU revision is: 0001974c (MIPS 74Kc)
[    0.000000] ath_sys_frequency: cpu srif ddr srif cpu 535 ddr 400 ahb 200
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] User-defined physical RAM map:
[    0.000000]  memory: 04000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] Zone PFN ranges:
[    0.000000]   Normal   0x00000000 -> 0x00004000
[    0.000000] Movable zone start PFN for each node
[    0.000000] early_node_map[1] active PFN ranges
[    0.000000]     0: 0x00000000 -> 0x00004000
[    0.000000] On node 0 totalpages: 16384
[    0.000000] free_area_init_node: node 0, pgdat 80295020, node_mem_map 81000000
[    0.000000]   Normal zone: 128 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 16256 pages, LIFO batch:3
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 16256
[    0.000000] Kernel command line: console=ttyS0,115200 rootfstype=squashfs init=/init mtdparts=ath-nor0:256k(u-boot),64k(u-boot-env),1024k(kernel),14720k(rootfs),256k(cfg),64k(EEPROM) mem=64M panic=3 quiet
[    0.000000] PID hash table entries: 256 (order: -2, 1024 bytes)
[    0.000000] Dentry cache hash table entries: 8192 (order: 3, 32768 bytes)
[    0.000000] Inode-cache hash table entries: 4096 (order: 2, 16384 bytes)
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Writing ErrCtl register=00000000
[    0.000000] Readback ErrCtl register=00000000
[    0.000000] Memory: 61760k/65536k available (2126k kernel code, 3704k reserved, 514k data, 132k init, 0k highmem)
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS:128
[    0.000000] plat_time_init: plat time init done
[    0.000000] Console: colour dummy device 80x25
[    0.000000] Calibrating delay loop... 266.24 BogoMIPS (lpj=133120)
[    0.022000] Mount-cache hash table entries: 512
[    0.024000] NET: Registered protocol family 16
[    0.027000] PCIe WLAN H/W found [tries: 1]
[    0.027000] PCI init:ath_pcibios_init
[    0.027000] ath_pcibios_init(419): PCI CMD write: 0x356
[    0.027000] registering PCI controller with io_map_base unset
[    0.032000] bio: create slab <bio-0> at 0
[    0.034000] pci 0000:00:00.0: reg 10 64bit mmio: [0x000000-0x1fffff]
[    0.034000] pci 0000:00:00.0: reg 30 32bit mmio pref: [0x000000-0x00ffff]
[    0.034000] pci 0000:00:00.0: supports D1 D2
[    0.034000] Returning IRQ 64
[    0.035000] Switching to clocksource MIPS
[    0.037000] NET: Registered protocol family 2
[    0.038000] IP route cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.039000] TCP established hash table entries: 2048 (order: 2, 16384 bytes)
[    0.039000] TCP bind hash table entries: 2048 (order: 1, 8192 bytes)
[... 625 lines cut ...]
[ 1922.968000] ol_ath_vap_set_param: VAP param unsupported param:33 value:0
[ 1922.968000] ol_ath_vap_set_param: VAP param unsupported param:34 value:0
[ 1923.164000]  ieee80211_ioctl_siwmode: vap->iv_opmode=1, new mode=2, valid=1
[ 1923.165000]  Scan in progress.. Cancelling it
[ 1923.186000] ol_ath_vap_set_param: VAP param unsupported param:35 value:0
[ 1946.331000] Deliver ol stop event, node count: 1
[ 1962.384000] br0: port 2(ath0) entering disabled state
[ 1962.409000] BW divisor: 0X411
[ 1962.446000] BW divisor: 0X411
[ 1962.497000] ADDRCONF(NETDEV_UP): ath0: link is not ready
[ 1962.500000] ol_ath_vap_set_param: VAP param unsupported param:35 value:0
[ 1962.500000] ol_ath_vap_set_param: VAP param unsupported param:33 value:0
[ 1962.500000] ol_ath_vap_set_param: VAP param unsupported param:34 value:0
[ 1962.709000]  ieee80211_ioctl_siwmode: vap->iv_opmode=1, new mode=2, valid=1
[ 1962.710000]  Scan in progress.. Cancelling it
[ 1962.731000] ol_ath_vap_set_param: VAP param unsupported param:35 value:0

WA.v8.5.1# cat /proc/mtd
dev:    size   erasesize  name
mtd0: 00040000 00010000 "u-boot"
mtd1: 00010000 00010000 "u-boot-env"
mtd2: 00100000 00010000 "kernel"
mtd3: 00e60000 00010000 "rootfs"
mtd4: 00040000 00010000 "cfg"
mtd5: 00010000 00010000 "EEPROM"

rotanid, 2018-04-15:
Hi!
as Gluon is based on OpenWrt, we depend on hardware support in OpenWrt.
see also #841 for info about this and other Ubnt AC devices.
currently, there's no support in OpenWrt but someone seems to be working on it:
https://forum.lede-project.org/t/ubiquiti-litebeam-5ac-23-support/10140/4

bookking3000, 2018-05-20:
Hi!
I've managed installing robimarkos' Firmware on the LBE5ac Gen2, it also has the QCA988X as 5Ghz Chip. Currently i have some networking issues with it,  it's not pingable and cant ping other hosts, but i'll look into that

bookking3000, 2018-12-01:
I just installed [this](https://github.com/TobleMiner/openwrt-prereleases-bin/tree/master/litebeam-5ac) one using the tftpboot command from uboot [Intructions](https://openwrt.org/toh/ubiquiti/ubiquiti_nanostation_ac_loco). Everything is working, except the LEDs

closed 2020-03-03 by mweinelt