freifunk-gluon/gluon #1370: issues with upgrading nanostation loco m2 to newer builds (or flashing gluon-factory via tftp)

rotanid, 2018-04-20:
when i upgraded a Ubiquiti NanoStation Loco M2 from a build based on a january commit of v2017.1.x to a march one , the device was "dead".
making it live again worked through flashing Stock AirOS 5.6.15 30572 170328 1107 via TFTP
but flashing a Gluon factory build via TFTP didn't work
also a LEDE 17.01.4 factory image via TFTP didn't work
also a OpenWrt Snapshot factory image via TFTP didn't work
after downgrading to AirOS 5.5.11 from the stock webinterface of AirOS 5.6.15, i could flash the Gluon january build via TFTP again.
afterwards, i could also flash a current build based on the "next" branch of Gluon via TFTP successfully, which didn't work either via TFTP nor sysupgrade when running on the AirOS 5.6.15 bootloader?!?
after flashing this next branch build, i downgraded with "sysupgrade -n" to the v2017.1.x branch successfully
also downgrading to a january v2017.1.x build and upgrading back to a current v2017.1.x build worked now.

i'm not really sure what conclusions to draw here:
- no problems at all when everything started with an AirOS 5.5.11 build
- no gluon factory flash via TFTP when you start with an AirOS 5.6.15 build (firmware check failed)
- soft-brick with newer Gluon builds (started between mid january and mid march) when you start with an AirOS 5.6.15 build

i'm happy to do additional tests, just tell me what to test exactly (no serial console, though!)

the goal should be to fix sysupgrade for devices already out there without having to downgrade AirOS

tackin, 2018-04-20:
Just to narrow the search ... We have a Loco M2 XM with gluon 2017.1.5. It is up and running 4 weeks now.

cajuskamer, 2018-04-22:
With the update from gluon2017.1.5 to gluon2017.1.6 last night, the "Ubiquiti NanoStation loco M2" router went offline. :(

rotanid, 2018-04-22:
@cajuskamer same question to you, do you know which firmware your Loco ran **before** the 2017.1.5 ?

csamsel, 2018-04-22:
a bit offtopic, but i ordered a Loco M2 yesterday for use with gluon / ffac, any tips to not brick it?

cajuskamer, 2018-04-22:
@rotanid the lost routers are on our experimental branch:
gluon 2016.2.7 (stable) -> gluon 0.11.2 ->  gluon 0.11.5 ->  gluon 0.11.6 (offline)
see [firmware ffdo](https://wiki.ffdo.de/Technik/Firmware) for details.

rotanid, 2018-04-23:
sigh... can't reproduce the problem anymore. someone else has to check it...

flashed AirOS 5.6 via TFTP, flashed a v2017.1.x build from november 2017 via TFTP which works great, updated to the january build via "sysupgrade -n" successfully.
a subsequent update to a build from march and one from april does also work with "sysupgrade -n".

flashed AirOS 5.6 via TFTP, flashed a v2016.2.7.1 build via WebUI, which works. updated to the january v2017.1.x build via "sysupgrade -n" successfully.

flashed AirOS 5.6 via TFTP, flashed a v2016.2.7.1 build via WebUI, which works. update to a v2017.1.x build from march via autoupdater successfully.

flashed a FFDO v2016.2.7 build via "sysupgrade -n", update to v2017.1.2, 2017.1.5, v2017.1.6 (each via config mode with keeping the settings) -> no problems

@cajuskamer do you have any idea which AirOS version your device ran when you first flashed it to Gluon? maybe even know if initial Gluon flash was via TFTP or WebUI from AirOS?

rotanid, 2018-04-23:
i managed to reproduce it after neoraider hinted, that flashing AirOS via TFTP doesn't change the bootloader.
after i flashed AirOS 5.6 via the WebUI of AirOS 5.5 i ran on the newer (2015) bootloader. you can check the bootloader version by copying the content of /dev/mtd0 and doing "strings mtd0" on the result. (not possible on the device itself, content has to be copied via scp to a "normal" linux machine)
with the new bootloader, flashing Gluon v2017.1.6 results in soft-brick, no matter if it is via sysupgrade or via the AirOS WebUI with the factory image. Gluon v2017.1.5 doesn't have a problem.

rotanid, 2018-04-23:
we managed to identify 6bd839f361d8210d8a52b3b6948e28aab9275c64 as the commit which introduced the problem.
but we don't know so far, what the problem is and how to fix it, we need a serial console log for this.
until someone supplies this log of a bricked device, i will try builds with newer kernel versions.

2tata, 2018-04-24:
Serial Log from the last working firmware:
 * Firmware-Version: 20180225
 * Gluon-Version: v2017.1.x
 * Commit ID: 64e0e28b24a78473d95ed7ea82ae72ed35265867
 * Download-fastd: https://firmware.ffnw.de/fastd/20180225
 * Download-l2tp: https://firmware.ffnw.de/l2tp/20170225

```
BusyBox v1.23.2 (2017-08-23 11:48:25 CEST) built-in shell (ash)

root@EmdenDevMeshRichtfunk:/# Watchdog handover: fd=3
- watchdog -
Sending TERM to remaining processes ... dnsmasq sh sysupgrade ubus ubusd logd haveged Aug 23 12:07:02 haveged: haveged: Stopping due to signal 15

netifd uhttpd dnsmasq 
Sending KILL to remaining processes ... 
Switching to ramdisk...
Performing system upgrade...
Unlocking firmware ...

Writing from <stdin> to firmware ...     
Appending jffs2 data from /tmp/sysupgrade.tgz to firmware...TRX header not found
Error fixing up TRX header
    
Upgrade completed
Rebooting system...
umount: can't umount /dev: Device or resource busy
umount: can't umount /tmp: Device or resource busy
[  345.120000] reboot: Restarting system

U-Boot 1.1.4.2-s594 (Dec  5 2012 - 15:23:07)

Board: Ubiquiti Networks XM board (rev 1.2 e0a2)
DRAM:  32 MB
Flash:  8 MB
PCIe WLAN Module found (#1).
Net:   eth0, eth1
Hit any key to stop autoboot:  0 
## Booting image at 9f050000 ...
   Image Name:   MIPS LEDE Linux-4.4.116
   Created:      2017-09-25   9:38:49 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1363016 Bytes =  1.3 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

[    0.000000] Linux version 4.4.116 (tata@keks) (gcc version 5.4.0 (LEDE GCC 5.4.0 r3840+42-788312ca59) ) #0 Mon Sep 25 09:38:49 2017
[    0.000000] bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 00019374 (MIPS 24Kc)
[    0.000000] SoC: Atheros AR7241 rev 1
[    0.000000] Determined physical RAM map:
[    0.000000]  memory: 02000000 @ 00000000 (usable)
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] No valid device tree found, continuing without
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x0000000001ffffff]
[    0.000000] Primary instruction cache 64kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, VIPT, cache aliases, linesize 32 bytes
[    0.000000] Built 1 zonelists in Zone order, mobility grouping on.  Total pages: 8128
[    0.000000] Kernel command line:  board=UBNT-BM mtdparts=spi0.0:256k(u-boot)ro,64k(u-boot-env)ro,7552k(firmware),256k(cfg)ro,64k(EEPROM)ro console=ttyS0,115200 rootfstype=squashfs,jffs2 noinitrd
[    0.000000] PID hash table entries: 128 (order: -3, 512 bytes)
[    0.000000] Dentry cache hash table entries: 4096 (order: 2, 16384 bytes)
[... 482 lines cut ...]
[   62.233241] br-client: port 2(local-port) entered forwarding state
[   62.239597] br-client: port 2(local-port) entered forwarding state
[   62.245871] br-client: port 1(bat0) entered forwarding state
[   62.251586] br-client: port 1(bat0) entered forwarding state
[   63.243522] batman_adv: bat0: Changing gw mode from: off to: client
[   63.393071] batman_adv: bat0: gw_sel_class: Changing from: 20 to: 1
[   63.462701] batman_adv: bat0: hop_penalty: Changing from: 30 to: 15
[   63.525936] batman_adv: bat0: orig_interval: Changing from: 1000 to: 5000
[   63.655372] br-wan: port 1(eth0) entered forwarding state
[   64.235593] br-client: port 2(local-port) entered forwarding state
[   64.245378] br-client: port 1(bat0) entered forwarding state
[   65.878894] batman_adv: bat0: Adding interface: br-wan
[   65.884082] batman_adv: bat0: The MTU of interface br-wan is too small (1500) to handle the transport of batman-adv packets. Packets going over this interface will be fragmented on layer2 which could impact the performance. Setting the MTU to 1532 would solve the problem.
[   65.908292] batman_adv: bat0: Interface activated: br-wan
[   66.429592] IPv6: ADDRCONF(NETDEV_UP): client0: link is not ready
[   66.684065] device client0 entered promiscuous mode
[   66.787233] IPv6: ADDRCONF(NETDEV_CHANGE): client0: link becomes ready
[   66.794117] br-client: port 3(client0) entered forwarding state
[   66.800265] br-client: port 3(client0) entered forwarding state
[   68.795346] br-client: port 3(client0) entered forwarding state
[   74.539766] batman_adv: bat0: Adding interface: mesh-vpn
[   74.545182] batman_adv: bat0: The MTU of interface mesh-vpn is too small (1312) to handle the transport of batman-adv packets. Packets going over this interface will be fragmented on layer2 which could impact the performance. Setting the MTU to 1532 would solve the problem.
[   74.569585] batman_adv: bat0: Interface activated: mesh-vpn

```

T-X, 2018-04-24:
Here is the serial output I get from the soft-bricked Ubiquiti Nanostation M2 after I had flashed a Gluon master about 3 weeks ago:

<pre><code>
$ picocom -b 115200 /dev/ttyUSB0 
picocom v2.2

port is        : /dev/ttyUSB0
flowcontrol    : none
baudrate is    : 115200
parity is      : none
databits are   : 8
stopbits are   : 1
escape is      : C-a
local echo is  : no
noinit is      : no
noreset is     : no
nolock is      : no
send_cmd is    : sz -vv
receive_cmd is : rz -vv -E
imap is        : 
omap is        : 
emap is        : crcrlf,delbs,

Type [C-a] [C-h] to see available commands

Terminal ready

U-Boot 1.1.4.2-s956 (Jun 10 2015 - 10:54:50)

DRAM:  32 MB
Flash:  8 MB
PCIe WLAN Module found (#1).
Net:   eth0, eth1
Board: Ubiquiti Networks XM board (rev 1.0 e012)
Hit any key to stop autoboot:  0 
## Booting image at 9f050000 ...
   Image Name:   MIPS LEDE Linux-4.4.120
   Created:      2018-03-04   9:26:34 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1364045 Bytes =  1.3 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

</code></pre>

Here things just seem to hang.
 
With aborted autoboot, here is some information gathered from within U-Boot:

<pre><code>
U-Boot 1.1.4.2-s956 (Jun 10 2015 - 10:54:50)

DRAM:  32 MB
Flash:  8 MB
PCIe WLAN Module found (#1).
Net:   eth0, eth1
Board: Ubiquiti Networks XM board (rev 1.0 e012)
Hit any key to stop autoboot:  0 
## Booting image at 9f050000 ...
   Image Name:   MIPS LEDE Linux-4.4.120
   Created:      2018-03-04   9:26:34 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1364045 Bytes =  1.3 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum ... OK
   Uncompressing Kernel Image ... OK

Starting kernel ...

eth0, eth1
Board: Ubiquiti Networks XM board (rev 1.0 e012)
Hit any key to stop autoboot:  0 
ar7240> 
ar7240> flinfo

Bank # 1: mx25l64 (Id: 0xc22017)
	Size: 8 MB in 128 sectors
ar7240> imls
Image at 9F050000:
   Image Name:   MIPS LEDE Linux-4.4.120
   Created:      2018-03-04   9:26:34 UTC
   Image Type:   MIPS Linux Kernel Image (lzma compressed)
   Data Size:    1364045 Bytes =  1.3 MB
   Load Address: 80060000
   Entry Point:  80060000
   Verifying Checksum ... OK
ar7240> iminfo

## Checking Image at 81000000 ...
   Bad Magic Number
ar7240> version

U-Boot 1.1.4.2-s956 (Jun 10 2015 - 10:54:50)
ar7240> mtdparts

device nor0 <ar7240-nor0>, # parts = 6
 #: name			size		offset		mask_flags
 0: u-boot                	0x00040000	0x00000000	0
 1: u-boot-env            	0x00010000	0x00040000	0
 2: kernel                	0x00100000	0x00050000	0
 3: rootfs                	0x00660000	0x00150000	0
 4: cfg                   	0x00040000	0x007b0000	0
 5: EEPROM                	0x00010000	0x007f0000	0

active partition: nor0,0 - (u-boot) 0x00040000 @ 0x00000000

defaults:
mtdids  : nor0=ar7240-nor0
mtdparts: mtdparts=ar7240-nor0:256k(u-boot),64k(u-boot-env),1024k(kernel),6528k(rootfs),256k(cfg),64k(EEPROM)
ar7240> 
</code></pre>

rotanid, 2018-04-24:
with work from @NeoRaider and tests by @2tata and myself we identified a commit as the possible culprit which was part of the Linux Kernel 4.4.120, probably this commit: https://lkml.org/lkml/2018/3/2/442

NeoRaider, 2018-04-26:
Fixed with 4a2f7945672e5561bd8ce18b2406675720bc5fd8

closed 2018-04-26 by NeoRaider

rotanid, 2018-04-26:
everyone reading this, please test the fix in the v2017.1.x branch as well as in the master branch, so we can be sure the next release doesn't have a problem again

T-X, 2018-04-28:
Looks good here, too! Tested a Ubiquiti NanoStation M2 xm, which I had bricked previously, with a new firmware provided by @mweinelt which should include this fix. And now it's running fine again.

Thanks a lot, @NeoRaider!

tackin, 2018-04-28:
I know, it probably was not affected but my Loco M2 XM with (probably) older bootloader also updated with 2017.1.7 via ssh and autoupdater . But I needed to run autoupdater twice.

Thanks to NeoRaider and his brilliant mind.

rotanid, 2018-04-28:
@tackin the "need to run twice" sometimes happens. i don't have a serial log of these events, but i guess it's due to a safety measure (problem during update, better reboot and try again later than going into an undefined state)