freifunk-gluon/gluon #1371: Block malformed (?) multicast listener query

RalfJung, 2018-04-20:
Starting a week ago, we have seen strange bursts of traffic in our network. After some analysis, the cause turned out to be as follows:
A single client kept sending, with a frequency of around 2sec, a "multicast listener query" with a sender address of `fe80::1`, asking for the multicast address `::`. Every such request probed a "multicast listener report" from pretty much every client in the network, leading to ~200 kBit/s of additional broadcast traffic. Multiplied by 500 nodes, this explains the ~100 MBit/s of additional traffic we have been seeing.

Is it possible to have the nodes block such traffic before it enters the mesh? I doubt a multicast address of `::` is valid.

Also, we have no idea if the client doing this is malicious or just broken. It's MAC starts with 4c:09:d4, which belongs to [Arcadyan](https://de.wikipedia.org/wiki/Arcadyan), so this seems to be some kind of active network device and not a laptop or phone. Maybe.

T-X, 2018-04-20:
Hi @RalfJung,

> Is it possible to have the nodes block such queries before they even enter the mesh? I doubt a multicast address of :: is valid.

Yes! We have gluon-ebtables-segment-mld for exactly that:

https://gluon.readthedocs.io/en/v2017.1.x/package/gluon-ebtables-segment-mld.html

With the next release (or in master already) this behavior will become the default one with no more need to add this package to the site.mk anymore. However, for v2017.1.x you still need to add it which is highly recommended.

RalfJung, 2018-04-20:
Seems like that will disable multicast group stuff entirely? Fair enough, if that's not needed for basic network functionality this is probably for the better. Thanks, firmware with this package is currently being built!

T-X, 2018-04-20:
No, we do not disable it entirely. Instead we mark the mesh/bat0 side as a "multicast router port" and all local multicast traffic to the mesh will be unconditionally handed over to batman-adv or ebtables then. And then we let those figure out what to do about it.

Yes, absolutely, MLD is required for IPv6 to work probably when snooping switches/bridges are involved. But we should be good to go with this multicast router port setting. Sorry, I still need to document the current architecture concerning multicast in Gluon with batman-adv.

RalfJung, 2018-04-20:
> Sorry, I still need to document the current architecture concerning multicast in Gluon with batman-adv.

It's okay, I'm going to wait for that. :)  Please also explain there how the gateways have to be configured, in particular the bridge that bat0 is in (if any); I imagine that's also relevant and right now I am just guessing that configuration.