freifunk-gluon/gluon #1378: HW support for GL.iNet AR-750

Commifreak, 2018-05-03:
-> https://www.gl-inet.com/ar750/

OpenWRT supports it: https://openwrt.org/toh/hwdata/gl.inet/gl.inet_gl-ar750 since commit: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=2e5252d346e2ec832a203af778b5c1d949f0ae5f

Is there a chance that gluon/FF support it?

mweinelt, 2018-05-03:
OpenWrt support was added after LEDE 17.01 and has not been backported. Therefore support is only possible until after Gluons master branch was upodated to the next OpenWrt release or if the device support was backported to 17.01, but it is quite late for that imo.

mweinelt, 2018-05-04:
If the device was in OpenWrt master you will need to do a backport to lede-17.01 first.

achterin, 2018-05-04:
Why would I have to do that? Right now the backport for "TP-Link Archer C7 v4" is almost accepted and the code for this comes straight from OpenWrt master as well.
Sorry if this is a dumb question as I am fairly new to the gluon development.

rotanid, 2018-05-04:
@mweinelt as upstream does reject almost any hardware related backport, we don't apply this restriction as far as i know.

Commifreak, 2018-05-04:
@achterin I've tried ti build gluon with the ffharz site:

```
make GLUON_TARGET=ar71xx-generic GLUON_REGION=eu
```

which exited normally. I've got several images now, but none that has AR750 in it. Where do I find an image for the AR750 to flash it with gluon? Am I missing something?

...time passes...

Got it! ->
```
./output/images/sysupgrade/gluon-ffharz-0.9.1~20180504-gl-ar750-sysupgrade.bin
```

But thats for sysupgrade only, where's the factory image?

pbriesch, 2018-05-04:
So far as I know there is OpenWrt/LEDE preinstalled. Therefore no factory image is needed, only sysupgrade.

Commifreak, 2018-05-04:
Yep, just read that on the openwrt device description. I'll give it a try. Stay tuned ;)

achterin, 2018-05-08:
I think I've messed up the port config. There was an openwrt change on master, that was not applied onto lede-2017.1, that changed the port naming.
I've updated my branch and it would be nice if you could rebuild and test again.
Sorry for the trouble.

Commifreak, 2018-05-08:
No problem, I build it again asap and let you know if it works!

Commifreak, 2018-05-10:
Tested it. Updated local repo by using git pull. The new built image also does not work :(

Still no connection to the device (neither dhcp nor static address)

What else could now be wrong?

Commifreak, 2018-05-10:
It is working fine!

[image]

I'll test some functions, but it seems to work well. Thanks for your help!

Maybe this patch is included in gluon soon ;)

achterin, 2018-05-10:
Awesome. Before the backport to v2017.1.x will be merged, I'll have to do another backport to master first.
I will submit a PR tomorrow and it would be nice if you could build and test that as well.

Commifreak, 2018-05-11:
Why do I have to update site conf? I would copy the ffharz as usual or do I need to adapt something?

achterin, 2018-05-11:
Because master uses some renamed and new config values. I had to update our site.conf and site.mk as well. Otherwise you will get compile errors. If you need help, hop onto irc and I'll help you.

Commifreak, 2018-05-14:
Thanks.

Where are those change documented? The test-build from your repo was GLUON 2017.1.7, master is currently 14 days beyond 2017.1.7 - what do I need to adapt [here](https://github.com/ffharz/site-ffharz) to get this site working on a test build based on [your master backport](https://github.com/achterin/gluon/tree/gl-ar750-backport-master)?

achterin, 2018-05-14:
I did the following changes to ffharz site config to compile run without errors:

site.conf
- rename 'site_seed' => 'domain_seed'
- add 'vxlan' option to 'mesh'
- remove 'gluon_ssid_changer' package because it needs changes

site.mk
- rename 'GLUON_ATH10K_MESH' => 'GLUON_WLAN_MESH'

i uploaded a gist containing the diff: https://gist.github.com/achterin/5589e72a7ea9561599b29e662a710e8a

achterin, 2018-05-14:
I checked this [forum thread](https://forum.freifunk.net/t/aenderungen-an-gluon-beim-wechsel-zu-v2018-1-x/16340/32). I dont know if there is a documentation available right now as the current master has no official release yet. Maybe @NeoRaider or @rotanid can help.

Commifreak, 2018-05-14:
Oh, thanks. Informative thread.

rotanid, 2018-05-15:
let's stick with the forum thread until there's a major release ( v2018.1 ) with release notes

PR #1395 (2018-05-17): ar71xx-generic: Add support for GL.iNet GL-AR750
This PR backports hw support for GL.iNet GL-AR750.
The backport against 2017.1.x got already tested by @Commifreak and he will test this patchset asap. As of now, this is should be seen as untested until he reports back.

Fixes #1378
closed 2018-05-17 by rotanid