freifunk-gluon/gluon #1380: batman-adv: broken TT announcements with empty vlans

lemoer, 2018-05-03:
Short story:
- Some (affected) nodes send TT announcements for an empty vlan with vlan_id = 0.
    - This behaviour is a bug! Batman-adv should not send TT announcements for empty vlans.
- The receiving batman-adv can not handle incomming TT announcements with empty vlans. (The crc check routine *batadv_tt_global_check_crc()* ) fails.
- As the receiving node thinks, the crc is broken, it does a full table request then.
- The announcing node sends a TT full table to the receiving node then, which also contains the empty vlan, so *batadv_tt_global_check_crc()* fails again on the receiver side.
- This causes a lot of unneeded TT traffic.

Script to check how many announcing nodes are affected:
``` shell
#!/bin/sh

origs=$(batctl o | tr -s '\*' ' ' | cut -d ' ' -f 2 | sort | uniq)
tg=$(batctl tg)

for o in $origs; do
	if echo -n "$tg" | grep $o | grep -v '\-1'>/dev/null; then
		echo ok
	else
		echo affected
	fi
done
```

Hotfix which fixes *batadv_tt_global_check_crc()* on the receiving nodes (which does not fix the sending issue):
``` patch
From f23a9d3f375dcf8e57197480ac260baf06738074 Mon Sep 17 00:00:00 2001
From: root <root@sn07.s.ffh.zone>
Date: Mon, 23 Apr 2018 02:24:43 +0200
Subject: [PATCH] batman-adv: fix bug when vlan crc is 0x00000000

---
 net/batman-adv/translation-table.c | 8 ++++++--
 1 file changed, 6 insertions(+), 2 deletions(-)

diff --git a/net/batman-adv/translation-table.c b/net/batman-adv/translation-table.c
index 7615f237..677dc5eb 100644
--- a/net/batman-adv/translation-table.c
+++ b/net/batman-adv/translation-table.c
@@ -3007,8 +3007,12 @@ static bool batadv_tt_global_check_crc(struct batadv_orig_node *orig_node,
 
 		vlan = batadv_orig_node_vlan_get(orig_node,
 						 ntohs(tt_vlan_tmp->vid));
-		if (!vlan)
-			return false;
+		if (!vlan) {
+			if (tt_vlan_tmp->crc == 0x00000000)
+				continue;
+			else
+				return false;
+		}
 
 		crc = vlan->tt.crc;
 		batadv_orig_node_vlan_put(vlan);
-- 
2.11.0

```
[-> This patch is also available as patch for gluon](https://raw.githubusercontent.com/freifunkh/site/lede/patches/0002-fix-when-vlan-crc-is-00000000.patch)

Comparison before and after the hotfix patch on a node:
```
Before:
=======

root@lemoer-nightly:/tmp# (batctl s; sleep 10; batctl s;) | grep tt_request_tx
	tt_request_tx: 2837110
	tt_request_tx: 2837306
root@lemoer-nightly:/tmp# (batctl s; sleep 10; batctl s;) | grep tt_request_tx
	tt_request_tx: 2837368
	tt_request_tx: 2837542

=> ~200 req/s

After:
========

root@lemoer-nightly:/lib/gluon# (batctl s; sleep 10; batctl s;) | grep tt_request_tx
	tt_request_tx: 7806
	tt_request_tx: 7813
root@lemoer-nightly:/lib/gluon# (batctl s; sleep 10; batctl s;) | grep tt_request_tx
	tt_request_tx: 7815
	tt_request_tx: 7822

=> < 1 req/s
```

Comparison before and after hotfix patch on the supernodes. Patch was applied on the freifunk nodes after 4/23 in evening:  
[image]

Status:
- Needs upstream patch.

rotanid, 2018-05-11:
possibly fixed in upstreams maint branch with https://git.open-mesh.org/batman-adv.git/commit/e4687b4be274da6180fc15b327419851fb681ec9
but i'm not 100% sure if that is the related fix...

lemoer, 2018-05-11:
Yes. That's one related fix. But this only supresses the emission of broken packets.

When a certain amount of nodes, which are not upgraded yet, exist, the ogm receivers will still send a lot of fulltable requests. So a second patch (like the one in my first message) on the receivers might help here by ignoring those broken packets. It's not clear yet, whether such a patch will be accepted upstream.

ecsv, 2018-06-04:
> possibly fixed in upstreams maint branch with https://git.open-mesh.org/batman-adv.git/commit/e4687b4be274da6180fc15b327419851fb681ec9
but i'm not 100% sure if that is the related fix...

This patch was added to 2017.1.x via #1412. In master, it was added via 184dc543174da96651b00c0db9a8bd657e78dfca.

In next it is still pending. Would be fixable by updating routing feed to https://github.com/openwrt-routing/packages/commit/013dcc6817eac52d7684667dccac0c27b7607ecf (master branch) or https://github.com/openwrt-routing/packages/commit/a22f490620e72c7bae7ff20fe1c67da10b84013b (openwrt-18.06 branch).

NeoRaider, 2018-06-04:
I just rebased next onto master.

closed 2018-06-04 by NeoRaider