freifunk-gluon/gluon #1381: No wifi connection via Freifunk on 11s (TP-Link C59)

mokkin, 2018-05-03:
Private WLAN is working
Connection via WLAN on Freifunk AP not. After pressing "wifi" command it works again.
Firmware: http://firmware.leipzig.freifunk.net:8004/srv2/gluon/20180401/
(based on OpenWRT Trunk)
btw: 5 GHz mesh is disabled due to unstable ath10k

```
[542421.723705] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542422.723895] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542423.723777] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542424.724154] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542425.724058] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542426.724432] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542427.724478] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542428.724487] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542429.724587] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542430.724737] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542431.724857] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542432.725052] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542433.725044] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542434.725145] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542435.725480] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542436.725345] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542437.725520] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542438.725516] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542439.725643] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542440.726063] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542441.725872] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542442.726236] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542443.726375] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542444.726480] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542445.726506] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542446.726735] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542447.726583] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542448.726746] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542449.726900] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542450.727096] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542451.727025] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542452.727208] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542453.727516] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542454.727500] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542455.727710] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542456.727599] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542457.727660] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542458.457277] ath10k_pci 0000:00:00.0: No VIF found for vdev 1
[542458.463234] ath10k_pci 0000:00:00.0: received addba event for invalid vdev_id: 1
[542458.474536] ath10k_pci 0000:00:00.0: peer-unmap-event: unknown peer id 2
[542458.492444] br-client: port 5(client0) entered disabled state
[542458.503710] device client0 left promiscuous mode
[542458.508801] br-client: port 5(client0) entered disabled state
[542458.531325] br-client: port 4(client1) entered disabled state
[542458.570237] device client1 left promiscuous mode
[542458.575240] br-client: port 4(client1) entered disabled state
[542458.727940] ath10k_pci 0000:00:00.0: Spurious quick kickout for STA cc:fd:17:ec:a3:be
[542458.858176] device wlan0 left promiscuous mode
[542458.863116] br-wan: port 3(wlan0) entered disabled state
[542458.906145] ------------[ cut here ]------------
[542458.911219] WARNING: CPU: 0 PID: 2511 at backports-2017-11-01/net/mac80211/sta_info.c:940 ieee80211_sta_eosp+0x65c/0xa34 [mac80211]
[542458.923600] Modules linked in: ath9k nf_conntrack_ipv6 iptable_nat ipt_REJECT ipt_MASQUERADE ebtable_nat ebtable_filter ebtable_broute ath9k_common ath10k_pci ath10k_core xt_time xt_tcpudp xt_state xt_quota xt_pkttype xt_physdev xt_owner xt_nat xt_multiport xt_mark xt_mac xt_limit xt_conntrack xt_comment xt_addrtype xt_TCPMSS xt_REDIRECT xt_LOG xt_CT nf_reject_ipv4 nf_nat_redirect nf_nat_masquerade_ipv4 nf_conntrack_ipv4 nf_nat_ipv4 nf_nat nf_log_ipv4 nf_defrag_ipv6 nf_defrag_ipv4 nf_conntrack_rtcache nf_conntrack mac80211 iptable_mangle iptable_filter ip_tables ebtables ebt_vlan ebt_stp ebt_snat ebt_redirect ebt_pkttype ebt_mark_m ebt_mark ebt_limit ebt_ip6 ebt_ip ebt_dnat ebt_arpreply ebt_arp ebt_among ebt_802_3 br_netfilter ath9k_hw ath em_cmp em_nbyte sch_dsmark sch_pie sch_gred em_meta sch_teql
[542458.997255]  act_ipt sch_multiq sch_prio em_text sch_codel sch_red sch_sfq sch_fq act_skbedit act_mirred em_u32 cls_u32 cls_tcindex cls_flow cls_route cls_fw sch_htb sch_hfsc ledtrig_usbport batman_adv libcrc32c cfg80211 compat ip6t_REJECT nf_reject_ipv6 nf_log_ipv6 nf_log_common ip6table_mangle ip6table_filter ip6_tables x_tables dummy ip_tunnel veth tun vxlan udp_tunnel ip6_udp_tunnel ehci_platform ehci_hcd act_police cls_basic sch_tbf sch_ingress gpio_button_hotplug usbcore nls_base usb_common crc16 crc32c_generic crypto_hash
[542459.044716] CPU: 0 PID: 2511 Comm: hostapd Tainted: G        W       4.9.87 #0
[542459.052269] Stack : 804e765a 00000042 00000000 00000001 8658d724 8046b327 8041e800 000009cf
[542459.061057]         804e37a0 000003ac 87658fcc 00000001 87658fcc 800a9240 80424030 80470000
[542459.069847]         00000003 000003ac 804221e8 86671a7c 87658fcc 800d6f6c 804e765a 00000077
[542459.078637]         8047d000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[542459.087427]         00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
[542459.096217]         ...
[542459.098843] Call Trace:
[542459.101459] [<8006b0f0>] show_stack+0x70/0x8c
[542459.106073] [<80080064>] __warn+0x110/0x118
[... 116 lines cut ...]
[542468.437877] IPv6: ADDRCONF(NETDEV_UP): mesh1: link is not ready
[542468.775171] IPv6: ADDRCONF(NETDEV_CHANGE): mesh1: link becomes ready
[542469.942797] batman_adv: bat0: Adding interface: mesh1
[542469.948190] batman_adv: bat0: Interface activated: mesh1
root@E09-HH-2OG-1:~# wifi
root@E09-HH-2OG-1:~# wifi
root@E09-HH-2OG-1:~# 
root@E09-HH-2OG-1:~# 
root@E09-HH-2OG-1:~# 
root@E09-HH-2OG-1:~# 
root@E09-HH-2OG-1:~# packet_write_wait: Connection to 192.168.1.xxx port 22: Broken pipe
user@pc:~$ ssh root@192.168.1.xxx

BusyBox v1.27.2 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt SNAPSHOT, r6567+7-db893ec
 -----------------------------------------------------
root@E09-HH-2OG-1:~# 
```

rotanid, 2018-05-03:
"based on OpenWrt trunk" - so you're running a firmware based on the "next" branch of Gluon?
if yes, could you please report if the issue also affects you if you user Gluons "master" branch?
so far, the master branch is our main focus

mokkin, 2018-05-04:
As far as I know, the master branch is based on LEDE 17.01 release. In this release the device (Archer c59) wasn't included, so I decided to compile gluon-next with OpenWRT to receive an image for the device and having the latest bugfixes.
Should I try it again with master?

rotanid, 2018-05-04:
we have support in the master branch:
https://github.com/freifunk-gluon/gluon/blob/master/targets/ar71xx-generic#L197
it was integrated with this PR in January:
https://github.com/freifunk-gluon/gluon/pull/1281

rotanid, 2018-05-25:
as at least three devices of the model Archer C59 are running here with master branch without problems, the problem might be in the "next" branch only or related to the specific site.conf/site.mk of @mokkin

mokkin, 2018-06-12:
I have built a new gluon based on master as requested. I'll let you know about the results.

rotanid, 2018-07-12:
@mokkin any update on this? can we close the issue?

mokkin, 2018-07-12:
Actually not. We are still observing the issue.

blocktrron, 2018-07-12:
@mokkin can you observe this issue on multiple devices? I've tried an C58 and a C59 and was not able to reproduce this issue. Is it working without Private WiFi enabled?

closed 2018-08-22 by rotanid