freifunk-gluon/gluon #1382: batman-adv: TT sync issues, 'WI' flags on multicast TT, continued

T-X, 2018-05-04:
In @lemoer's setup it was observed that multicast entries with both a 'W' and 'I' flag were still spreading even though (most) nodes were having the according [TT sync fix](https://git.open-mesh.org/batman-adv.git/commit/382d020fe3fa528b1f65f8107df8fc023eb8cacb).

It seems that the TT sync fix so far only fixed direct TT responses, but not intermediate TT responses. The issue seems to lie in batadv_tt_tvlv_generate() here:

https://elixir.bootlin.com/linux/latest/source/net/batman-adv/translation-table.c#L2955

There the mangled flags are still used which is wrong.

Status:
* I was able to reproduce the issue in a 4 node setup
* [This](https://github.com/FreifunkVogtland/batman-adv/commit/83fea7fd23b97a3613b79d3be51959620a0c574a) hotfix/workaround from @ecsv is currently applied in @lemoer's setup at Freifunk Hannover to mitigate this issue
* A proper fix is WIP

T-X, 2018-05-05:
A fix for batman-adv was suggested here: https://patchwork.open-mesh.org/patch/17336/

rotanid, 2018-05-11:
fixed in upstreams maint branch: https://git.open-mesh.org/batman-adv.git/commit/d65daee8617b29c1ddcc949ce3a5ec24f7a1e1af

ecsv, 2018-06-04:
This patch was added to 2017.1.x via #1412. In master, it was added via 184dc543174da96651b00c0db9a8bd657e78dfca.

In next it is still pending. Would be fixable by updating routing feed to 013dcc6817eac52d7684667dccac0c27b7607ecf (master branch) or a22f490620e72c7bae7ff20fe1c67da10b84013b (openwrt-18.06 branch).

NeoRaider, 2018-06-04:
I just rebased next onto master.

closed 2018-06-04 by NeoRaider