freifunk-gluon/gluon #1415: gluon-status-page: show additional info like site, domain, etc.

rotanid, 2018-06-04:
inspired by #1402 
show the pretty names for additional data (site, domain, roles), and hide the labels for unavailable fields

mweinelt, 2018-11-26:
Domain pretty names is pretty expensive right now. We get the value in `nodeinfo.system.domain_code` and have to look at all files in `/lib/gluon/domains/*.json`.
A package that does this is `gluon-config-mode-domain-select` but that's because it shows all pretty names anyway.

rotanid, 2018-11-26:
showing the non-pretty names would still be an improvement ;)

mweinelt, 2018-11-26:
Really? `ffda_da_220_230` doesn't sound like anything I'd put on a status page.

rotanid, 2018-11-26:
more than nothing, although not ideal.
judging from our experience, no one except people with deep technical knowledge uses the status page anyway...

ambassador86, 2018-11-27:
I am preparing a patch for this in https://github.com/FreifunkChemnitz/gluon/tree/status-page-extension I will add a PR as soon as I am happy with it.

mweinelt, 2018-12-18:
If traffic limits are in place it would be great to publish them on the status-page as well.

rotanid, 2019-05-27:
added wifi channels, from #1727

lemoer, 2020-01-09:
@rotanid nexthop is done

AiyionPrime, 2021-03-09:
Something I lacked in the last years  and compensated by memorizing more and more router models was the base frequency of a nodes processor, as well as his installed ram. Is that something someone else missed as well?

@rotanid May you explain what `roles` is expected to contain?
Like whether a node is a supernode or not, based on `nodeinfo.vpn`?

AiyionPrime, 2021-03-11:
TIL: site.conf provides a 'roles' parameter:

> Nodes will announce their role inside the mesh. This will allow in the backend to distinguish between normal, backbone and service nodes or even gateways (if they advertise that role). It is up to the community which roles to define.

example:
```
roles = {
  default = 'node',
  list = {
    'node',
    'test',
    'backbone',
    'service',
  },
},
```

https://gluon.readthedocs.io/en/latest/user/site.html

AiyionPrime, 2021-04-12:
Huh, seems like I did not commit the related changes back then. Sorry.
Find the implemenatation of `system.role` in https://github.com/freifunk-gluon/gluon/pull/2200.

Adorfer, 2021-05-02:
please add **"batman  MainIF/MAC"**
(nodeid is in most (all?) cases different, many batctl-functions require the MainIF-MAC, not the one from the status page)

Adorfer, 2021-05-02:
please add 
- "Mesh-Interface-Names"  (to see if MoW/MoL is enabled, or if mesh on radio0/1 is on)
- "Client interface names" (so see if Lan-Client or AP on radio0/1 is enabled)

blocktrron, 2021-05-02:
> or if mesh on radio0/1 is on

This is already provided by the mesh signal histogram

AiyionPrime, 2021-05-02:
@rotanid 
> roles

is done since 490c522351e237ad43ab06c9d0bb8774feaca01e and

> wifi channels

as well: 0ce961eb8c27df967e2dc3599ed2518a1d2e0faf.

AiyionPrime, 2021-06-07:
@rotanid
> VPN type (active / fastd / l2tp / ...)

is done since 16c5c64141ff0147943408cf7483ac64caa4479a

PR #2217 (2021-06-07): gluon-status-page: add bandwidth limit
**blocked** by #2216

There will be a bandwidth limit on the statuspage, eventually.

And likewise this might finally resolve #1415 .

This router has the necessary commits applied:
http://meshvpnabstracttest.n.ffh.zone/cgi-bin/status or
http://[2001:678:978:114:b64b:d6ff:fe22:4ebc]/cgi-bin/status

[image]

I implemented a bit-formatting function that looks like this, whether this should go in statuspage or a util file is up to decide.

```lua
local function formatBits(v, factor)
    --to format a value given in kilobits use a factor of 1000
    factor =  [... cut]
closed 2021-06-07 by blocktrron