freifunk-gluon/gluon #1417: CPE210 sysupgrade renders device into paper weight

1977er, 2018-06-05:
Upgrading a CPE210 to the current gluon master (03 Jun 2018) puts the device into a bootloop with no way to restore an old firmware.

User reports a "Data loaded outside the reserved load area" message on a normal boot. Trying to flash a different firmware (stock firmware or older known working gluon) via tftp is quitted with "incorrect file".

The filesize of the sysupgrade file is suspiciously larger than the factory and larger than prior sysupgrade files. (4 MB vs. 8 MB) @rotanid reports equally large filesizes in gluon-next.

[image]

rotanid, 2018-06-05:
file sizes with our builds:
v2017.1.x factory: 4.38M
v2017.1.x sysupgrade: 4.5M
master factory: 4.2M
master sysupgrade: 7.44M
next factory: 4.41M
next sysupgrade: 7.44M

this effect is the same for OpenWrt images, comparing lede 17.01.4 release with current snapshot:
https://downloads.openwrt.org/releases/17.01.4/targets/ar71xx/generic/
https://downloads.openwrt.org/snapshots/targets/ar71xx/generic/

NeoRaider, 2018-06-05:
The increased fimeware size is expected, as we reorganized the partition layout. The smaller factory is normal as well (the sysupgrade image is a plain flash dump, while the factory image stores each partition individually).

NeoRaider, 2018-06-05:
Which CPE210 revision is this? We've had reports before that TFTP recovery using stock/factory images is broken since version 1.1 (possibly there is a special recovery firmware format now?).

Can you give me the Gluon image that caused this issue?

1977er, 2018-06-05:
The user report dates back to May 24th, but unfortunately we do not archive our nightly builds. But I guess the current nightly is affacted as well. Try this one: https://build.ffh.zone/job/gluon-nightly/ws/download/images/sysupgrade/

1977er, 2018-06-05:
In two cases it was a CPE210 v1.1.

NeoRaider, 2018-06-05:
I am working on a fix for the issue, I hope to finish it by tomorrow.

hanmob, 2018-06-06:
As a trial:
 CPE210 V1.0 and CPE510 V1.0 has been sysupgraded with vHN20180603 from above source.
 Anormal status verified : missing ipv4 config page  and no reply onping6 reply ff02::1
 Recovery using old factory image  TP_PharOS_1.3.3_160705 and a switch between LAN1 and tftpd worked. Then vHN20180603 factory could be flashed and looks like working well.

commit 0563473a (2018-06-06): ar71xx: fix boot with increased kernel size on CPE/WBS 210/510

Fixes: d139a13563ef ("Add various patches to deal with bigger kernels")
Fixes #1417
closed 2018-06-06 by NeoRaider