freifunk-gluon/gluon #1419: batman-adv: vanishing multicast TT entries

T-X, 2018-06-05:
@lemoer has reported the following, additional problem here:

* [Missing TT entry leads to inconsistent but stable TT state](https://www.open-mesh.org/issues/355)

T-X, 2018-06-08:
Patches are available in the ticket referenced above. Needs testing and reviewing now.

ecsv, 2018-06-13:
The openwrt-routing PRs with the mentioned patches were merged:

* https://github.com/openwrt-routing/packages/pull/384 
* https://github.com/openwrt-routing/packages/pull/385
* https://github.com/openwrt-routing/packages/pull/386

gluon's master and next branch have to adjusted to point to a newer revision. Right now, they point to 013dcc6817eac52d7684667dccac0c27b7607ecf (older openwrt-master). So they most likely have to change their modules' PACKAGES_ROUTING_COMMIT to point to the current master f785dc0e9b19baf121f3440ef995d58fa18e5ef1

rotanid, 2018-06-13:
i'm running PR #1431 on 4 nodes at the moment, tomorrow there will be at least 50

rotanid, 2018-06-13:
fixed by #1431 in v2017.1.x branch and 046103e5a25607a54e4ceb996275e067028b30ba in master branch

closed 2018-06-13 by rotanid