freifunk-gluon/gluon #1433: Pull Request Template

mweinelt, 2018-06-13:
We need a pull request template to improve on applying our guidelines and better able to track progress of pull requests.

@skorpy2009 and @rotanid wanted to take care of that. I recommend HackMD to collaborately build the text and neoraider recommended to use HTML comments to hide explanations etc..

skorpy2009, 2018-07-12:
This depends on #1438 & #1434

mweinelt, 2018-07-13:
GitHub allows for several different types of ISSUE_TEMPLATEs, probably the same for pull requests, so we don't need to cram everything into a single template.

https://blog.github.com/2018-05-02-issue-template-improvements/

via https://github.com/ffrgb/meshviewer/tree/develop/.github/ISSUE_TEMPLATE

rotanid, 2018-08-08:
@mweinelt you can only add a single PR template at the moment, as far as i can see it

mweinelt, 2018-08-08:
Sure, but better start somewhere, GitHub will follow at some point.

rotanid, 2018-08-08:
sure, feel free to do a pull request adding templates ;)

christf, 2019-02-27:
which problem are we addressing with these templates?

mweinelt, 2019-02-27:
Lack of information and structure in pull requests. Like when you add a new device you'd get the device checklist by default. 

The issue is that we currently can only have on PR template as it looks.

mweinelt, 2022-03-26:
This did not pan out, which is fine, since nobody seems to be complaining about the current state of PR descriptions.

closed 2022-03-26 by mweinelt