freifunk-gluon/gluon #1437: guidelines for Gluon

rotanid, 2018-06-13:


mweinelt, 2019-06-02:
Lua:
- names: lowercase, connected through underscores (`some_variable`)

bobcanthelpyou, 2019-06-02:
There are some short guidelines hidden in the docs, already:

Development Guidelines
----------------------
lua should be used instead of sh whenever sensible. The following criteria
should be considered:

- Is the script doing more than just executing external commands? if so, use lua
- Is the script parsing/editing json-data? If so, use lua for speed
- When using sh, use jsonfilter instead of json_* functions for speed

Code formatting may sound like a topic for the pedantic, however it helps if
the code in the project is formatted in the same way. The following rules
apply:

- use tabs instead of spaces
- trailing whitespaces must be eliminated

https://gluon.readthedocs.io/en/latest/dev/basics.html#development-guidelines
https://github.com/freifunk-gluon/gluon/blob/master/docs/dev/basics.rst#development-guidelines

mweinelt, 2020-04-11:
Think we codified this in the .editorconfig file mostly. I believe the documentation was updated with that in mind. Closing.

closed 2020-04-11 by mweinelt