freifunk-gluon/gluon #1447: simple-tc reduces bandwidth far below given limit

RalfJung, 2018-06-26:
I have set
```
uci set simple-tc.mesh_vpn.limit_ingress=10000
uci set simple-tc.mesh_vpn.limit_egress=1000
uci commit simple-tc && /etc/init.d/tunneldigger restart
```
With these settings, when I do a speedtest, I get <3 MBit/s download speed (and ~1 MBit/s upload as expected).

With an ingress limit of `30000`, download speed increases to ~5 MBit/s.

When I turn off simple-tc, I get 25 MBit/s for downloads (my DSL line has 30 MBit/s so that's as expected).

Seems like simple-tc reduces the download speed much further than it should? I had a look at the device load during these tests, but there was always >50% idle.

This is with a WR841N/ND v10 and a firmware based on Gluon v2018.1.8, in a tunneldigger/l2tp-based network.

NeoRaider, 2018-06-26:
TL;DR: ingress filters are bad, the behaviour is known.

We don't have any way to control how fast a VPN peer sends traffic; such a feature would need to be implemented in the VPN software, or possibly a layer on top of the VPN that transmits the limit and installs an egress filter on the other side.

The only thing our ingress filter does is to drop packets when we are over the limit, and hope that TCP connections behave sanely and reduce throughput. For non-TCP protocols, the situation is even worse, as they often do not have any kind of control (so they will simply experience packet loss when the ingress filter kicks in). And even for TCP, behaviour in the presence of packet loss can be very bad with some TCP implementations, especially when there are multiple parallel TCP streams.

RalfJung, 2018-06-27:
I see. So what one would want is for tunneldigger to tell the broker on the gateway side to please do egress filtering. I can see how that makes more sense. :)
I think I even saw TODOs for that somewhere in the tunneldigger sources or so.

I can look into adding support for that in tunneldigger. Does simple-tc have support for cooperating with the VPN client in that regard?

kaechele, 2018-06-29:
Tunneldigger already has that support (-L flag on the client makes the server set the requested downstream bandwidth limit). It's just that the support on the Gluon side is missing to actually pass that flag to tunneldigger when
1. a limit is set in config mode and
2. tunneldigger is the method used for VPN.

RalfJung, 2018-07-26:
Awesome, thanks a lot!