freifunk-gluon/gluon #1452: gluon-status-page: neighbours on wired and vpn meshes are not visible

lemoer, 2018-06-30:
[image]

As the network inspector (in top right of the screenshot) shows, the neighbour info is not loaded at all. When the event stream is accessed manually by curl (as seen in bottom right) everything is returned correctly. So the issue should be located somewhere in the javascript code.

lemoer, 2018-06-30:
[-> Link to the Router](http://[2a02:790:ff:1019:e068:b9ff:fe97:a786]/cgi-bin/status)

lemoer, 2018-06-30:
Browsers I tried:
- Firefox 59.0.2 (64-bit)
- Chromium Version 66.0.3359.139

tobby88, 2018-06-30:
Looks like the link to the router doesn't work - maybe this one does :) http://[2a02:790:ff:1019:e068:b9ff:fe97:a786]/cgi-bin/status

NeoRaider, 2018-06-30:
Your node is missing the *gluon-status-page-mesh-batman-adv* package. Either add it explicitly, or add both *status-page* and *mesh-batman-adv-15* (or -14) to GLUON_FEATURES.

It is interesting that it doesn't even show the names of the neighbour nodes; the respondd nodeinfo collection should work independently of the batadv-specific components. I'm leaving this open as a reminder to look into this, but I don't consider it release-critical.

rotanid, 2018-06-30:
i just tried out the most-recent master branch and the status page is correctly showing the VPN and wired-mesh partners.

lemoer, 2018-06-30:
Ah, I think we missed to add the package!

Two questions arise for me:
- Maybe we should ensure that *gluon-status-page* can't be installed without either *gluon-status-page-mesh-batman-adv* or *gluon-status-page-mesh-babel*?
- Is there an option, so we could make features "non optional", so one has to decide between "mesh-batman-adv" and "mesh-babel"?

CodeFetch, 2018-07-03:
@lemoer I think you can build a virtual packet which is being selected by gluon-status-page-mesh-batman-adv and gluon-status-page-mesh-babel. It could be called gluon-status-page-provider-mesh. You can select it with DEPENDS:=+gluon-status-page-provider-mesh. 

gluon-status-page could then depend on while not selecting gluon-status-page-provider-mesh thus failing attempts to build the status page without one of those packages. E.g. DEPENDS:=gluon-status-page-provider-mesh (without the + sign).

Maybe @NeoRaider knows a more elegant solution. I think there was something like "PROVIDES", but I can't find information about that.

NeoRaider, 2018-07-03:
In theory, the status page should work fine without a protocol-specific provider, so I wouldn't add a dependency (at the moment, the nodeinfo request is only started when the either batman-adv or wireless has found a client on an interface - we should probably change that eventually).

rotanid, 2018-07-12:
please open a new issue or PR for the things discussed in the last 3 comments, as the actual problem of this issue is already solved.

closed 2018-07-12 by rotanid