freifunk-gluon/gluon #1457: Problem with update process

bschelm, 2018-07-08:
Hi all.
We have a problem where the update process does not properly finish (Freifunk Hannover, gluon-v2017.1)
What I can see in my SSH session when I run autoupdater is, that some processes are shut down and the updater is looking for the new firmware and tries to download it.
However, when the download does not succeed, the update process is aborted and the router tries to restart the services it has shut down earlier.
Mostly I have lost the connection already, so I cannot see what happens exactly, but my guess is, that some services don't start up properly and the router hangs in a state, where it routes some traffic, but is inaccessible otherwise.
In that state, I cannot SSH to the router any more and it does not show up in the map as online, but still shows connections to other routers and I can also connect to routers behind it (but only veeeeeery slowly).
The problem seems to be a massive load on the router and a reboot of the router mostly helps.
However, it's not always possible to restart the router because it is not physically accessible (only by SSH).
I have seen routers that have a load of 10 when the problem occurs, but have only around 0.1 load after reboot. So far I had 4 cases out of 20 (3x WR841 and 1x WR1043).
I don't know if this is really a problem that can be fixed in gluon because Hannover has a problem with massive router count and is just now in the process of splitting the net into smaller segments where the speed and load is much improved.
However I thought it'd be good to have this case public, if only for documentation and information.

Regards
Bernd

1977er, 2018-07-08:
bschelm refers to this firmware: https://github.com/freifunk-gluon/gluon/commit/d210cfcad9493797ed2cf35f65e9bd5f0a905c69
with this site conf: https://github.com/freifunkh/site/commit/b682f72041e25a71b696f055b10b053b89ec52ae

https://hannover.freifunk.net/wiki/Freifunk/FirmwareReleases#vH6

mweinelt, 2018-07-08:
Without serial access this sounds like a lot of guesswork I'm afraid. 

The referenced version uses the new (mid january) autoupdater already and I haven't had issues with that so far.

rotanid, 2018-07-08:
we also had no problems, so i agree with @mweinelt , without a serial log it's like searching for a needle in a haystack..

bschelm, 2018-07-08:
Like I said: more for documentation and information. :-)
I can also not reproduce it so even serial console would not help much.
Thanks anyway

NeoRaider, 2018-07-08:
The information you provided is inconsistent... @bschelm is talking about Gluon v2017.1, while @1977er cites a Gluon master commit. Overall, the issue sounds like #1258, which was fixed in v2017.1.8 and recent master versions.

closed 2018-07-08 by rotanid