freifunk-gluon/gluon #1463: nextnode changes: comment in Release nodes or accept non array string

A-Kasper, 2018-07-10:
I'm trying to build my first 2018.1.x images. I like the "changes to site.conf" part in the release nodes.
my build failed with:

`site.conf error: expected next_node.name to be an array, but it is "info.freifunk-bochum.de" (a string value)`

next_node.name now needs to be an array: {'name'} or {'name', 'anothername'}

I would like to suggest to add this in release nodes or if possible just to autoconvert the string to a single element array during build. Maybe this is also just a bug in site.conf check

rotanid, 2018-07-11:
the related change/commit: 9711afaf6980542f58383adbe24b58c678163fc6
did we miss this in the release notes or was it intentional @NeoRaider ?

NeoRaider, 2018-07-12:
We forgot to mention this in the release notes. I plan to fix this (and a few other omissions I noticed) on the weekend.

commit 3bf5e1c8 (2018-07-14): docs: releases/v2018.1: add note regarding next_node.name change

Fixes: #1463
commit 18aab98c (2018-07-14): docs: releases/v2018.1: add note regarding next_node.name change

Fixes: #1463
closed 2018-07-14 by NeoRaider