freifunk-gluon/gluon #1465: autoupdater: clearify feedback

A-Kasper, 2018-07-10:
When I tried the autoupdater I got:
Retrieving manifest from http://build.freifunk-bochum.de/stable/sysupgrade/stable.manifest 
autoupdater: warning: no matching firmware found (model tp-link-tl-wr1043n-nd-v3)

There was a firmware for this model. but no new firmware. 
I would suggest to clearify the feedback and give back "no matching new firmware found" or better: "no new firmware found for model tp-link-tl-wr1043n-nd-v3 with firmware $fwstring" instead of no matching firmware found.

rubo77, 2018-07-10:
:+1: 

I got the same misleading error when trying to downgrade a router from 2018.1.x to 2017.1.x, where I changed the release-info file, so the autoupdater downloaded the manifest, but stopped anyway (luckily) by rejecting a downgrade. (I don't know, how it notices, that this would be a downgrade, but i'm fine with this).

rotanid, 2018-07-11:
we have never supported downgrades in Gluon.

closed 2018-07-11 by rotanid

rubo77, 2018-07-11:
I think Kaspar has another issue (maybe you already had the latest firmware? Or your versioning didn't mark the newer firmware with a bigger number?

I think it is a good idea to make this message more verbose.

Please reopen this

NeoRaider, 2018-07-11:
The message for no new firmware is "No new firmware available." "warning: no matching firmware found (model tp-link-tl-wr1043n-nd-v3)" is printed when no matching line was found at all.

Note that the manifest format has changed (there is a new file size field), so the new autoupdater will not understand manifest generated by older versions of Gluon.

A-Kasper, 2018-07-11:
yes. it seems to be the new format. I installed it on a node and tested if the autoupdater is able to connect and worried if it worked correct when i saw the no image message. with a test against a master manifest everything worked fine. ==> can be closed