freifunk-gluon/gluon #1467: Status LED bei FW Update

klausdint, 2018-07-10:
lässt es sich vielleicht einrichten, dass wenn ein Firmware Update beim FF
Router durchgeführt wird (egal ob automatisch oder manuell durch
autoupdater -f oder über das Webinterface) dies durch das Blinken einer
Status LED wie der Power LED wie auch bei der Originalfirmware angezeigt
wird? So könnte verhindert werden, dass die Firmware gebricked wird, wenn
man das Gerät mal neu starten möchte, sich aber nicht sicher ist, ob nicht
gerade ein Firmware Update stattfindet. Man könnte dies dann direkt am
Gerät feststellen, ähnlich wie es z.B. auch beim Update der FritzBoxen der
Fall ist.

blocktrron, 2018-07-12:
@klausdint I've just tried sysupgrading an Archer C7 and it's status LED was blinking while flashing the image. Maybe you've tried a device which LED is statically powered and therefore can't be controlled by OpenWRT?

Adorfer, 2018-07-13:
Wenn man es rabulistich angehen wollte würde ich behaupten: Topic kann geschlossen weren, da zumindest bei >80% der Gluon-Router TP-Link-Geräte zum Einsatz kommen. Bei diesen blinkt während des Firmwareupdates auch bei Original-FW nicht die PowerLED. 
Die dahingehende Behauptung des Threadstarters ist nicht nachvollziehbar. 

Umgekehrt: Ja, irgendwelche Status-LEDs blinken zu lassen ist gut. Das Softbrick-Risiko ist bei aktuellen Firmwares aber geringt, da z.B. mit zwei Images (UBNT) gearbeitet wird oder aber heute schon während des Upgrades die Online-LED rasch blinkt "in der kritischen Phase" (siehe Post von blocktrron, aber beim 841er ist das ebenfalls so.)

NeoRaider, 2018-07-13:
In OpenWrt, this is currently solved on a per-target basis (ar71xx does not have any upgrade indication at the moment AFAICT from the code).

I generally don't care about what the devices do with stock firmware - I would prefer to have consistent behaviour on all devices with OpenWrt/Gluon (as far as we have a status LED that can be controlled). In any case, this must be decided and solved in OpenWrt.

klausdint, 2019-02-11:
Done: https://bugs.openwrt.org/index.php?do=details&task_id=2120

blocktrron, 2019-02-11:
@klausdint This is no upstream issue as all targets in OpenWRT support update indication except for ar71xx. As ar71xx in the meantime became legacy, don't expect a fix there.

For the next time: upstream issues are best created in english (just like gluon issues)

rotanid, 2019-02-12:
@blocktrron and for us, is there anything that can be improved or has to be done to support it like upstream?

mweinelt, 2019-02-12:
@rotanid As we just make use of the upstream sysupgrade utility I would think it is completely within OpenWrt's purview.

blocktrron, 2019-02-12:
@rotanid Maybe my statement was a bit misunderstanding - it is an upstream issue BUT as ar71xx ist considered legacy, i'm not sure if we should add a patch for this minor feature/bugfix (not sure what it is)

Anyway - this should be everything there is needed to let the status LED blink on a running update. It's not tested but maybe @klausdint wants to test and give feedback? https://github.com/blocktrron/openwrt/commit/01f12589873412ac3e9231a0d6eb6cf7897518c2

klausdint, 2019-02-13:
> blocktrron/openwrt@01f1258

Hello, I've tried it with a TP Link WR841n and it didn't worked there

klausdint, 2019-02-13:
Also it was not working with a TP Link WDR 3600 v1. The WR841n were v9 and
v8

CodeFetch, 2020-02-08:
Does it work with ath79?

blocktrron, 2020-02-08:
Was implemented upstream with https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=9e1e4324690ecf3874acfab76566e856f133bf6f

If anybody wants this implemented in 2020.1, a backport PR can be opened. That said, it works on OF-platforms since 2018.2.

closed 2020-02-08 by blocktrron