freifunk-gluon/gluon #1477: Check site earlier in make process

rubo77, 2018-07-14:
At the moment, you will be confronted of syntax errors in your site.conf after about 30 minutes build time.

It would save you much hassle, if the check_site script would run once at the start of the build process.

NeoRaider, 2018-07-14:
Syntax errors will be detected very early (after the OpenWrt tools and host Lua have been built).

This is not the case for other (non-syntax) errors: We can't run check_site earlier, as the site checks depend on the package selection. The exact package selection is handled by the OpenWrt build process (and may even differ per-model!), Gluon only seeds this build process by generating `.config` etc.

Note this is only a problem for the initial build. Incremental builds take about ~3 minutes, so you get feedback reasonable fast.

closed 2018-07-14 by NeoRaider

rubo77, 2019-12-30:
related: #1901