freifunk-gluon/gluon #1481: Archer C2600 – Mesh-on-LAN interface unavailable

jotzt, 2018-07-15:
On my Archer C2600 v1.1, the Mesh-on-LAN interface is not available after installation (gluon master 2018-07-13).

`/etc/config/network`
```
config interface 'mesh_lan'
        option ifname 'eth1'
        option mesh 'bat0'
        option igmp_snooping '0'
        option transitive '1'
        option index '4'
        option macaddr '12:64:92:06:91:3c'
        option proto 'gluon_wired'
        option disabled '0'
```
The interface is coming up after adding the following lines (copied from a WR1043 config)

```
network.mesh_lan.type='bridge'
network.mesh_lan.auto='1'
```

NeoRaider, 2018-07-15:
We do not make mesh_lan a bridge anymore since Gluon v2018.1; *auto* defaults to 1. (I also see the obsolete option `mesh 'bat0'`, which isn't used anymore since Gluon v2016.2.x; it seems our upgrade script do not clean this up completely...).

Is it possible that your eth1 interface appears in any other interface definitions like 'client'? I can't think of any other reason why the interface wouldn't come up.

jotzt, 2018-07-15:
> I also see the obsolete option mesh 'bat0', which isn't used anymore since Gluon v2016.2.x; 
... and had been pasted into the issue by accident, see the edited version. Sorry for that.

I figured out that this is not an issue but a feature as I was used to connect my laptop to a MoL port before and and connecting to the router via its link local address of the MoL-port.

NeoRaider, 2018-07-15:
That didn't really answer my question, and you changed the interface name in the example... Is eth1.1 used by any other interface definition?

jotzt, 2018-07-15:
As far as I can recall, the situation was as follows:

* first firmware installation with Gluon V2018.1 -> eth1
* `sysupgrade -n <gluon-master>` -> eth1
* `firstboot` -> eth1.1

There was no manual interface configuration except the brigde/auto-setting

NeoRaider, 2018-07-15:
How did you enable Mesh-on-LAN? If you did so via the command line, did you also disable the LAN port in the 'client' interface?

`firstboot` and `sysupgrade -n` should always yield the same configuration; I assume you did not upgrade to master with `-n`, so the old (untagged) switch configuration was retained (which is fine).

jotzt, 2018-07-15:
Mesh-on-LAN was enabled by default. `site.config`: `mesh_on_lan = true,`

NeoRaider, 2018-07-15:
I just read your edit. Everything seems in order, multicast pings might well be filtered by the firewall.

closed 2018-07-15 by NeoRaider