freifunk-gluon/gluon #1493: gluon-config-mode-geo-location: difference between code and docs

rotanid, 2018-07-25:
in the lua code which was added before Gluon 2016.2 the altitude field is optional and it is shown if there is no explicit setting to hide it.
the expanded site docs, which were added in march ( https://github.com/freifunk-gluon/gluon/commit/486c2e482184cbe21f8459d29653225f24f84f6c#diff-0d44e6a3fa2aba2dc4a79afbfcc4ab1f ) state:
`By default, no altitude fields are shown by the *gluon-config-mode-geo-location* package.`
which is the opposite of the code.

so either the docs or the code have to be changed?

NeoRaider, 2018-08-14:
Hmm, indeed, the documentation doesn't match the current behaviour. I think that the altitude field is not really useful for most communities, so I would be fine with either fix (changing the code or the documenatation).

I propose one more change to the behaviour: at the moment, the fields will be shown if there is a value even when it is disabled in site.conf. This seems weird to me; we should 1) always follow the site.conf setting and 2) delete the altitude value from UCI when a form without altitude field is saved.

rubo77, 2018-08-14:
I could only think of this Szenario:

In a multi domain setup there could be one
domain with and one without altitude. And with a hood selector the domain could change in a node back and for so the altitude could overdue a time in the domain without altitude.

NeoRaider, 2018-08-15:
@rubo77 *show_altitude* is not allowed in domain configs. (The multidomain feature exists for a single purpose: to create multiple *completely identical* router configurations that simply don't mesh with each other. All site fields that do not match this usecase are disallowed in domain configs.)

rubo77, 2018-08-16:
Then, I think there is nothing in the way to delete an eventually stored altitude field with uci.

commit b4346ff9 (2018-08-19): gluon-config-mode-geo-location: make altitude show behaviour match docs

Do not show by default, delete old altitude value when a form without
altitude shown is saved.

Fixes #1493
closed 2018-08-19 by NeoRaider