freifunk-gluon/gluon #1495: kmod-ath10k-ct spamminess

mweinelt, 2018-07-26:
For some reason we are using the -ct driver for ath10k in combination with a non-ct firmware. The logs below seem to result from ct modifications. Decoding them seems to be proprietary(?), as no tooling but a service is offered here: https://lists.openwrt.org/pipermail/openwrt-devel/2017-June/007877.html.

We should test if we can migrate to the stock kmod-ath10k driver instead.

```
[  128.680532] ath10k_pci 0000:01:00.0: htt peer map vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[  129.416356] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  129.416383] ath10k: [0000]: 311A0200 1B4CFC07 84584300 311A0200 0128FC17 01000000 04F00F00 03900000
[  129.416396] ath10k: [0008]: 00000000 FF000000
[  129.416408] ath10k_pci 0000:01:00.0: ATH10K_END
[  159.595208] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[  160.418733] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  160.418759] ath10k: [0000]: CE950200 025C000C 28BAB5F0 A02A0000 01000000
[  160.418772] ath10k_pci 0000:01:00.0: ATH10K_END
[  631.853770] ath10k_pci 0000:01:00.0: htt peer map vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[  632.462310] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  632.462338] ath10k: [0000]: D7F60900 1B4CFC07 14594300 D7F60900 0128FC17 01000000 04F00F00 03900000
[  632.462350] ath10k: [0008]: 00000000 FF000000
[  632.462362] ath10k_pci 0000:01:00.0: ATH10K_END
[  662.816022] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[  663.464926] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[  663.464953] ath10k: [0000]: A5720A00 025C000C 28BAB5F0 A02A0000 01000000
[  663.464965] ath10k_pci 0000:01:00.0: ATH10K_END
[ 2586.651578] ath10k_pci 0000:01:00.0: htt peer map vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[ 2587.652642] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 2587.652671] ath10k: [0000]: D7812800 1B4CFC07 A4594300 D7812800 0128FC17 01000000 04F00F00 03900000
[ 2587.652683] ath10k: [0008]: 00000000 FF000000
[ 2587.652696] ath10k_pci 0000:01:00.0: ATH10K_END
[ 2617.611533] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[ 2617.655181] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 2617.655207] ath10k: [0000]: A2FD2800 025C000C 28BAB5F0 A02A0000 01000000
[ 2617.655220] ath10k_pci 0000:01:00.0: ATH10K_END
[ 8876.817256] ath10k_pci 0000:01:00.0: htt peer map vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[ 8877.249106] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 8877.249135] ath10k: [0000]: E1C98A00 1B4CFC07 345A4300 E1C98A00 0128FC17 01000000 04F00F00 03900000
[ 8877.249148] ath10k: [0008]: 00000000 FF000000
[ 8877.249160] ath10k_pci 0000:01:00.0: ATH10K_END
[ 8907.793879] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[ 8908.252012] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[ 8908.252038] ath10k: [0000]: BD458B00 025C000C 28BAB5F0 A02A0000 01000000
[ 8908.252051] ath10k_pci 0000:01:00.0: ATH10K_END
[10408.434823] ath10k_pci 0000:01:00.0: htt peer map vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[10409.396064] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[10409.396092] ath10k: [0000]: 32B8A200 1B4CFC07 C45A4300 32B8A200 0128FC17 01000000 04F00F00 03900000
[10409.396104] ath10k: [0008]: 00000000 FF000000
[10409.396117] ath10k_pci 0000:01:00.0: ATH10K_END
[10687.838745] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[10688.422707] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[10688.422734] ath10k: [0000]: BE15A700 025C000C 28BAB5F0 A02A0000 01000000
[10688.422747] ath10k_pci 0000:01:00.0: ATH10K_END
[12966.370454] ath10k_pci 0000:01:00.0: htt peer map vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[12966.638752] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[12966.638781] ath10k: [0000]: AFAFCA00 1B4CFC07 545B4300 AFAFCA00 0128FC17 01000000 04F00F00 03900000
[12966.638793] ath10k: [0008]: 00000000 FF000000
[12966.638806] ath10k_pci 0000:01:00.0: ATH10K_END
[13002.041789] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[13002.641861] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[13002.641887] ath10k: [0000]: 533ECB00 025C000C 28BAB5F0 A02A0000 01000000
[13002.641900] ath10k_pci 0000:01:00.0: ATH10K_END
[65362.924968] ath10k_pci 0000:01:00.0: htt peer map vdev 0 peer 6c:19:c0:0d:c5:84 id 130
[65363.630064] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[65363.630093] ath10k: [0000]: 5A5DFD03 1B4CFC07 E45B4300 5A5DFD03 0128FC17 03000000 04F00F00 03100100
[65363.630107] ath10k: [0008]: 00000000 FFFF0000
[65363.630119] ath10k_pci 0000:01:00.0: ATH10K_END
[65927.945189] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 0 peer 6c:19:c0:0d:c5:84 id 130
[65928.683967] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[65928.683996] ath10k: [0000]: 58310604 025C000C 6C19C00D C5840000 01000000 58310604 324CFC07 E45B4300
[65928.684010] ath10k_pci 0000:01:00.0: ATH10K_END
[65978.620092] ath10k_pci 0000:01:00.0: htt peer map vdev 0 peer 6c:19:c0:0d:c5:84 id 130
[65978.687898] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[... 896 lines cut ...]
[313937.263255] ath10k: [0008]: B0442903 284CFC07 40000000 B7442903 234CFC0F 01000000 01000000 01000000
[313937.263268] ath10k_pci 0000:01:00.0: ATH10K_END
[313945.261836] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[313945.261866] ath10k: [0000]: 52642903 284CFC07 40000000 5A642903 234CFC0F 01000000 01000000 01000000
[313945.261884] ath10k: [0008]: BA652903 284CFC07 40000000 C2652903 234CFC0F 01000000 01000000 01000000
[313945.261901] ath10k: [0016]: E2662903 284CFC07 40000000 EA662903 234CFC0F 01000000 01000000 01000000
[313945.261914] ath10k_pci 0000:01:00.0: ATH10K_END
[313946.339771] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 0 peer 8c:f5:a3:60:0b:ae id 75
[313947.262003] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[313947.262034] ath10k: [0000]: 5D6B2903 144CFC07 40010000 5F6B2903 025C000C 8CF5A360 0BAE0000 01000000
[313947.262048] ath10k: [0008]: 5F6B2903 324CFC07 B4704300
[313947.262061] ath10k_pci 0000:01:00.0: ATH10K_END
[316024.718343] ath10k_pci 0000:01:00.0: htt peer map vdev 0 peer 70:e0:27:5c:01:eb id 213
[316328.722228] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 0 peer 70:e0:27:5c:01:eb id 213
[316329.469169] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[316329.469199] ath10k: [0000]: D2A44E03 025C000C 70E0275C 01EB0000 01000000 D2A44E03 324CFC07 44714300
[316329.469213] ath10k_pci 0000:01:00.0: ATH10K_END
[316645.497962] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[316645.497991] ath10k: [0000]: 33955303 244CFC17 24704300 05000000 00000000 40000000 01000000
[316645.498006] ath10k_pci 0000:01:00.0: ATH10K_END
[318672.244706] ath10k_pci 0000:01:00.0: removing peer, htt peer unmap vdev 0 peer 28:ba:b5:f0:a0:2a id 21
[318672.692049] ath10k_pci 0000:01:00.0: ath10k_pci ATH10K_DBG_BUFFER:
[318672.692076] ath10k: [0000]: C7427303 025C000C 28BAB5F0 A02A0000 01000000
[318672.692090] ath10k_pci 0000:01:00.0: ATH10K_END
```

blocktrron, 2018-07-26:
We replaced kmod-ath10k-ct with kmod-ath10k which made the spamminess disappear. 

It also reduced memory in-use from around 50% to 25% on an Archer C7 v2. 11s is still working.

rotanid, 2018-07-26:
nice, sounds almost too good to be true - and doesn't necessarily have to work for every ath10k device

blocktrron, 2018-07-26:
@rotanid If it works it should work on every dualband AC device with a QCA9880. QCA9886 devices (Archer C58-C60) already use kmod-ath10k. 

Some statistics:
[kmod-ath10k-ct](https://stats.darmstadt.freifunk.net/d/000000021/router-meshviewer-export?var-node=c4e984f60b72&refresh=1m&panelId=3&fullscreen&orgId=1&from=1532478291340&to=1532635714088)

[kmod-ath10k](https://stats.darmstadt.freifunk.net/d/000000021/router-meshviewer-export?var-node=60e3274ff06b&refresh=1m&panelId=3&fullscreen&orgId=1&from=1532628176935&to=1532635636510)

rotanid, 2018-07-31:
@mweinelt @blocktrron i'd say we apply this change in master, we still have enough time to revert it as the next release won't happen tomorrow.
please open a pull request, thank you!

(btw. as ecsv asked on over at #1440: are your tests based on gluon master / openwrt?)

Casey1979, 2018-07-31:
So sieht es bei einem Archer C2600 nach ein paar Tagen aus, immernoch alles sauber und aufgeräumt, allerdings sowohl mit ct Treiber als auch ct Firmware.
Man beachte insbesondere diese Zeile, die wohl der OpenWRT 18.06 Upstream mitgebracht hat:
```
[   30.381484] ath10k_pci 0001:01:00.0: NOTE:  Firmware DBGLOG output disabled in debug_mask: 0x10000000
```

```
root@TL-ARCHER-C2600:~# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 4.14.54 (x@X) (gcc version 7.3.0 (OpenWrt GCC 7.3.0 r7140+7-a7b2764879)) #0 SMP Thu Apr 12 05:50:02 2018
[    0.000000] CPU: ARMv7 Processor [512f04d0] revision 0 (ARMv7), cr=10c5787d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, PIPT instruction cache
[    0.000000] OF: fdt: Machine model: TP-Link Archer C2600
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] On node 0 totalpages: 122880
[    0.000000] free_area_init_node: node 0, pgdat c0b27b00, node_mem_map ddc39000
[    0.000000]   Normal zone: 960 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 122880 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x88/0x40c with crng_init=0
[    0.000000] percpu: Embedded 15 pages/cpu @ddc01000 s29324 r8192 d23924 u61440
[    0.000000] pcpu-alloc: s29324 r8192 d23924 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121920
[    0.000000] Kernel command line:
[    0.000000] Bootloader command line (ignored): console=ttyHSL1,115200n8 root=mtd:rootfs rootfstype=squashfs
[    0.000000] PID hash table entries: 2048 (order: 1, 8192 bytes)
[    0.000000] Dentry cache hash table entries: 65536 (order: 6, 262144 bytes)
[    0.000000] Inode-cache hash table entries: 32768 (order: 5, 131072 bytes)
[    0.000000] Memory: 478384K/491520K available (4832K kernel code, 161K rwdata, 1280K rodata, 1024K init, 250K bss, 13136K reserved, 0K cma-reserved, 0K highmem)
[    0.000000] Virtual kernel memory layout:
[    0.000000]     vector  : 0xffff0000 - 0xffff1000   (   4 kB)
[    0.000000]     fixmap  : 0xffc00000 - 0xfff00000   (3072 kB)
[    0.000000]     vmalloc : 0xde800000 - 0xff800000   ( 528 MB)
[    0.000000]     lowmem  : 0xc0000000 - 0xde000000   ( 480 MB)
[    0.000000]     pkmap   : 0xbfe00000 - 0xc0000000   (   2 MB)
[    0.000000]     modules : 0xbf000000 - 0xbfe00000   (  14 MB)
[    0.000000]       .text : 0xc0208000 - 0xc07b80f0   (5825 kB)
[    0.000000]       .init : 0xc0a00000 - 0xc0b00000   (1024 kB)
[    0.000000]       .data : 0xc0b00000 - 0xc0b28400   ( 161 kB)
[    0.000000]        .bss : 0xc0b2a000 - 0xc0b68aec   ( 251 kB)
[    0.000000] SLUB: HWalign=64, Order=0-3, MinObjects=0, CPUs=2, Nodes=1
[    0.000000] Hierarchical RCU implementation.
[    0.000000] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[    0.000000] clocksource: dg_timer: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 305801671480 ns
[    0.000008] sched_clock: 32 bits at 6MHz, resolution 160ns, wraps every 343597383600ns
[    0.000020] Switching to timer-based delay loop, resolution 160ns
[    0.000237] Calibrating delay loop (skipped), value calculated using timer frequency.. 12.50 BogoMIPS (lpj=62500)
[    0.000258] pid_max: default: 32768 minimum: 301
[    0.000379] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000396] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes)
[    0.000960] CPU: Testing write buffer coherency: ok
[    0.001689] Setting up static identity map for 0x42300000 - 0x42300060
[    0.001841] Hierarchical SRCU implementation.
[    0.002559] smp: Bringing up secondary CPUs ...
[    0.004332] smp: Brought up 1 node, 2 CPUs
[    0.004345] SMP: Total of 2 processors activated (25.00 BogoMIPS).
[    0.004356] CPU: All CPU(s) started in SVC mode.
[    0.015261] VFP support v0.3: implementor 51 architecture 64 part 4d variant 2 rev 0
[    0.015419] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.015444] futex hash table entries: 512 (order: 3, 32768 bytes)
[    0.015559] pinctrl core: initialized pinctrl subsystem
[    0.016540] NET: Registered protocol family 16
[    0.016779] DMA: preallocated 256 KiB pool for atomic coherent allocations
[    0.018115] cpuidle: using governor ladder
[    0.018179] cpuidle: using governor menu
[    0.038838] msm_bus_fabric_init_driver
[    0.040230] usbcore: registered new interface driver usbfs
[    0.040318] usbcore: registered new interface driver hub
[... 409 lines cut ...]
[   31.083842] ath10k_pci 0000:01:00.0: 10.4 wmi init: vdevs: 16  peers: 48  tid: 96
[   31.083869] ath10k_pci 0000:01:00.0: msdu-desc: 2500  skid: 32
[   31.159244] ath10k_pci 0000:01:00.0: wmi print 'P 48/48 V 16 K 144 PH 176 T 186  msdu-desc: 2500  sw-crypt: 0'
[   31.159905] ath10k_pci 0000:01:00.0: wmi print 'free: 33100 iram: 16188 sram: 16448'
[   31.428405] IPv6: ADDRCONF(NETDEV_UP): mesh0: link is not ready
[   32.561055] IPv6: ADDRCONF(NETDEV_CHANGE): mesh1: link becomes ready
[   34.210090] IPv6: ADDRCONF(NETDEV_CHANGE): mesh0: link becomes ready
[   34.407246] batman_adv: bat0: Adding interface: mesh1
[   34.407275] batman_adv: bat0: Interface activated: mesh1
[   34.615817] batman_adv: bat0: Adding interface: mesh0
[   34.615846] batman_adv: bat0: Interface activated: mesh0
[... cut ...]
[   36.692781] batman_adv: bat0: MLD Querier appeared
[   36.954040] NOHZ: local_softirq_pending 08
[  118.873910] NOHZ: local_softirq_pending 08
[  139.361140] NOHZ: local_softirq_pending 08
[  200.794664] NOHZ: local_softirq_pending 08
[  323.676150] NOHZ: local_softirq_pending 08
[  344.157950] NOHZ: local_softirq_pending 08
[  405.598623] NOHZ: local_softirq_pending 08
[  548.960243] NOHZ: local_softirq_pending 08
[  630.881595] NOHZ: local_softirq_pending 08
[  692.322440] NOHZ: local_softirq_pending 08
root@TL-ARCHER-C2600:~#
```

closed 2018-08-01 by rotanid

greearb, 2018-08-01:
Is the memory use just related to the ath10k-ct logging, which should be disabled by default in recent OpenWRT?

mweinelt, 2018-08-03:
I personally don't own any ath10k devices, sorry.

PR #1499 (2018-08-01): ar71xx-generic: prefer kmod-ath10k for 802.11s mesh
- removes cryptic debug messages from kernel ring buffer and
- reduces memory usage by as much as 20-25 MiB 

resolves #1495