freifunk-gluon/gluon #1501: add brcm47xx support

Loader009, 2018-08-03:
Hey there,
I own a spare Netgear WNR3500L, which I would like to flash gluon on it.
But sadly there is no brcm47xx support.

Can you add support for it?
Alternatively (if there is nearly nothing to change), I can test it first locally and report if anything is wrong.

OpenWRT link:
http://archive.openwrt.org/releases/17.01.4/targets/brcm47xx/mips74k/

mweinelt, 2018-08-03:
It is unlikely that BCM radios will support 802.11s. Can you try and flash OpenWrt 18.06 and dump information like `iw phy`? Also it would be relevant to know the specific hardware revision.

Loader009, 2018-08-03:
I have the first revision, WNR3500L v1.
I will get dmesg and iw phy tomorrow.

OpenWRT wiki:
https://openwrt.org/toh/netgear/wnr3500l

PS: Didn't know that 18.06 is out, I am at LEDE 17.01.4 with my other router.

Loader009, 2018-08-04:
Here is the data:
[dmesg.log](https://github.com/freifunk-gluon/gluon/files/2259354/dmesg.log)
[iw phy.log](https://github.com/freifunk-gluon/gluon/files/2259355/iw.phy.log)

```
	Supported interface modes:
		 * IBSS
		 * managed
		 * AP
		 * AP/VLAN
		 * monitor
		 * mesh point
```

I think the router is qualified for mesh networking?

mweinelt, 2018-08-04:
> 	interface combinations are not supported

This means it can't do both AP and Mesh at the same time.

closed 2018-08-04 by rotanid