freifunk-gluon/gluon #1505: C2600: client wifi does not work properly - mark ipq806x as broken

Noodlesalat, 2018-08-07:
Hi,
I have some problems with the client wifi connection on the TP-Link C2600. When I try to connect to the node it disconnects me a few times until the connection is established.

Here is a part of `logread` while connecting:
```
Tue Aug  7 23:55:10 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: authenticated
Tue Aug  7 23:55:10 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: associated (aid 1)
Tue Aug  7 23:55:10 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MACADDRESS
Tue Aug  7 23:55:12 2018 daemon.notice hostapd: client0: AP-STA-DISCONNECTED MACADDRESS
Tue Aug  7 23:55:12 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: disassociated
Tue Aug  7 23:55:13 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Aug  7 23:55:20 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: authenticated
Tue Aug  7 23:55:20 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: associated (aid 1)
Tue Aug  7 23:55:20 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MACADDRESS
Tue Aug  7 23:55:21 2018 daemon.notice netifd: client (1739): cat: write error: Broken pipe
Tue Aug  7 23:55:22 2018 daemon.notice hostapd: client0: AP-STA-DISCONNECTED MACADDRESS
Tue Aug  7 23:55:22 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: disassociated
Tue Aug  7 23:55:23 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Aug  7 23:55:25 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: authenticated
Tue Aug  7 23:55:25 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: associated (aid 1)
Tue Aug  7 23:55:25 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MACADDRESS
Tue Aug  7 23:55:27 2018 daemon.notice hostapd: client0: AP-STA-DISCONNECTED MACADDRESS
Tue Aug  7 23:55:27 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: disassociated
Tue Aug  7 23:55:28 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: deauthenticated due to inactivity (timer DEAUTH/REMOVE)
Tue Aug  7 23:55:29 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: authenticated
Tue Aug  7 23:55:29 2018 daemon.info hostapd: client0: STA MACADDRESS IEEE 802.11: associated (aid 1)
Tue Aug  7 23:55:29 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MACADDRESS
```

I also tested firmware without the ct-packages:

- **with** the ct-packages the client wifi does **not** work, but the wifi mesh works great
- **without** the ct-packages the client wifi works, but the mesh network does **not** appear

rotanid, 2018-08-07:
- which Gluon version (or if it's not a released version: which commit?) are you testing? have you tested latest commits?
- which meshing standard are you using? ibss or 802.11s - and have you tried the other standard?

Noodlesalat, 2018-08-07:
> which Gluon version (or if it's not a released version: which commit?) are you testing? have you tested latest commits?

[d87c4b521b2e891155241c01b98a7ac90a8883b9](https://github.com/freifunk-gluon/gluon/commit/d87c4b521b2e891155241c01b98a7ac90a8883b9). I also tried the gluon master branch around yesterday.

> which meshing standard are you using? ibss or 802.11s - and have you tried the other standard?

Currently 802.11s and I have not tested ibss.

rotanid, 2018-08-07:
actually we're more interested in the case "without the ct-packages the client wifi works, but the mesh network does not appear" as on other devices, the -ct packages are only necessary for ibss meshing.
could you try again with the kmod without ct and the firmware without ct and check if the mesh interface is there on the device itself and post the complete dmesg from the boot process?

Noodlesalat, 2018-08-07:
ct
=
**logread**
```
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.015423] futex hash table entries: 512 (order: 3, 32768 bytes)
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.015536] pinctrl core: initialized pinctrl subsystem
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.016519] NET: Registered protocol family 16
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.016759] DMA: preallocated 256 KiB pool for atomic coherent allocations
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.018096] cpuidle: using governor ladder
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.018161] cpuidle: using governor menu
Thu Apr 12 07:59:18 2018 kern.err kernel: [    0.038818] msm_bus_fabric_init_driver
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.040215] usbcore: registered new interface driver usbfs
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.040303] usbcore: registered new interface driver hub
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.040400] usbcore: registered new device driver usb
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.040456] pps_core: LinuxPPS API ver. 1 registered
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.040468] pps_core: Software ver. 5.3.6 - Copyright 2005-2007 Rodolfo Giometti <giometti@linux.it>
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.040500] PTP clock support registered
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.042360] clocksource: Switched to clocksource dg_timer
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.045607] NET: Registered protocol family 2
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.046199] TCP established hash table entries: 4096 (order: 2, 16384 bytes)
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.046241] TCP bind hash table entries: 4096 (order: 3, 32768 bytes)
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.046294] TCP: Hash tables configured (established 4096 bind 4096)
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.046383] UDP hash table entries: 256 (order: 1, 8192 bytes)
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.046409] UDP-Lite hash table entries: 256 (order: 1, 8192 bytes)
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.046563] NET: Registered protocol family 1
Thu Apr 12 07:59:18 2018 kern.debug kernel: [    0.046610] PCI: CLS 0 bytes, default 64
Thu Apr 12 07:59:18 2018 kern.warn kernel: [    0.047725] No memory allocated for crashlog
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.047959] workingset: timestamp_bits=30 max_order=17 bucket_order=0
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.053052] squashfs: version 4.0 (2009/01/31) Phillip Lougher
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.053069] jffs2: version 2.2 (NAND) (SUMMARY) (LZMA) (RTIME) (CMODE_PRIORITY) (c) 2001-2006 Red Hat, Inc.
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.061295] io scheduler noop registered
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.061312] io scheduler deadline registered (default)
Thu Apr 12 07:59:18 2018 kern.warn kernel: [    0.062975] qcom-pcie 1b500000.pci: 1b500000.pci supply vdda not found, using dummy regulator
Thu Apr 12 07:59:18 2018 kern.warn kernel: [    0.063094] qcom-pcie 1b500000.pci: 1b500000.pci supply vdda_phy not found, using dummy regulator
Thu Apr 12 07:59:18 2018 kern.warn kernel: [    0.063187] qcom-pcie 1b500000.pci: 1b500000.pci supply vdda_refclk not found, using dummy regulator
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.064014] OF: PCI: host bridge /soc/pci@1b500000 ranges:
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.064056] OF: PCI:    IO 0x0fe00000..0x0fefffff -> 0x0fe00000
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.064081] OF: PCI:   MEM 0x08000000..0x0fdfffff -> 0x08000000
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.163076] qcom-pcie 1b500000.pci: link up
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.163252] qcom-pcie 1b500000.pci: PCI host bridge to bus 0000:00
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.163274] pci_bus 0000:00: root bus resource [bus 00-ff]
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.163293] pci_bus 0000:00: root bus resource [io  0x0000-0xfffff] (bus address [0xfe00000-0xfefffff])
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.163308] pci_bus 0000:00: root bus resource [mem 0x08000000-0x0fdfffff]
Thu Apr 12 07:59:18 2018 kern.debug kernel: [    0.163324] pci_bus 0000:00: scanning bus
Thu Apr 12 07:59:18 2018 kern.debug kernel: [    0.163372] pci 0000:00:00.0: [17cb:0101] type 01 class 0x060400
Thu Apr 12 07:59:18 2018 kern.debug kernel: [    0.163469] pci 0000:00:00.0: calling pci_fixup_ide_bases+0x0/0x44
Thu Apr 12 07:59:18 2018 kern.debug kernel: [    0.163541] pci 0000:00:00.0: supports D1
Thu Apr 12 07:59:18 2018 kern.debug kernel: [    0.163552] pci 0000:00:00.0: PME# supported from D0 D1 D3hot
Thu Apr 12 07:59:18 2018 kern.debug kernel: [    0.163567] pci 0000:00:00.0: PME# disabled
Thu Apr 12 07:59:18 2018 kern.debug kernel: [    0.163823] pci_bus 0000:00: fixups for bus
Thu Apr 12 07:59:18 2018 kern.info kernel: [    0.163847] PCI: bus0: Fast back to back transfers disabled
Thu Apr 12 07:59:18 2018 kern.debug kernel: [    0.163865] pci 0000:00:00.0: scanning [bus 01-ff] behind bridge, pass 0
Thu Apr 12 07:59:18 2018 kern.debug kernel: [    0.163986] pci_bus 0000:01: scanning bus
Thu Apr 12 07:59:18 2018 kern.debug kernel: [    0.164162] pci 0000:01:00.0: [168c:0040] type 00 class 0x028000
[... cut ...]
[    0.000000]   Normal zone: 960 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 122880 pages, LIFO batch:31
[    0.000000] random: get_random_bytes called from start_kernel+0x88/0x40c with crng_init=0
[    0.000000] percpu: Embedded 15 pages/cpu @ddc01000 s29324 r8192 d23924 u61440
[    0.000000] pcpu-alloc: s29324 r8192 d23924 u61440 alloc=15*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 121920
[

Noodlesalat, 2018-08-07:
This seems interesting to me:
```
Thu Apr 12 08:00:16 2018 kern.warn kernel: [ 279.632599] ath10k_pci 0001:01:00.0: must load driver with rawmode=1 to add mesh interfaces
```

rotanid, 2018-08-07:
it does. but at this point, i'm out of ideas, i did my part to get the relevant information into this issue...

NeoRaider, 2018-08-07:
That's an indicator that the ath10k driver/firmware do not support 11s. Moving from the -ct driver/firmware to the QCA version should help.

Casey1979, 2018-08-08:
I did some test again too but with Mesh and Privatewifi instead of Clientwifi because the Networks are still under attack which required MFP and Encryption.
Clientnetwork alone works with every combination of Firmware and Driver.
Meshnetwork alone works with either Driver but needs the CT-Firmware because the QCA 9980 Firmware has still no 11s Support as it seems.
Mesh+Privatewifi together works with either Driver but has Problems with the Clients connecting unreliable while Mesh still works.
The cause is still unknown, i tried to test with old Ledefirmwares but Openwrt doesn't accept the old ones, i didn't force it though.
It may be either the CT-Firmware or a Problem in OpenWRT, maybe it came with 18.06 because i think it wouldn't have been gone unnoticed if this Bug existed during the testing with the Lede based Firmwares.
Btw. VHT Support without Client or Private Network doesn't seem to work either, probably never worked.

rotanid, 2018-08-08:
@Casey1979 you reported in May, that everything was working. https://github.com/freifunk-gluon/gluon/issues/1088#issuecomment-386901948
can one of you try again with a firmware from back then to double check this?
if it way working before, we have a good chance to fix it.

rotanid, 2018-08-20:
in that case, we should mark the device/target as broken again, yes....

rotanid, 2018-08-25:
@kpanic23 on IRC you reported that you don't have any problems with the c2600 - have you tried a rather new build, if not, which commit is your current build based on?

Noodlesalat, 2018-08-25:
I am already testing kpanics build which I downloaded from [ftp://ftp.ff3l.net/Archer-C2600/](ftp://ftp.ff3l.net/Archer-C2600/). I am going to publish the results when the tests are finished.

Noodlesalat, 2018-08-26:
I still encounter problems with @kpanic23's Firmware. The initial connection works better now. The connection is established right at the first try.
But my Phone is reconnecting every few minutes/seconds.

Here the outputs of `logread`:
### 2,4 GHz
```
### 5 GHz
```
Sun Aug 26 22:21:33 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:21:33 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:21:33 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:22:18 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:22:18 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:22:18 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:22:30 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:22:30 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:22:30 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:23:54 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:23:54 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:23:54 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:25:31 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:25:31 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:25:31 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:26:11 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:26:11 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:26:11 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:26:58 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:26:58 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:26:58 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:27:14 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:27:14 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:27:14 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:27:57 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:27:57 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:27:57 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:28:32 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:28:32 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:28:32 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:29:17 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:29:17 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:29:17 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:29:47 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:29:47 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:29:47 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:31:02 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:31:02 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:31:02 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
Sun Aug 26 22:31:49 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: authenticated
Sun Aug 26 22:31:49 2018 daemon.info hostapd: client0: STA MAC-ADDRESS IEEE 802.11: associated (aid 1)
Sun Aug 26 22:31:49 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED MAC-ADDRESS
```

Noodlesalat, 2018-08-29:
Now I can reproduce the issue with three different devices running Android. The strange thing is, that it now behaves like I decribed in my frist post in this issue. Possibly this is affected by the devices I use?!
Furthermore meshing is still untested with @kpanic23 's firmware.

mweinelt, 2018-08-29:
As you said that you have no problems with the current OpenWrt master maybe there is firmware we can backport to resolve these issues.

blocktrron, 2018-08-29:
@mweinelt judging from kalles repository (https://github.com/kvalo/ath10k-firmware) we are already running the latest firmware (which is 3 years old).

rotanid, 2018-09-03:
target was marked as broken with commit dddd2cd01144407b04cb0a6d471c1381e7295545

closed 2018-09-03 by rotanid

BIGFAAT, 2019-09-26:
802.11s (even encrypted) together with AP mode on 5ghz just work fine on OpenWrt SNAPSHOT r11111 for me. Maybe you can reconsider the support of this device in Gluon in the future?

rotanid, 2019-09-26:
@BIGFAAT wait until #1791 is merged and then you can test the device following our https://github.com/freifunk-gluon/gluon/wiki/Device-Integration-checklist

PR #1526 (2018-09-03): ipq806x: mark as broken because of unstable wifi drivers (v2018.1.x)
closes #1505