freifunk-gluon/gluon #1507: ipv6 only config mode

MoritzFago, 2018-08-08:
If we would drop v4 config mode support, we could save some space on the space constrained targets.

mweinelt, 2018-08-08:
AFAIK: Windows 7/8 do not support RADV. That means users on these systems would need to do manual IP configuration, which isn't exactly user friendly.

rotanid, 2018-08-08:
i think the first step before IPv6 only has to be dual stack.
is IPv6 possible at the moment? is it accessible using something easier than a link-local address?

blocktrron, 2018-08-08:
In case saving space is the primary goal, i think it makes more sense to simply drop DHCP and make configuring yourself a static IP-address mandatory for those devices.

NeoRaider, 2018-08-08:
1. We will need some kind of DHCP for client addresses with Layer 3 meshing anyways
2. dnsmasq is deeply integrated in OpenWrt, removing it will take a lot of effort (and possibly ongoing maintenance)
3. I think there are better ways to deal with space-constrained targets. One option would be to replace the JFFS2 overlay with something simpler (like the tar-based snapshot overlays OpenWrt already supports; RAM usage would need to be evaluated).

I still think adding IPv6 support is generally a good idea, but only for the sake of future proofing the config mode, not for any other reason. Adding a ULA address (from which prefix?) in config mode seems like a good first step.

Note that I would like to see the config mode in its current form (i.e. as a separate mode of operation that requires a reboot to change into or out of) go away altogether, but that's more of a long-term idea and certainly out of scope for this ticket.

rotanid, 2018-08-08:
so we would welcome an issue (or even better a patch) about adding dual-stack to the config-mode, but ipv6 only won't happen in the near future.
to start this clean, i'll close here and welcome a fresh one with an adjusted direction.

closed 2018-08-08 by rotanid