freifunk-gluon/gluon #1510: Config Mode Hostname

mweinelt, 2018-08-08:
In config mode provide a hostname that the node resolves to itself, so we can use a name to reference to its IP addresses. Just like we already do with the nextnode address.

This requires us to set the nameserver DHCP option, if we don't do that already, and set up RDNSS.

### Example

#### site.conf
```lua
config_mode = {
    name = { 'setup', 'setup.ffda.io' },
}
```

NeoRaider, 2018-08-08:
Unfortunately, this conflicts with the idea that the user is in a private WLAN for internet access while connected to the node in config mode (e.g. for loading a map to display). The node should not advertise DNS in config mode to avoid breaking internet connectivity...

kaechele, 2018-09-13:
Maybe [umdns](https://openwrt.org/docs/guide-developer/mdns#umdns) could be useful here?