freifunk-gluon/gluon #1519: RFC: Mention updater Bug and workarounds in 2018.1.0 Release notes

A-Kasper, 2018-08-24:
I think it would be good, if the bug (and also how to avoid that the problem appears) is mentioned in the Release notes.

rotanid, 2018-08-24:
changing existing release notes doesn't help anyone, because the content of the following page will not be changed by your suggested change:
https://gluon.readthedocs.io/en/v2018.1/releases/v2018.1.html
we can only do this on v2018.1.1 release notes and i'm sure @NeoRaider will do this

A-Kasper, 2018-08-24:
that is correct. sorry.

NeoRaider, 2018-08-24:
We generally link the release notes in the stable branch rather than the release tag (  https://gluon.readthedocs.io/en/v2018.1.x/releases/v2018.1.html ), for the precise reason that we may want to update the release notes.

commit bc0d240d (2018-08-26): docs: releases/v2018.1: add regressions to known issues

Fixes #1519
closed 2018-08-28 by NeoRaider