freifunk-gluon/gluon #1520: GL.inet AR750 disconnects clients on 5 GHz WLAN

Badkappenfrieda, 2018-08-24:
I tried both the 2017.1.8 as well as the 2018.1.0 firmware (stable/sysupgrade) for the AR750. 
On 5 GHz, clients first wait a couple of seconds until they get an IP via DHCP. After a few more seconds, the clients are disassociated from the AP.

Badkappenfrieda, 2018-08-24:
From logread:

Fri Aug 24 23:27:14 2018 daemon.info hostapd: client0: STA 84:9f:b5:03:29:2a IEEE 802.11: disconnected due to excessive missing ACKs
Fri Aug 24 23:27:14 2018 daemon.notice hostapd: client0: AP-STA-DISCONNECTED 84:9f:b5:03:29:2a
Fri Aug 24 23:27:14 2018 kern.warn kernel: [  422.963297] ath10k_pci 0000:00:00.0: removing peer, htt peer unmap vdev 0 peer 84:9f:b5:03:29:2a id 6
Fri Aug 24 23:27:14 2018 kern.debug kernel: [  423.413681] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
Fri Aug 24 23:27:14 2018 kern.debug kernel: [  423.413719] ath10k: [0000]: E0B90600 1F4CFC13 ADEE0300 2B002C00 03000700 C801C901 E0B90600 1F4CFC13
Fri Aug 24 23:27:14 2018 kern.debug kernel: [  423.413740] ath10k: [0008]: E0B90600 89030004 A0000000 6C000000 EBB90600 1F4CFC13 ADEE0300 2C002D00
Fri Aug 24 23:27:14 2018 kern.debug kernel: [  423.413759] ath10k: [0016]: 04000800 C901CA01 EBB90600 1F4CFC13 EBB90600 89030004 A0000000 6D000000
Fri Aug 24 23:27:14 2018 kern.debug kernel: [  423.413779] ath10k: [0024]: EEB90600 100CFC0B 00243200 01000000 F4B90600 234CFC0F 01000000 01000000
Fri Aug 24 23:27:14 2018 kern.debug kernel: [  423.413797] ath10k: [0032]: 01000000 02BA0600 025C000C 849FB503 292A0000 01000000
Fri Aug 24 23:27:14 2018 kern.debug kernel: [  423.413812] ath10k_pci 0000:00:00.0: ATH10K_END
Fri Aug 24 23:27:18 2018 kern.warn kernel: [  427.160302] ath10k_pci 0000:00:00.0: htt peer map vdev 0 peer 84:9f:b5:03:29:2a id 6
Fri Aug 24 23:27:18 2018 daemon.info hostapd: client0: STA 84:9f:b5:03:29:2a IEEE 802.11: authenticated
Fri Aug 24 23:27:18 2018 daemon.info hostapd: client0: STA 84:9f:b5:03:29:2a IEEE 802.11: associated (aid 1)
Fri Aug 24 23:27:18 2018 daemon.notice hostapd: client0: AP-STA-CONNECTED 84:9f:b5:03:29:2a
Fri Aug 24 23:27:18 2018 kern.debug kernel: [  427.414229] ath10k_pci 0000:00:00.0: ath10k_pci ATH10K_DBG_BUFFER:
Fri Aug 24 23:27:18 2018 kern.debug kernel: [  427.414266] ath10k: [0000]: 05CB0600 1B4CFC07 04534300 05CB0600 0128FC17 01000000 08F00F00 03900102
Fri Aug 24 23:27:18 2018 kern.debug kernel: [  427.414280] ath10k: [0008]: FDFF0000 FF000000
Fri Aug 24 23:27:18 2018 kern.debug kernel: [  427.414294] ath10k_pci 0000:00:00.0: ATH10K_END

rotanid, 2018-08-25:
have you built the ibss or 11s variant?
have you tried "original" upstream OpenWrt Builds from LEDE 17.01.5 or OpenWrt 18.06.1 ?

jotzt, 2018-08-25:
I observed a similar problem with the TP Archer C2600 (OpenWRt 18.06 & 11s) as long as meshing is enabled. Did you try to disable mesh on 5 GHz? To me, it looks like there might be an issue with the ath10k driver.

blocktrron, 2018-08-25:
@Badkappenfrieda You could try `uci set wireless.client_radio0.disassoc_low_ack='0' && uci commit wireless && wifi`. As you seem to be using -ct driver (indicated #1495 ) you should consider trying to build a firmware with 11s and try again. 

If you problem disappears it's most likely an issue with the -ct driver in conjunction with your specific chip.

Badkappenfrieda, 2018-08-27:
I did not build the software myself but was using the FF3L (Freifunk Drei-Ländereck) Software. 

The UCI settings from @blocktrron helped stabilize the 5 GHz link. However, there seems to be an issue with routing or something. At least I do not get any connection via 5 GHz, all the while 2,4 GHz is working just fine.

blocktrron, 2018-08-27:
As the WiFi-Chip used in this particular device is not very common (QCA9887), it could indeed be that there are issues which are not yet known. Additionally it got backported to Gluon which might cause issues due to old ath10k driver / firmware. The QCA9887 did receive a firmware bump 8 months ago (https://github.com/openwrt/openwrt/commit/f6a8505de05b818561eaf403a4059a59286fb089)

You can try the following to pinpoint the issue: 
 - Disable mesh on 5GHz
 - Try OpenWRT 18.06 and check if the issue persists

In case OpenWRT 18.06 does not  have this issue, ask someone from ff3l for an Image based on Gluon master.

rotanid, 2019-02-16:
i do own this device now, too.
i just tried to reproduce your reported problem using this GL-AR750 running Gluon v2018.2.x branch.
I could connect to the routers 5GHz wifi easily with a thinkpad with an intel AC wifi card and with a LG G3 android phone.
i also tested 5GHz 11s mesh with a WDR4900, this also worked fine.
therefore i'll close this bugreport as "cant reproduce", it may be a problem with your community-specific firmware or with your client software.

closed 2019-02-16 by rotanid