freifunk-gluon/gluon #1522: Githubmessages in Freifunk slack

A-Kasper, 2018-08-29:
I liked the messages in the gluon channel in Freifunk slack. The channel seems to be silent since 26 july.

rotanid, 2018-08-29:
aha! we saw some slack webhook in the github configuration but didn't knew what it does and who it set up, so we removed it.

A-Kasper, 2018-08-29:
There is a gluon channel in Freifunk Slack. Every commit was posted there by a bot. As far as I know only the repository owner is able to setup this service. It would be nice if it could be re-enabled. I liked it.

rotanid, 2018-08-29:
and what is "Freifunk Slack"?
Does the repository owner need do create/have an account there? if yes, it probably won't happen. if not, we'll discuss.

A-Kasper, 2018-09-01:
It is the slack of the Freifunk Rheinland. It's open.
Somebody already setup this function. As you know there was the necessary api key in github configured. I'll have a look how to reenable the bot. I think the owner just needs to communicate or accept an api key.

rotanid, 2018-09-02:
as far as i know no one in the current gluon team uses the slack of freifunk rheinland - therefore it wouldn't be a good idea to give someone else an API key for the Gluon github organization...

NeoRaider, 2018-09-02:
I assume the integration is using a webhook. Readding a webhook for push notifications shouldn't be a problem.

nomaster, 2018-09-05:
The integration works with a dedicated GitHub Slack app that needs access to this project. So this can only be done of an orga or project member.

Alternately, we can use some RSS app to subscribe to release events.

mweinelt, 2018-11-25:
Does this kind of integration really need organization access? https://slack.github.com/

nomaster, 2018-11-25:
The Slack app has been released in a totally new, incompatible version back in July. Since then, the old version that is currently in use on Freifunk Slack is dysfunctional.

I recommend removing the old access and reinstalling the Slack app with the following command (in the channel #gluon on Freifunk Slack):
```
/github subscribe freifunk-gluon/gluon
```

mweinelt, 2018-11-25:
So nothing for us to do, correct? If so, I'd close the issue.

mweinelt, 2018-12-14:
Closing for lack of feedback.

closed 2018-12-14 by mweinelt