freifunk-gluon/gluon #1524: 2018.1.1 not building due to issue with iproute2 package

lcb01a, 2018-08-30:
Hi,

While building a new firmware using a clean buildroot for 2018.1.1 I encounter the following issue:

> make[5]: Entering directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0'
make[6]: Entering directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/lib'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/lib'
make[6]: Entering directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/ip'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/ip'
make[6]: Entering directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/tc'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/tc'
make[6]: Entering directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/bridge'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/bridge'
make[6]: Entering directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/misc'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/misc'
make[6]: Entering directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/genl'
make[6]: Nothing to be done for 'all'.
make[6]: Leaving directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/genl'
make[6]: Entering directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/tipc'
make[6]: *** No rule to make target 'ar71xx-tiny', needed by 'all'.  Stop.
make[6]: Leaving directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/tipc'
Makefile:50: recipe for target 'all' failed
make[5]: *** [all] Error 2
make[5]: Leaving directory '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0'
Makefile:154: recipe for target '/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/.built' failed
make[4]: *** [/home/cyrus/git/gluon-2018.1.1/lede/build_dir/target-mips_24kc_musl-1.1.16/iproute2-full/iproute2-4.4.0/.built] Error 2
make[4]: Leaving directory '/home/cyrus/git/gluon-2018.1.1/lede/package/network/utils/iproute2'
package/Makefile:105: recipe for target 'package/network/utils/iproute2/compile' failed
make[3]: *** [package/network/utils/iproute2/compile] Error 2
make[3]: Leaving directory '/home/cyrus/git/gluon-2018.1.1/lede'
package/Makefile:101: recipe for target '/home/cyrus/git/gluon-2018.1.1/lede/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_compile' failed
make[2]: *** [/home/cyrus/git/gluon-2018.1.1/lede/staging_dir/target-mips_24kc_musl-1.1.16/stamp/.package_compile] Error 2
make[2]: Leaving directory '/home/cyrus/git/gluon-2018.1.1/lede'
/home/cyrus/git/gluon-2018.1.1/lede/include/toplevel.mk:205: recipe for target 'world' failed
make[1]: *** [world] Error 2
make[1]: Leaving directory '/home/cyrus/git/gluon-2018.1.1/lede'
Makefile:148: recipe for target 'all' failed
make: *** [all] Error 2

When trying to build a 2018.1 firmware using the same configuration I encounter no issues.
Used config branch: https://github.com/ffddorf/gluon-site/tree/feature_multidomain

I suspect some issue with the iproute2 package that is causing this.

Cheers

NeoRaider, 2018-08-30:
Hmm, this is a duplicate, but I can't find the other ticket right now.

I fixed this in upstream iproute2 a while back (https://git.kernel.org/pub/scm/network/iproute2/iproute2.git/commit/?id=cfd6ccbfd0311d01a29b490c472ad79b6ad8036b), and it should be fixed in OpenWrt 18.06+. For older releases, the workaround is not to define an environment variable with the name *TARGETS*.

closed 2018-08-30 by NeoRaider

rotanid, 2018-08-30:
ack, i just built the Gluon tag and had no issue whatsover.

lcb01a, 2018-08-31:
Thanks for the clarification :) I adjusted my build wrapper script and it works now.