freifunk-gluon/gluon #1527: Derive target domain from location

mweinelt, 2018-09-04:
This probably needs to extend domains (and pretty names) with polygons so locations can be automatically selected based on the position.

2tata, 2018-09-04:
This will be already provided by #1393

rotanid, 2018-09-05:
yes, should be handled by that PR, review still pending:(

kpanic23, 2018-09-13:
@mweinelt 
Have a look at https://github.com/ff3l/packages
I've extended FFRGB's gluon-segment-mover to have this feature. The package gluon-config-mode-segment-mover can be used to be able to disable it in config mode.
It uses a web API to get the correct domain for the coordinates, since our geojson shape files would be larger than most devices' flash chips. The (super simple) web API can be found at https://github.com/ff3l/ff3l-api

2tata, 2018-09-23:
Shall we close this Issue due to its duplicated ? by the hoodselector PRs ?

rotanid, 2018-09-23:
@2tata an issue can be solved by a merged PR. therefore it stays open until the solution is merged :)

CodeFetch, 2019-06-29:
@rotanid Can be closed now. The suggestion by kpanic23 has been incorporated in #1773.

2tata, 2019-09-16:
solved by #1739

CodeFetch, 2019-09-16:
@rotanid Please close this issue.

closed 2019-09-16 by mweinelt