freifunk-gluon/gluon #1530: Browser cache, redirections and related issues

kpanic23, 2018-09-13:
When flashing multiple nodes without setting a custom name, they all end up with the same hostname, generated using the first node's host ID. 

I would propose adding "no-cache"-headers to the config mode pages as well as to the status page:

The redirection of different gluon version's status pages differ, if these get cached one can't reach the status page without clearing the browser's cache.

Example: You open the status page using the nextnode address. On Gluon 2018.1 it gets redirected to /cgi-bin/status.
Now, if you try to open a different node's status page running Gluon 2017.1 is not possible, since you get redirected automatically and that path does not exist in older Gluon versions.
The same happens if you downgrade to an older Gluon version.

That redirection should be done using HTTP 303 instead of 301, so it doesn't get cached.

rotanid, 2018-09-13:
that's a good idea, maybe you can develop, test and provide the patch for this? :)

mweinelt, 2018-09-30:
The redirect function in master uses a HTTP 302, which seems to be a temporary redirect, see https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-web/luasrc/usr/lib/lua/gluon/web/http.lua#L118-L122.

The index.html on the status-page is using a meta http-equiv refresh, see https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-status-page/files/lib/gluon/status-page/www/index.html.

I'm not too familiar how those are cached, so is it possible the caching happened before we got to gluon-web?