freifunk-gluon/gluon #1532: TP-Link TL-WR902AC (AC750) v1

Dark4MD, 2018-09-17:
Is it possible to support this device? It is fully supported since the nightlies for 18.06 and has everything working with the final 18.06. It has 8MB Flash and 64MB Ram. I have multiple of those and would like to use it with gluon.

blocktrron, 2018-09-17:
Theoretically yes. Although 64M RAM are pretty small for an ath10k device, so it won't be very future-proof in the long run (Perdonally, i would not "Recommend" the device to others).

There might be issues with the 5GHz WiFi as it is using the same chip as the device in #1520 . But we didn't got the information if this issue still persists in OpenWRT18.06.

Dark4MD, 2018-09-27:
I just build a working Image myself with the gluon Master and the ffh site conf for the device. For me the 5GHz WiFi works as far as i can tell (made an extra SSID just for the 5GHz radio to test it). Mesh on both Radios is active and working on my two devices i have and my clients still have a working connection on the 5GHz radio.

Dark4MD, 2018-09-27:
Correct entry btw for ar71xx-generic is:
device tp-link-tl-wr902ac-v1 tl-wr902ac-v1

blocktrron, 2018-09-27:
@Dark4MD Could you verify the things outlined in the linked comment (expecially the button function, led mapping and primary mac-adress)?

https://github.com/freifunk-gluon/gluon/pull/1494#issue-203848523

blocktrron, 2018-09-28:
@Dark4MD Thanks for providing this checklist - helps a lot!

One more thing: Could you provide the output of `lua -e 'print(require("platform_info").get_image_name())'? This is important to generate the correct filename for the autoupdater.

> radio1 (which is the 2,4GHz radio) gets the hardware mac

So we need to acquire Node-ID from WiFi1. 

`

Dark4MD, 2018-09-28:
```
root@ManuWTFWR902ACSelfBuild:~# lua -e 'print(require("platform_info").get_image_name())'
tp-link-tl-wr902ac-v1

```

Dark4MD, 2018-10-02:
@blocktrron It works. The Node-ID is the same as the Mac Address on the Case and the device works without Problems.