freifunk-gluon/gluon #1536: fastd vpn encryption/performance option missing in configmode

Grotax, 2018-09-24:
Today I received the question why the option to configure the fastd encryption was missing form the web-interface.
To be sure I did a extra build without any extra packages.
Still the option is missing.
Maybe the site config is incorrect but I think it's fine.
[image]

Site config: https://github.com/ffsh/site/blob/testing/site.conf
Images: https://firmware.grotax.de/testing/
Build with https://github.com/freifunk-gluon/gluon/commit/51b7928a55155dc2a462cef5d4c045a81d9c1e0c

NeoRaider, 2018-09-24:
The page is provided by the "web-mesh-vpn-fastd" GLUON_FEATURES entry. It is not included by default, as many sites don't want to allow configuring the encryption.

closed 2018-09-24 by NeoRaider

Grotax, 2018-09-24:
Thank you for the quick response :)