freifunk-gluon/gluon #1538: Support for TP-Link WBS210 v2.0

datendelphin, 2018-09-26:
I got a new TP-link WBS210 and it reports v2.0 on the status page.

mweinelt, 2018-09-29:
Not supported in OpenWrt yet and no pull request yet. You might want to ask around and send your device to someone who is able to build the support.

Until that is resolved Gluon support is out of the question, sorry.

datendelphin, 2018-10-05:
I guess it is the same hardware as v1.2 I added following strings:

                       "WBS210(TP-LINK|UN|N300-2|00000000):2.0\r\n"
                       "WBS210(TP-LINK|US|N300-2|55530000):2.0\r\n"
                       "WBS210(TP-LINK|EU|N300-2|45550000):2.0\r\n"

to .support_list of tplink-safeloader.c, compiled openwrt and could upload the image and log in via ssh.

it looks similar to the patch included for CPE210 V2. I'll have a look at where I can report that to openwrt

mweinelt, 2018-11-26:
Support added to OpenWrt master in 5c5bf8b8658a588423f6ec445d7ef6a36f99a396. You'll probably also need 2524febf7927a1bf430d64b7790feb126023e3d1.

datendelphin, 2019-01-19:
I compiled openwrt with the dirty hack i mentioned above, and that works and did not brick my device. Now I'm trying to go the route with the mentioned patches, building dedicated v2 images, though I did not manage yet to build them, mostly due to lack of time and this is the first time tinkering with openwrt.

RalfJung, 2019-02-13:
> Support added to OpenWrt master in 5c5bf8b8658a588423f6ec445d7ef6a36f99a396. You'll probably also need 2524febf7927a1bf430d64b7790feb126023e3d1.

This looks like it is about the CPE210 v2, not the WBS210 v2?

datendelphin, 2019-02-13:
The expectation is that it is mainly the same hardware, except that the WBS has antenna connectors while the CPE has internal antennas.
My device in the mean time doesn't want to flash with the factory firmware properly. Still trying with tftp.

RalfJung, 2019-02-13:
FWIW, we have just successfully flashed a WBS v2 with our firmware, after patching it as described above.  Wifi meshing is working as well.

rotanid, 2019-02-13:
@RalfJung so you could go through our checklist and create a pull request maybe

RalfJung, 2019-02-14:
Well I don't have the device nor did I do the flashing, but maybe I can get the people that did to do that.^^

Which checklist? I checked the README and CONTRIBUTING files and found no checklist.

bobcanthelpyou, 2019-02-14:
@RalfJung maybe the device integration checklist: https://github.com/freifunk-gluon/gluon/wiki/Device-Integration-checklist

gobold1979, 2019-02-18:
I'm one of those @RalfJung mentioned, the one who bought a WBS210 V2 and flashed the "FrankenFirmware" onto it. After that worked, I bought another 10 WBS210. While we did flash the first one from the normal login, we used TFTP to flash the other ten. It worked, but it's different compared to other TP-Link routers though. It expects the TFTP-Server to run on a host with IP 192.168.0.100 instead of 192.168.0.66 and it asks for "recovery.bin" instead of something bearing the router specifications in the file name.

I didn't make the firmware, but we'll see what we can do to get into OpenWRT and from there down the stream. It contains a minor bug that I don't understand yet: On the map, it shows up as a WBS210 V2.[insert a whole lot of bad encoding symbols here]0. Aside from that, it works.

A WBS510 is on its way, we'll see what happens.

gobold1979, 2019-02-19:
> Moin, is the recommended "fix" already tested to not brick a router? We got a User in our community who would like to use a WBS210 but only got a v2 version.

I can at least say, that thanks to TFTP, it is not possible to brick these things by just flashing something weird onto them.

gobold1979, 2019-02-19:
JFTR: The WBS510 has arrived and it's a V1.20. But that's not what this thread is about anyway.

gobold1979, 2019-02-19:
It seems like our Firmware ignores the PoE-Passthrough-Setting. Does anyone have an idea how to deal with that?

mweinelt, 2019-02-19:
> It seems like our Firmware ignores the PoE-Passthrough-Setting. Does
> anyone have an idea how to deal with that?

What is it exactly that is being ignored?

Power over Ethernet should be toggleable from withing the Network tab in
the config mode. It represents a subset of gpio switches below `uci show
system.gpio_switch`.

Compare
https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-web-network/luasrc/lib/gluon/config-mode/model/admin/network.lua#L110-L130

gobold1979, 2019-02-22:
> What is it exactly that is being ignored? Power over Ethernet should be toggleable from withing the Network tab in the config mode.

That's exactly what I mean. In detail, let's compare the expected behaviour, as it can be observed as actual behaviour on the WBS510 V1.2 to the actual behaviour of a WBS 210 V2.0:

For the WBS510V1.2...
1. Go to the network tab ("Erweiterte Einstellungen" -> "Netzwerk")
2. Check the box next to Activate PoE Passthrough ("PoE-Passthrough aktivieren")
3. Click "Save" / "Speichern"
4. Switch to the wizard
5. Click "Save and reboot" / "Speichern und neustarten"
6. The device reboots. After a minute or so, it activates PoE passthrough, i.e. the LAN port starts to provide passive PoE.

For the WBS210V2.0
Steps 1 through 5 as above.
6. It reboots, but there's no PoE on the LAN port.

> It represents a subset of gpio switches below `uci show system.gpio_switch`. Compare https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-web-network/luasrc/lib/gluon/config-mode/model/admin/network.lua#L110-L130

I don't have any knowledge on that, but I'll have a look. Thanks for the link!

citronalco, 2019-06-14:
Here are highly experimental patches for OpenWrt and Gluon adding support for the WBS 210 V2.0:
[ftp://ftp.geierb.de/gluon/](ftp://ftp.geierb.de/gluon/)
Most likely PoE passthrough won't work (didn't test it), but it builds an own image for the wbs210v2 which can be flashed using TP-Links web GUI, the model shows up correctly and Wifi works as expected.

citronalco, 2019-09-23:
> Here are patches for OpenWrt and Gluon adding support for the WBS 210 V2.0:
> ftp://ftp.geierb.de/gluon/
> Most likely PoE passthrough won't work (didn't test it), but it builds an own image for the wbs210v2 which can be flashed using TP-Links web GUI, the model shows up correctly and Wifi works as expected.
> I have no real clue about the whole thing, so I'd highly appreciate if someone else could take over.

As nobody else seems to be interested, I'll try to bring these patches into shape.
Flashing the factory image using the web ui works, but a sysupgrade on command line fails with
```
Image metadata not found
Invalid image magic '003a0d3d'. Expected '01000000'.
Image check 'platform_check_image' failed.
```
Does anyone know how to solve this?

MTRNord, 2019-09-28:
I am interested but sadly didn't have the time to test it yet as I don't own the device myself but only a community member who really would like to use his device. If I can do any tests that help, please ping me. But keep in mind I do not have the device on my desk. Max would be to build a firmware, get it to the device owner and remote shell into it if it works :)

citronalco, 2019-09-28:
It works properly, but unfortunately won't be merged. See my pull request #1827 
Porting the device was not that hard, I could build upon old gluon pull requests for similar TP-Link devices. But porting it to OpenWRT's target ath79, as it is required now for gluon, is probably above my abilities.

MTRNord, 2019-09-28:
Ok :/ I don't have the required knowledge to do the port either.

rotanid, 2019-09-28:
> But porting it to OpenWRT's target ath79, as it is required now for gluon

this is not true.
Gluon does not support hardware, that is not supported by OpenWrt.
And OpenWrt does not contain WBS210 v2 at the moment, neither in ar71xx target nor in ath79 target. 
Therefore, Gluon does not accept a patch as was always the policy, so we don't carry downstream patches forever which aren't upstream.

Regarding OpenWrt, they do not accept new devices into ar71xx target as it got replaced by ath79.
Even if they would accept a device, it would need to be supported in ath79 *before*

RalfJung, 2019-10-02:
But, isn't it a consequence of that those OpenWRT and gluon policies that the device could only get into gluon if it gets ported to ath79? Or am I missing something?

rotanid, 2019-10-02:
> But, isn't it a consequence of that those OpenWRT and gluon policies that the device could only get into gluon if it gets ported to ath79? Or am I missing something?
> 
> If what I said is accurate, how is the statement "porting to ath79 is required for gluon" not true? 

you are right

TomTheDragon, 2019-11-03:
The WBS210 v2 is now supported by Openwrt
https://github.com/openwrt/openwrt/commit/3930aab2cb067d04f5c9ebbf1af57744e19355eb

rotanid, 2019-11-03:
great, so as soon as Gluon has the ath79 target we can support this device via a backport of the linked patch.

until then, someone could now try to get the ar71xx variant supported in openwrt, too.
but maybe they reject it for ar71xx, there's no clear policy that i know of.

mweinelt, 2020-03-03:
We are ready to tackle this one on master.

TomTheDragon, 2020-09-02:
Any updates on the WBS210 v2 support of Gluon?

mweinelt, 2020-09-02:
We rely on third-party contributions for device support, since we need someone to verify the hardware support.

rotanid, 2020-09-02:
> We rely on third-party contributions for device support, since we need someone to verify the hardware support.

... unless someone donates such a device.

citronalco, 2020-09-10:
Gluon master still uses an OpenWRT branch too old to support WBS210 v2.
If I'm not wrong it would currently not only be testing and then enabling the device in Gluon, but also backporting all the stuff.

rotanid, 2020-09-10:
you could try it with the "next" branch and as soon as next branch becomes master, you can create your Pull Request

citronalco, 2020-09-12:
Did a quick(!) test: Device works with the factory image that's built when adding the line
`device('tp-link-wbs210-v2', 'tplink_wbs210-v2')"`
to targets/ath79-generic
Still have to check sysupgrade and LEDs.

But: The device has two ethernet ports: eth0 with PoE, and eth1(without PoE).
Config mode is on eth0, while WAN is on eth1. As this is an outdoor device and can only be powered using the PoE port, this seems somewhat impractical to me. Any hint where I can set WAN to eth0?

NeoRaider, 2020-09-12:
- Switch LAN and WAN in `package/gluon-core/luasrc/lib/gluon/upgrade/020-interfaces`
- Add the device to the lists of outdoor devices in `package/gluon-core/luasrc/usr/lib/lua/gluon/platform.lua`
 to get Config Mode on WAN instead of LAN

citronalco, 2020-09-15:
Done. Config Mode and WAN are both on the PoE port, sysupgrade works, LEDs are ok. Everything is fine.

As far as I know it's not possible to make a PR in the "next" branch. So if anyone's eager to test or not able to wait until the next Gluon release here's the relevant commit: https://github.com/citronalco/gluon/commit/2e6c7fbdc4d412ef1e13f502a9e51b6872ca405e

AiyionPrime, 2022-06-16:
I think this is resolved since the merge of #2299.
If someone would close this?

closed 2022-06-16 by rotanid