freifunk-gluon/gluon #1561: 5GHz doesn't work on Archer C7 with 2018.1.1

oliver, 2018-11-02:
I have flashed gluon-ffhb-2018.1.1+bremen1-tp-link-archer-c7-v2.bin (from Freifunk Bremen community) onto an Archer C7 v2. The device appears to basically work, except for 5GHz wifi. There is an ibss1 network but no ibss0.

On a similar device the gluon-ffhb-2017.1.8+bremen1-tp-link-archer-c7-v2.bin firmware works fine, including 5GHz mesh and client network.

oliver, 2018-11-02:
Here is the dmesg output for the 2018.1.1 device:
[dmesg-2018.1.1.ac7.txt](https://github.com/freifunk-gluon/gluon/files/2544234/dmesg-2018.1.1.ac7.txt)

mweinelt, 2018-11-02:
Please post the output of `uci show wireless`.

rotanid, 2018-11-03:
for us 5 GHz is working on Archer C7 v2, but we're not using ibss but 802.11s, so maybe its related to this difference

mweinelt, 2018-11-03:
And maybe `iw dev`?

oliver, 2018-11-03:
```
root@og-ac7-testing-2:~# iw dev
phy#1
	Interface client1
		ifindex 15
		wdev 0x100000003
		addr f4:f2:6d:70:b2:76
		ssid bremen.freifunk.net
		type AP
		channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
		txpower 20.00 dBm
	Interface ibss1
		ifindex 12
		wdev 0x100000002
		addr 0e:9e:42:44:91:ae
		ssid mesh.ffhb.de
		type IBSS
		channel 6 (2437 MHz), width: 20 MHz, center1: 2437 MHz
		txpower 20.00 dBm
```

mweinelt, 2018-11-03:
Something is wrong with the firmware load.

```
[   10.977818] ath10k driver, optimized for CT firmware, probing pci device: 0x3c.
[   10.986486] PCI: Enabling device 0000:01:00.0 (0000 -> 0002)
[   10.992429] ath10k_pci 0000:01:00.0: pci irq legacy oper_irq_mode 1 irq_mode 0 reset_mode 0
[   11.214967] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/pre-cal-pci-0000:01:00.0.bin failed with error -2
[   11.225851] ath10k_pci 0000:01:00.0: Falling back to user helper
[   11.360684] firmware ath10k!pre-cal-pci-0000:01:00.0.bin: firmware_loading_store: map pages failed
[   11.371219] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/fwcfg-pci-0000:01:00.0.txt failed with error -2
[   11.381924] ath10k_pci 0000:01:00.0: Falling back to user helper
[   11.544417] firmware ath10k!fwcfg-pci-0000:01:00.0.txt: firmware_loading_store: map pages failed
[   11.757678] ath10k_pci 0000:01:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[   11.767084] ath10k_pci 0000:01:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[   11.780112] ath10k_pci 0000:01:00.0: firmware ver 10.2.4-1.0-00016 api 5 features no-p2p,raw-mode,mfp crc32 0c5668f8
[   11.790911] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/board-2.bin failed with error -2
[   11.801512] ath10k_pci 0000:01:00.0: Falling back to user helper
[   11.900849] firmware ath10k!QCA988X!hw2.0!board-2.bin: firmware_loading_store: map pages failed
[   11.910088] ath10k_pci 0000:01:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[   13.008685] ath10k_pci 0000:01:00.0: wmi print 'P 145 V 16 T 443'
[   13.014927] ath10k_pci 0000:01:00.0: wmi print '_wlan_rtt_enable'
[   13.023000] ath10k_pci 0000:01:00.0: htt-ver 2.1 wmi-op 5 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[   13.032669] ath10k_pci 0000:01:00.0: no channel configured; ignoring frame(s)!
[   13.040018] ath10k_pci 0000:01:00.0: no channel configured; ignoring frame(s)!
[   13.047462] ath10k_pci 0000:01:00.0: no channel configured; ignoring frame(s)!
```

oliver, 2018-11-03:
Output of `iw list`: [iw-list.txt](https://github.com/freifunk-gluon/gluon/files/2545092/iw-list.txt)

oliver, 2018-11-03:
@mweinelt: I think with 2017.1.8 there are similar firmware load errors, but 5 GHz works there. Here's some dmesg output from an AC7v2 with 2017.1.8:

```
[   11.087224] ath10k_pci 0000:01:00.0: could not fetch firmware file 'ath10k/QCA988X/hw2.0/firmware-5.bin': -11
[   11.097379] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/firmware-4.bin failed with error -2
[   11.108237] ath10k_pci 0000:01:00.0: Falling back to user helper
[   11.263193] firmware ath10k!QCA988X!hw2.0!firmware-4.bin: firmware_loading_store: map pages failed
[   11.272480] ath10k_pci 0000:01:00.0: could not fetch firmware file 'ath10k/QCA988X/hw2.0/firmware-4.bin': -11
[   11.282640] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/firmware-3.bin failed with error -2
[   11.293505] ath10k_pci 0000:01:00.0: Falling back to user helper
[   11.447880] firmware ath10k!QCA988X!hw2.0!firmware-3.bin: firmware_loading_store: map pages failed
[   11.457142] ath10k_pci 0000:01:00.0: could not fetch firmware file 'ath10k/QCA988X/hw2.0/firmware-3.bin': -11
[   11.656609] ath10k_pci 0000:01:00.0: qca988x hw2.0 target 0x4100016c chip_id 0x043202ff sub 0000:0000
[   11.666018] ath10k_pci 0000:01:00.0: kconfig debug 1 debugfs 1 tracing 0 dfs 1 testmode 0
[   11.678317] ath10k_pci 0000:01:00.0: firmware ver 10.1.467-ct-_fW-019-60dd7c5 api 2 features wmi-10.x,has-wmi-mgmt-tx,txstatus-noack,wmi-10.x-CT,ra
temask-CT crc32 f59d154a
[   11.693943] ath10k_pci 0000:01:00.0: Direct firmware load for ath10k/QCA988X/hw2.0/board-2.bin failed with error -2
[   11.704546] ath10k_pci 0000:01:00.0: Falling back to user helper
[   11.800537] firmware ath10k!QCA988X!hw2.0!board-2.bin: firmware_loading_store: map pages failed
[   11.809825] ath10k_pci 0000:01:00.0: board_file api 1 bmi_id N/A crc32 bebc7c08
[   12.720566] ath10k_pci 0000:01:00.0: 10.1 wmi init: vdevs: 16  peers: 127  tid: 256
[   12.737963] ath10k_pci 0000:01:00.0: wmi print 'P 128 V 8 T 410'
[   12.744120] ath10k_pci 0000:01:00.0: wmi print 'msdu-desc: 1424  sw-crypt: 0'
[   12.751369] ath10k_pci 0000:01:00.0: wmi print 'alloc rem: 25908 iram: 25808'
[   12.805422] ath10k_pci 0000:01:00.0: htt-ver 2.1 wmi-op 2 htt-op 2 cal file max-sta 128 raw 0 hwcrypto 1
[   12.921710] ath: EEPROM regdomain: 0x0
[   12.921724] ath: EEPROM indicates default country code should be used
[   12.921732] ath: doing EEPROM country->regdmn map search
[   12.921748] ath: country maps to regdmn code: 0x3a
[   12.921757] ath: Country alpha2 being used: US
[   12.921765] ath: Regpair used: 0x3a
[   12.958794] batman_adv: B.A.T.M.A.N. advanced 2013.4.0 (compatibility version 14) loaded
[   13.055471] u32 classifier
[   13.058218]     input device check on
[   13.061982]     Actions configured
[   13.067524] Mirror/redirect action on
[   13.174410] Bridge firewalling registered
[   13.195464] Ebtables v2.0 registered
[   13.201665] ip_tables: (C) 2000-2006 Netfilter Core Team
```

oliver, 2018-11-03:
For comparison, here's the full output of `dmesg`, `iw dev` and `iw list` of an AC7v2 with 2017.1.8+bremen1 (this device: https://map.bremen.freifunk.net/#!/en/map/60e3274fdac3):
* [dmesg-2017.1.8.5ghztestnode.txt](https://github.com/freifunk-gluon/gluon/files/2545178/dmesg-2017.1.8.5ghztestnode.txt)
* [iw-dev.2017.1.8.5ghztestnode.txt](https://github.com/freifunk-gluon/gluon/files/2545179/iw-dev.2017.1.8.5ghztestnode.txt)
* [iw-list.2017.1.8.5ghztestnode.txt](https://github.com/freifunk-gluon/gluon/files/2545180/iw-list.2017.1.8.5ghztestnode.txt)

NeoRaider, 2018-11-03:
The firmware load warnings look completely normal. I'll have a closer look at the logs tomorrow.

oliver, 2018-11-03:
Copying the ath10k firmware from 2017.1.8 appears to fix this: the node now meshes on 5GHz and also has a 5GHz client, according to Meshviewer.

This is the firmware I've used: [firmware-2.bin.gz](https://github.com/freifunk-gluon/gluon/files/2545300/firmware-2.bin.gz) (sha256sum: 4112d78161a2c20e7a9490ea8d04e9f919508bafb097d88761dcce026c26f93f). I have copied this file from the AC7 with 2017.1.8 mentioned above ("5ghztestnode") from `/lib/firmware/ath10k/QCA988X/hw2.0/firmware-2.bin`.

On the 2018.1.1 device I have copied the firmware-2.bin to `/lib/firmware/ath10k/QCA988X/hw2.0/firmware-2.bin` and also renamed the existing `/lib/firmware/ath10k/QCA988X/hw2.0/firmware-5.bin` file (to `firmware-5.bin.disabled`) so that it is not detected by the driver.

mweinelt, 2018-11-03:
So did Candelatech drop/break IBSS support? They still advertise it for both versions of their firmware.

oliver, 2018-11-03:
I think the firmware in 2018.1.1 is not from Candelatech but is the original Qualcomm FW.

mweinelt, 2018-11-03:
See the conditional packaging here: https://github.com/freifunk-gluon/gluon/blob/v2018.1.x/targets/ar71xx-generic#L5-L8

oliver, 2018-11-03:
Used ath10k firmwares:
* 2017.1.1 has 10.1.467-ct-_fW-019-60dd7c5 (no idea where this FW originates from exactly; from a superficial look at the files at https://www.candelatech.com/ath10k-10.1.php in the 10.1.467-ct-19  section I haven't found it)
* 2018.1.1 has 10.2.4-1.0-00016 (which is identical to the one from https://github.com/kvalo/ath10k-firmware/blob/master/QCA988X/hw2.0/10.2.4-1.0/firmware-5.bin_10.2.4-1.0-00016, ie. original Qualcomm FW)

The ath10k driver appears to be same version in both Gluon releases, judging from these dmesg lines:
```
Loading modules backported from Linux version wt-2017-01-31-0-ge882dff19e7f
Backport generated by backports.git backports-20160324-13-g24da7d3c
ath10k driver, optimized for CT firmware, probing pci device: 0x3c.
```

oliver, 2018-11-03:
@mweinelt: nice detective work! Your suggested commands fixed it:

```
opkg update
opkg remove ath10k-firmware-qca988x
opkg install ath10k-firmware-qca988x-ct
```

oliver, 2018-11-03:
@mweinelt: we use IBSS in Bremen, so we would probably use `GLUON_WLAN_MESH ?= ibss`? Or did I misunderstand the GLUON_WLAN_MESH option?

mweinelt, 2018-11-03:
Exactly, but please remember to migrate to 11s, for example after v2018.2 is released, which probably contains #1555.

oliver, 2018-11-03:
@mweinelt: thanks for your help!

closed 2018-11-04 by rotanid

genofire, 2018-11-05:
@mweinelt wait #1555 is only for domain change - so i could change configuration in `site.conf` but `GLUON_WLAN_MESH` is in `site.mk` and is used during build - how should this be possible?

mweinelt, 2018-11-05:
Ah funny, there was a time when enabling both IBSS and 11s at the same time was possible in the same firmware. Not everything will work, like 5 GHz on QCA988x radios, but I think with `GLUON_WLAN_MESH=11s` one domain with IBSS and the other with 802.11s could still be possible.

@blocktrron Do you agree?

blocktrron, 2018-11-05:
This is the case for ar71xx indeed as ath9k supports both protos simultaneously.

This is not the case for ipq40xx as it uses ath10k on both frequency bands.

PR #1562 (2018-11-04): docs: releases/v2018.1: add remark about GLUON_WLAN_MESH variable
fixes #1561
commit e56acfbe (2018-11-04): docs: releases/v2018.1: add remark about GLUON_WLAN_MESH variable

fixes #1561
commit 1649358d (2018-11-10): docs: releases/v2018.1: add remark about GLUON_WLAN_MESH variable

fixes #1561