freifunk-gluon/gluon #1569: Support for "make dist"

yanosz, 2018-11-15:
Hello,

is:

* Gluon Builds are done using `make world` with some parameter are specified.
* Doing a gluon build is hardly done calling make world only one time
  * Different architectures require sucessive calls
  * Different side-configuration require sucessive calls (esp. for minor-changes - i.e. batman-adv-Version), that cannot be implemented in a multidomain setup

As an effect, different communities are creating different, elaborate build configurations.

Should:

* Gluon should support an additional target "make dist"
  * This target should build all available platforms (selecting explicit ones should be possible though)
  * This target should accept multiple site-configurations. **Validation** should be done **before** anything else (esp. sth. with  effort) is done
  * The target should create a folder called dist include the full tree of all images and packages # @

christf, 2018-11-23:
I think it would be helpful to standardize the builds on a gluon-project level instead of community level. Another idea was to specify a Jenkinsfile and have the parameters in place to customize.

For the make dist approach - how would parameters be stored?

yanosz, 2019-04-25:
Ok, thinking about this for a while, I think, that theses features are missing in Gluons Makefile
- [ ] Build more than one platform
- [ ] Use Envirnomental variables (i.e. yielding envsubst) for things, that cannot be hardcoded in the config.
- [ ] Validations before build

If `make dist` boils down to that, a jenkins job should be easy.

neocturne, 2025-04-04:
Hmm, I've been thinking about this for a bit, and while it would of course be possible to add an "all-targets" Make target, I don't think there are any usecases for it. Locally, I would never want to build all targets; in CI, you want a separate job for each target.

We are currently using Github's CI because it provides us with free runners. I can see a few things that could be improved here:
- Modularize Github Actions to make it easier for communities to reuse parts
- Implement templates for alternative CI that can be set up on owned infrastructure (Jenkins, Gitlab, ...)