freifunk-gluon/gluon #1572: [RFC] Funding of Gluon developers?

CodeFetch, 2018-11-17:
Maybe I'm pessimistic, but I have the feeling that Gluon development and the widespread usage and demand for new features is reaching a point where some of the next steps require a degree of commitment that can't be easily achieved by volunteers only. 
One option could be to ally with other mesh firmware developers e.g. from LibreMesh, but that deserves an own issue.
Another option could be to support projects financially that aim to develop features or fix bugs for which we have a common demand.
There a grants for projects like Gluon as it can have a huge impact on network infrastructure in development countries in the future and already has e.g. in German Freifunk communities, which we don't make use of at the moment.
The question of fairness must of course be considered and if Gluon chooses this path, it is beyond dispute that there will be people that feel treated unfair. But at the moment every Gluon developer sacrifices his or her free time and even financial means just for the cause and often does not receive the appreciation deserved. 
If we look at the history of linux, opening the gates to companies not only accelerated the development by nth degrees, spreaded its use, but also opened up new fields of employment.

My question is simple, but requires consensus: Do you think we can agree on projects that can receive financial support in the future, which would otherwise not be realised or do you think it's better to accept a possible slow down due to increasing complexity for keeping independence?

rotanid, 2018-11-17:
as there still is an open project which is connected with a monetary bounty - and no one is reaching for that money - i don't see how more money should help?
even if we would welcome your suggestion and we find someone paying more money: the paid projects will always be the difficult ones, that are stuck at the moment, not the easy ones. and currently there IS an open (difficult) project waiting for someone to solve it.
and, even if [...]: i don't think we can pay for workhours like a normal company, it would always be in the form of a bounty for solving something...

CodeFetch, 2018-11-17:
@rotanid I totally agree with that. "Fixing" SquashFS is a really hard nut, but there are other probably easier tasks. Following come to my mind spontaneously:
- WiFi fallback updater
- WireGuard L2/fastd in kernel space
- N2N VPN
- Migration to ath79
- Support and testing of MT7620 devices
- Writing a tool for easier flashing of Xiaomi routers
- Better mesh support for other Mediatek chipsets (nbd would do that for money)
- (harder and not that critical atm) MCCA

rotanid, 2018-11-17:
projects that don't benefit Gluon alone would be better suited i guess. (the squashfs stuff and the ftrace is one of those, also WG, fastd, ath79, chipset/driver stuff from your list)
but i think this does not necessarily need approval by Gluon, someone could already organize something like this without involving Gluon-Devs - so the issue probably is rather, that you need someone organizing something like this, 2nd finding people paying for this and not the approval by someone which you might think is an authority.
am i right again? ;)

CodeFetch, 2018-11-17:
@rotanid Yes, it needs the knowledge and involvement of Gluon devs to evaluate and organise this.

christf, 2019-02-27:
We already have an implementation for the n2n vpn. I just chose not to raise it as a PR because it is for fastd only and I now find wireguard more interesting as it has received more review and is faster.

mweinelt, 2019-02-27:
> And quite frankly I think instead of reworking towards gluon web I'd rather get rid of the frontends and status page altogether.

What?! Where is this coming from?

CodeFetch, 2019-05-03:
I close it as I feel that funding projects to bring Gluon forward seems to be accepted. If anyone has a concrete idea what he or she would like to do, something that is hard to implement without financial help, please tell me and I try to get some funding for it.

closed 2019-05-03 by CodeFetch