freifunk-gluon/gluon #1578: ER-X broken with Gluon 2018.1.2

kpanic23, 2018-11-23:
Updating an UBNT EdgeRouter X to Gluon v2018.1.2 reproducibly results in a brick.
The EdgeRouter X SFP seems to not be affected.
Further investigation is needed.

heini66, 2018-11-23:
root@VIN-FFNH-ERX-01:~# batctl n
Error - interface bat0 is not present or not a batman-adv interface
batman-adv module isn't loading on er-x. er-x-sfp working...
https://pastebin.com/niMzkU08
https://pastebin.com/93RPXDss

ecsv, 2018-11-23:
Please test #1580. And make sure that you really rebuild everything

rubo77, 2018-11-24:
On the top you see the repository and branch FreifunkVogtland:batadv-kmod-deps/v2018.1.x

Checkout that gluon branch and build again

collimas, 2018-11-24:
Ah, ok. I'll try and report.

collimas, 2018-11-24:
Everything is working fine now. Seems there will be a 2018.1.3 bugfix release in the near future. Thanks a lot!

rotanid, 2018-11-24:
fixed with 8647b782f758679be7287ffdfc4579a5e0b868b8 #1580

closed 2018-11-24 by rotanid