freifunk-gluon/gluon #1584: 5GHz Mesh doesn't work with 2018.1

oliver, 2018-11-25:
After the general 5GHz wifi problems with AC7 and 2018.1.1+bremen1 firmware were solved (in #1561), we found out that _meshing_ on 5GHz still does not work.

This affects all FFHB 2018.1 firmwares so far, and has (only) been tested on Archer C7 v2 so far.

Symptoms are that a 5GHz-only device (eg. CPE510) does not create a working wifi mesh connection to the AC7.

Example device: http://og-ac7-testing-2.nodes.ffhb.de (https://map.ffhb.de/#!/en/map/f4f26d70b277).
The AC7 is in the same room as a CPE510 (https://map.ffhb.de/#!/en/map/c4e984b0a84a) and a WDR3600 (https://map.ffhb.de/#!/en/map/c4e984d5138e), both with 2017.1.8+bremen1 (ie. stable) firmware.
The Stable-FW devices are meshing nicely. The AC7 is meshing only via 2.4GHz. As result, the WDR3600 meshes with both other devices while the CPE510 meshes only with WDR3600.

Interestingly the devices appear to see each other via IBSS, but still don't get a mesh connection working:
```
root@cpe510-og-1:~# iw dev ibss0 station dump
Station aa:85:d3:37:f0:5e (on ibss0)
	inactive time:	0 ms
	rx bytes:	167231521
	rx packets:	1267310
	tx bytes:	15690420
	tx packets:	65650
	tx retries:	2531
	tx failed:	0
	rx drop misc:	0
	signal:  	-58 [-60, -62] dBm
	signal avg:	-59 [-61, -63] dBm
	tx bitrate:	144.4 MBit/s MCS 15 short GI
	rx bitrate:	144.4 MBit/s MCS 15 short GI
	expected throughput:	46.875Mbps
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	0
	beacon interval:100
	short slot time:yes
	connected time:	5629 seconds
Station 0e:9e:42:44:91:aa (on ibss0)
	inactive time:	30 ms
	rx bytes:	73632280
	rx packets:	567327
	tx bytes:	0
	tx packets:	0
	tx retries:	0
	tx failed:	0
	rx drop misc:	0
	signal:  	-61 [-63, -65] dBm
	signal avg:	-60 [-63, -63] dBm
	tx bitrate:	6.0 MBit/s
	authorized:	yes
	authenticated:	yes
	associated:	yes
	preamble:	long
	WMM/WME:	yes
	MFP:		no
	TDLS peer:	no
	DTIM period:	0
	beacon interval:100
	short slot time:yes
	connected time:	2544 seconds
```

Nothing strange in `batctl if`:
```
root@og-ac7-testing-2:~# batctl if
mesh-vpn: active
ibss1: active
ibss0: active
primary0: active
```

I'm AFK now for a while; will post dmesg etc. later, but there's not much to see there. What info would be useful to debug this?

mweinelt, 2018-11-25:
Please post the output of `batctl n`

oliver, 2018-11-25:
```
root@og-ac7-testing-2:~# batctl n
Error - no valid command or debug table specified: n
[...]
root@og-ac7-testing-2:~# batctl -v
batctl 2013.4.0 [batman-adv: 2013.4.0]
```

oliver, 2018-11-25:
Various command results of an AC7 v2 with 2018.1.1+bremen2:
* `dmesg`: [dmesg.og-ac7-testing-2.txt](https://github.com/freifunk-gluon/gluon/files/2613343/dmesg.og-ac7-testing-2.txt)
* `iw dev`: [iw-dev.og-ac7-testing-2.txt](https://github.com/freifunk-gluon/gluon/files/2613352/iw-dev.og-ac7-testing-2.txt)
* `iw list`: [iw-list.og-ac7-testing-2.txt](https://github.com/freifunk-gluon/gluon/files/2613345/iw-list.og-ac7-testing-2.txt)

Command results of another AC7 v2 with 2018.1.1+bremen2:
* `dmesg`: [dmesg-2017.1.8.5ghztestnode.txt](https://github.com/freifunk-gluon/gluon/files/2613347/dmesg-2017.1.8.5ghztestnode.txt)
* `iw dev`: [iw-dev.2017.1.8.5ghztestnode.txt](https://github.com/freifunk-gluon/gluon/files/2613353/iw-dev.2017.1.8.5ghztestnode.txt)
* `iw list` is exactly identical between 2018.1.1+bremen2 and 2017.1.8+bremen1

oliver, 2018-11-25:
```
root@og-ac7-testing-2:~# modinfo batman-adv
module:		/lib/modules/4.4.153/batman-adv.ko
version:	2013.4.0
description:	B.A.T.M.A.N. advanced
author:		Marek Lindner <lindner_marek@yahoo.de>, Simon Wunderlich <siwu@hrz.tu-chemnitz.de>
license:	GPL
depends:	
```
```
root@og-ac7-testing-2:~# lsmod | grep batman
batman_adv            106595  0 
```
```
root@og-ac7-testing-2:~# lsmod | wc -l
18
```

On 2018.1.1+bremen2 there are 18 modules loaded; on 2017.1.8+bremen1 there are 117 modules loaded! This is probably related to #1580, but I don't know whether it actually causes these problems.

Anyway, here's `lsmod` output for old and new firmware:
* [lsmod.2017.1.8.5ghztestnode.txt](https://github.com/freifunk-gluon/gluon/files/2613359/lsmod.2017.1.8.5ghztestnode.txt)
* [lsmod.og-ac7-testing-2.txt](https://github.com/freifunk-gluon/gluon/files/2613360/lsmod.og-ac7-testing-2.txt)

mweinelt, 2018-11-25:
> On 2018.1.1+bremen2 there are 18 modules loaded; on 2017.1.8+bremen1 there are 117 modules loaded! This is probably related to #1580, but I don't know whether it actually causes these problems.

Sounds like it. Please retry with the latest v2018.1.x commit and report back if that fixes the issue.

NeoRaider, 2018-11-25:
The low number of loaded module is expected, we started building as much as possible into the kernel.

NeoRaider, 2018-11-30:
To check whether the issue is in ath10k or in batadv, try if a simple ping (over IPv6 link-local) on ibss0 works. 

Unfortunately, the logs don't show anything unusual. Does the issue only occur between ath9k and ath10k devices, or is it reproducible with two ath10k devices as well?

NeoRaider, 2018-11-30:
One more thing to test: Since v2018.1, we set the htmode of 11ac devices to 'VHT20' in `/etc/config/wireless`, in older versions, it was always set to 'HT20'.

blocktrron, 2018-11-30:
`rx_bitrate` is missing on `iw dev ibss0 station dump`, thy may be a hint.

oliver, 2018-12-07:
(sorry for the silence on my side, I'm not online a lot at the moment and will probably get to this issue in 2019. Thanks for the comments, I will try these hints)

rotanid, 2018-12-07:
@oliver we will close this ticket now. feel free to reopen as soon as you can provider additional information like requested

blocktrron, 2018-12-09:
@oszilloskop Can you link your binary firmware files?

oszilloskop, 2018-12-09:
@blocktrron 
You will find the binary files here: https://dl.ffm.freifunk.net/firmware/test/

rotanid, 2018-12-10:
@blocktrron he pointed out on IRC, that it also doesn't work with v2017.1.7 - this does not contain the mentioned patch.
interestingly though, v2017.1.8 does work for Freifunk Bremen if i understood correctly what @oliver wrote.
the low-hanging fruit, err, conclusion: one of both tests went wrong or it's a different issue.
i think(!) v2017.1.x does not have a general issue with 5 GHZ ibss mesh as this would have hopefully been reported last year - and it works for FF Bremen as far as i understand.

oliver, 2018-12-19:
Short update: ping via ibss0 (5 GHZ IBSS) shows some weirdness on 2018.1.3: `ping6 ff02::1%ibss0` returns only the local IPv6 address but no replies from other nodes. But doing the same via ibss1 shows replies from the other node that is connected via 2.4 GHz.

But if I _directly_ ping the link-local IPv6 of another node via ibss0 (ie. not via broadcast address), I do get replies. And afterwards I will also get a reply from that node with `ping ff02::1%ibss0`; and `iw dev ibss0 station dump` now actually shows an "rx bitrate:" for the connection to this node. And on the Gluon status page the name of the remote node will now also appear (before, there was only the MAC shown).

I don't know enough about wifi IPv6 broadcasting to understand what's the cause or effect here. So maybe the broadcast ping is indeed broken and this causes the Batman problem; or maybe Batman (or something else) is broken, and that also causes the IBSS connection to stay in some "idle" state where broadcast pings don't work.

But at least this shows that in general some data can be transferred over 5 GHz IBSS with 2018.1.

NeoRaider, 2018-12-20:
Might be a power save issue. You can try disabling it using `iw dev <dev> set power_save off`.

NeoRaider, 2018-12-20:
Please also try the Gluon master, which is based on OpenWrt 18.06.

oliver, 2018-12-20:
I've just installed 2018.1.3+bremen1 on a CPE510 v1 which is a 5GHz-only device, and mesh works fine (see https://map.ffhb.de/#!/en/map/c4e984b0a84a and http://cpe510-og-1.nodes.ffhb.de/). The device successfully meshes with another node. The CPE510 uses the ath9k driver.

So this problem doesn't affect all devices. So far the ath10k driver with -ct firmware shows the problem, while the ath9k driver works.

oliver, 2018-12-20:
@oszilloskop can you check which driver is used on the devices which don't work for you? What does `lsmod | grep ath` show?

oliver, 2018-12-22:
> @blocktrron he pointed out on IRC, that it also doesn't work with v2017.1.7 - this does not contain the mentioned patch.
> interestingly though, v2017.1.8 does work for Freifunk Bremen if i understood [...]

@rotanid: good thing you mentioned this! I just did a test of this specific functionality, and it looks like 5 GHz meshing on AC7 was already broken with 2017.1.8+bremen1 :-( So no, v2017.1.8 does _not_ really work for FFHB, and v2017.1.x _does_ have a general issue with 5 GHZ ibss mesh.

Or maybe my test is wrong. I have set up three devices with 2017.1.8+bremen1, all located in the same room:
* AC7v4 (https://map.ffhb.de/#!/en/map/704f57119260), dualband
* WDR3600 (https://map.ffhb.de/#!/en/map/c4e984d5138e), dualband
* CPE510 (https://map.ffhb.de/#!/en/map/c4e984b0cb42), 5GHz only

Result: the WDR3600 meshes with both devices. The AC7 and the CPE510 only mesh with the WDR3600. To me this indicates that the 5GHz mesh on AC7 doesn't work.

I guess we never systematically tested this, and the problem was never really noticed probably because the devices are still meshing via 2.4 GHz.

Anyway, I'll report back when I've found out more.

rotanid, 2018-12-28:
as discussed in person, please try v2016.2.x based firmware (with a device like Archer C7 v2) and also try current master (or, the same, v2018.2.x as soon as it is released)

rotanid, 2019-01-02:
thanks. now the last missing bit of information is if an ath10k devices works with v2016.2.x in IBSS 5 GHz mesh

mortzu, 2019-01-07:
I have the same issue on mANTBox 15s (ath10k). The mesh works in the moment I disabled the client wireless device.

oszilloskop, 2019-01-07:
@mortzu 
Nice founding. 
I can confirm that this workaround works fine with my two UniFi-AC-MESH (dual 2.4/5GHz, ath9k/ath10k).

> So this problem doesn't affect all devices. So far the ath10k driver with -ct firmware shows the problem, while the ath9k driver works.

Furthermore  I can confirm that my ath9k only device (Ubiquiti Loco M5 XW) does not have a 5GHz mesh problems. Now with correct working mesh partners, it meshed very well without that workaround.

mweinelt, 2019-01-07:
This essentially means that ath10k with the candelatech driver/firmware has become unusable for Gluon, as it does not support ap+ibss at the same time.

oszilloskop, 2019-01-07:
Oh sorry. My above tests were done with Gluon 2018.2.

rotanid, 2019-01-07:
@mweinelt "become" ? according to the tests done by @oliver the situation didnt change in any way - maybe that's nitpicking, but i think that's a difference.

blocktrron, 2019-01-07:
@oszilloskop 
This is expected behavior. ath9k is by far more open as ath10k. If the ath10k firmware does not support the service-set combination we need, we are - simply put - out of luck.

Maybe you can try an mt76 based device, apparently this driver sports IBSS support for 11ac (as it's "comparably libre" to ath9k), but it is flagged as broken for IBSS as it is untested.

oszilloskop, 2019-01-08:
I only affirmation the test results which were already reported by @oliver and @mortzu.
In my case, the result "not at the same time" is very important. ath10k can do ibss mesh, but not simultaneously with client network. This is different for me than "it does not work".

rotanid, 2019-01-08:
i adjusted the title of this issue accordingly.

CodeFetch, 2019-06-16:
IBSS support will be dropped as described in #1747. This issue will likely not be resolved in Gluon. The issue I've opened in ath10-ct still remains open. Thus it might get fixed at some point, but there won't be a Gluon-release with this fix. 

Thus I think this can be closed now.

rotanid, 2019-06-18:
@CodeFetch for now it's a known issue in the current and next Gluon release.

CodeFetch, 2019-06-18:
Okay. I thought that it won't get fixed is a reason for closing it as "won't fix".

mweinelt, 2019-06-18:
"Won't fix" is a reasonable assumption, so let me tag it like this.

mweinelt, 2019-09-25:
With the deprecation of IBSS meshing in the v2019.2 release cycle this issue is out of scope for Gluon.

Please migrate to 802.11s in a timely fashion.

closed 2019-09-25 by mweinelt