freifunk-gluon/gluon #1593: Deutsche Dokomentation

yanosz, 2018-12-01:
Is:
* Gluon's documentation is in English. 

Should:
* Gluon's documentation is in German, too

Ideas & Remarks:
* Every now and then I encounter people not speaking English. Gluon's doc is a barrier here, while Freifunk is focussing on transporting knowledge.
* For doing translations it is good to have seperated tracking process. Usually, people doing translations are different from people implementing stuff. It'd helpful to have nice workflow. E.g. 
  * Overview: What sections are outdated. 
  * (Ticket?)-based tracking integrated into the release process.
* Some parts are more revelant than others. I.e. deep dives into development are less relevant - Getting started is very relevant
* Lots of Freifunk-Communities have German docs for Gluon. Joint documentation efforts 
* IMHO doing translations is an easy way for people to join gluon. It should be possible to have a broad base.

CodeFetch, 2018-12-01:
Translate it and make a pull-request, but don't expect Gluon developers to keep the German translation topical.

rubo77, 2018-12-01:
Wie müsste man die deutschen Dateien in die Sphinx Ordner Struktur einfügen?

CodeFetch, 2018-12-02:
Someone who understands Gluon enough to translate the documentation should be able to handle po files. You don't want to reinvent the wheel. Readthedocs supports localization https://docs.readthedocs.io/en/latest/localization.html and so does the Github Wiki engine https://github.com/JustArchiNET/ArchiSteamFarm/wiki albeit both are more kind of a workaround.

NeoRaider, 2018-12-02:
In the last Mumble meetings we reinforced our decision that English is the official project language - the docs are in English, and English is the preferred language on the mailing lists and in our IRC channel. We would like contributors outside of Germany to be able to participate as easily as possible.

If someone creates a translation, we can link to it, but it should be clear that it is only a translation, and the English version is the only official documentation.

rubo77, 2018-12-02:
https://translate.googleusercontent.com/translate_c?depth=1&hl=de&nv=1&rurl=translate.google.com&sl=auto&sp=nmt4&tl=de&u=https://gluon.readthedocs.io/en/v2018.1.x/&xid=17259,15700023,15700122,15700124,15700149,15700186,15700190,15700201,15700237&usg=ALkJrhj3mhiVaLoiXxtP6PKOSkp2gvJM-Q

😉

kevin-olbrich, 2018-12-02:
I don't think, that additional doc langs would help in any topic. To debug Gluon / LEDE / OpenWRT, english is mandatory. Also, most vocabulary / technical terms like server, router, etc. are i18n and don't need to be translated.

Other langs are fine for how-to's or installation guides for the enduser and that is something that local communities should be providing.

christf, 2018-12-02:
On Sun, Dec 02, 2018 at 01:48:32PM -0800, Kevin Olbrich wrote:
> I don't think, that additional doc langs would help in any topic. To debug Gluon / LEDE / OpenWRT, english is mandatory. Also, most vocabulary / technical terms like server, router, etc. are i18n [...]
What is the quality of the result when applying deepl.com to a given 
gluon docs page to translate the content into german? 

Is it usable? 
Can we ask our user base to do this?

If the answer to the first question is yes and the second is no, maybe a 
bot can do the translation and keep it up to date?
-- 
()  ascii ribbon campaign - against html e-mail
/\  against proprietary attachments

bobcanthelpyou, 2018-12-02:
> If the answer to the first question is yes and the second is no, maybe a bot can do the translation and keep it up to date?

A user could use deepl on-demand if a translation is needed, then the translation is up2date, too.

By the way, is a translation by a service like deepl free to use and is it allowed to create public documents with it?

rubo77, 2018-12-22:
I think it is a great idea to provide a German translation...

My suggestion: clone this repo and translate it. When it's done, we can link to it on the starting page of the English Doko

yanosz, 2018-12-27:
@CodeFetch, well, "Gluon is not Freifunk" is a very difficult issue, that is completly off-topic. I'm not interested in discussion this, here.

Libremesh is already providing an Italian documentation for onboarding. Actually, this it what I'm thinking of. By that, translation is already an issue when thinking about a merger. Changing the title is a good idea in that context.

Please mind, that German docs on Gluon are not Freifunk specific, unless they provide insights different from their English counter-part. Nobody has proposed that so far. I don't understand your concern.

kaechele, 2019-01-16:
It's actually pretty easy to do:

`cd docs
make -e SPHINXOPTS="-D gettext_compact=0" gettext
sphinx-intl update -p _build/locale/ -l de`
Edit the *.po files in locales/de/* using your favorite PO-file editor (e.g. poEdit).
When done:
`make -e SPHINXOPTS="-D language='de'" html`

Feel free to take a stab at it.

Having the translations in the main repo is probably not feasible, as they will be outdated and unmaintained in 99% of the cases.
So keeping the translated files in a separate repo and maybe even using a service like Transifex to enable easier translations by the community would be advisable.

CodeFetch, 2019-01-17:
@kaechele I don't have a clue how to integrate this into ReadTheDocs. As you seem to understand it, can you please have a look at how to do it and maybe how to use e.g. Weblate or something else that has tight git integration?

kaechele, 2019-01-17:
Definitely willing to help here. But I guess that would require me having access to the ReadTheDocs settings.
Weblate or Pootle I can set up as well, but we'd need to discuss whether we want to manually sync in the translation updates or someone does this manually on a regular basis.

kaechele, 2019-01-17:
I set up this for testing: https://translate.kaechele.ca
Don't invest too much time there, as this is my personal test instance and data may go away or things may break without notice.
But please have a look if this is something that we would want to be doing.
Sphinx makes it a bit convoluted by breaking out each section of the docs into a single translatable file.
I'm now looking into piecing together everything to commit that back to the git repository.

CodeFetch, 2019-02-19:
Thank you very much! I like it. Weblate has a Github App for CI: https://github.com/apps/hosted-weblate
Can we use this somehow?

kaechele, 2019-02-19:
Yes, weblate supports github webhooks natively even when not using their hosted product.
To set it up you'd need administrator access to the repository settings though.

mweinelt, 2019-02-20:
I don't have an issue with setting up webhooks for certain events for this project.

mweinelt, 2022-03-26:
Closing as stale, I don't think we have enough contributions to make this happen.

closed 2022-03-26 by mweinelt