freifunk-gluon/gluon #1596: add squashfs images for x86

txt-file, 2018-12-03:
I would like to have x86 images with failsafe mode.

NeoRaider, 2018-12-03:
Does sysupgrade between ext4 and squashfs work smoothly? If that is the case, we can just replace our ext4 images with squashfs. I don't think we should provide two different images for the same targets.

mweinelt, 2019-01-14:
@txt-file Can you test the sysupgrade between ext4 and squahfs and report back?

txt-file, 2019-01-14:
I tried it with stock OpenWrt 18.06.1 in both directions without preserving the config. It worked and the device booted after flashing via sysupgrade.

rotanid, 2019-01-14:
i'm pretty sure that the question was about keeping/preserving config, otherwise it wouldnt make sense to remove an image type...

txt-file, 2019-01-14:
1. Nobody explicitly said that I should test to preserve the config.
2. I also think that without config preservation the upgrade is useless. This is why I explicitly said that I did not test this. Today I first wanted to make sure that changing the filesystem is possible.

I didn't manage to test all the thing but wanted to share my test results. My plan is to test config preservation under OpenWrt and gluon next Monday.

txt-file, 2019-01-21:
```
BusyBox v1.28.4 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 18.06-SNAPSHOT, r7653+12-939fa07b04
 -----------------------------------------------------
root@ffc-halloTest:~# mount
/dev/root on / type ext4 (rw,noatime,block_validity,delalloc,barrier,user_xattr)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/sda1 on /boot type ext4 (rw,noatime,block_validity,delalloc,barrier,user_xattr)
/dev/sda1 on /boot type ext4 (rw,noatime,block_validity,delalloc,barrier,user_xattr)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
root@ffc-halloTest:~# cat /lib/gluon/gluon-version
v2018.2-11-g692f61ec+
root@ffc-halloTest:~# cat /lib/gluon/release
b20190121-multi
root@ffc-halloTest:~# 
[…]
root@ffc-halloTest:/tmp# sysupgrade gluon-ffc-b20190121-multi-x86-generic.img 
Image metadata not found
Reading partition table from bootdisk...
Reading partition table from image...
Saving config files...
Commencing upgrade. Closing all shell sessions.
```
reboot
```
BusyBox v1.28.4 () built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 18.06-SNAPSHOT, r7653+12-939fa07b04
 -----------------------------------------------------
root@ffc-halloTest:~# mount
/dev/root on /rom type squashfs (ro,relatime)
proc on /proc type proc (rw,nosuid,nodev,noexec,noatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,noatime)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/loop0 on /overlay type f2fs (rw,lazytime,noatime,background_gc=on,no_heap,user_xattr,inline_xattr,inline_data,inline_dentry,flush_merge,extent_cache,mode=adaptive,active_logs=6)
overlayfs:/overlay on / type overlay (rw,noatime,lowerdir=/,upperdir=/overlay/upper,workdir=/overlay/work)
/dev/sda1 on /boot type ext4 (rw,noatime,block_validity,delalloc,barrier,user_xattr)
/dev/sda1 on /boot type ext4 (rw,noatime,block_validity,delalloc,barrier,user_xattr)
tmpfs on /dev type tmpfs (rw,nosuid,relatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,relatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,noatime)
root@ffc-halloTest:~#
```

rotanid, 2019-01-21:
@txt-file looks good, maybe add at least one sentence to your comment via the edit function saying what you have done and what the result was

PR #1634 (2019-02-03): target x86-*: change from ext4 to squashfs
This allows for smaller firmware images and makes it possible to reset a
device via firstboot command.

closes #1596
closed 2019-02-03 by NeoRaider