freifunk-gluon/gluon #1600: Firmware version 1.7.0 not considered newer than 1.7.0~rc1

RalfJung, 2018-12-18:
It used to be the case that (following Debian's handling of version numbers) firmware version 1.7.0 was considered newer than 1.7.0~rc1.  However, with the latest gluon 2018.1.3, that does not seem to work any more: We have rolled out 1.7.0~rc1 on our beta branch, and now we built 1.7.0, but even when doing a force update check the autoupdater just says "No new firmware available."

mweinelt, 2018-12-18:
The expectation is that `anything` is greater than `anything~` and that holds true for dpkg. Like this:

```
# 1.7.0 greater than 1.7.0~rc1
$ dpkg --compare-versions 1.7.0 gt 1.7.0~rc1; echo $?
0
# 1.7.0 not less than 1.7.0~rc1
$ dpkg --compare-versions 1.7.0 lt 1.7.0~rc1; echo $?
1
# 1.7.0 not equal 1.7.0~rc1
$ dpkg --compare-versions 1.7.0 eq 1.7.0~rc1; echo $?
1
```

The code was last changed when the new autoupdater implementation was merged. See  https://github.com/freifunk-gluon/packages/commit/49cb4b3fdb8b9db0abb81783b9b3b386a8ed6a92.

RalfJung, 2018-12-18:
> The expectation is that anything is greater than anything~ and that holds true for dpkg. Like this:

Exactly.  But somehow it doesn't seem to work any more in 2018.1.3 (the beta 1.7.0~rc1 and the stable 1.7.0 were both built from that version).

With `--force-version`, it is happy to install the new version:
```
# autoupdater -f -b experimental
Retrieving manifest from http://mgmt.saar.freifunk.net/firmware/experimental/sysupgrade/experimental.manifest ...
No new firmware available.
# autoupdater -f -b experimental --force-version
Retrieving manifest from http://mgmt.ffsaar/firmware/experimental/sysupgrade/experimental.manifest ...
Stopping cron...
Stopping haveged...
Stopping micrond...
Stopping sysntpd...
Stopping gluon-radvd...
Stopping uhttpd...
```

RalfJung, 2018-12-18:
> The expectation is that anything is greater than anything~ and that holds true for dpkg. Like this:

Exactly.  But somehow it doesn't seem to work any more in 2018.1.3 (the beta 1.7.0~rc1 and the stable 1.7.0 were both built from that version).

With `--force-version`, it is happy to install the new version:
```
# autoupdater -f -b experimental
Retrieving manifest from http://mgmt.saar.freifunk.net/firmware/experimental/sysupgrade/experimental.manifest ...
No new firmware available.
# autoupdater -f -b experimental --force-version
Retrieving manifest from http://mgmt.ffsaar/firmware/experimental/sysupgrade/experimental.manifest ...
Stopping cron...
Stopping haveged...
Stopping micrond...
Stopping sysntpd...
Stopping gluon-radvd...
Stopping uhttpd...
```

And it's not just one node, I noticed this because a user reported a problem on several of their nodes, and I can see the same on mine.