freifunk-gluon/gluon #1610: UniFi-AC-Mesh-Pro: Mixed up Port assignment

h03e, 2019-01-02:
This device seems to have mixed up WAN (labled MAIN) and LAN (labled SECONDARY) Ports. Mesh-VPN and Config-Mode are on the SECONDARY Port while PoE ist on MAIN. The Unifi AC Pro is configured the other way around. Short you need two cables to use it properly.
Also the detected MAC (78:8a:20:b0:ec:d9) is different by one digit compared to the lable (78:8a:20:b2:ec:d9).
I'd like to provide more info and can't find a solution myself.

NeoRaider, 2019-01-02:
Please provide the contents of `/tmp/sysinfo/board_name` and `/etc/config/network`.

@achterin Can you have a look at this?

h03e, 2019-01-02:
> /tmp/sysinfo/board_name

`unifiac-pro`

> /etc/config/network

```config interface 'loopback'
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fde1:26a1:335e::/48'

config interface 'wan6'
	option proto 'dhcpv6'
	option ifname 'br-wan'
	option ip6table '1'
	option sourcefilter '0'
	option peerdns '0'
	option reqprefix 'no'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'```

heini66, 2019-01-02:
ac-pro, same ...

achterin, 2019-01-02:
I've seen this with one out of six devices as well, but don't have access to it right now. I haven't found a reason why this occurred only on one device.
I'll have a look at the MAC problem asap.

blocktrron, 2019-01-03:
@h03e your outlook looks incomplete, could you check that?

heini66, 2019-01-03:
config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fdc1:def4:575f::/48'

config interface 'wan6'
	option proto 'dhcpv6'
	option ifname 'br-wan'
	option ip6table '1'
	option sourcefilter '0'
	option peerdns '0'
	option reqprefix 'no'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '2 0t'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '3 0t'

config interface 'wan'
	option igmp_snooping '1'
	option ifname 'eth0.1'
	option multicast_querier '0'
	option peerdns '0'
	option auto '1'
	option type 'bridge'
	option proto 'dhcp'
	option macaddr 'ea:4a:26:bf:fd:d0'

config rule6 'wan6_lookup'
	option mark '0x01/0x01'
	option lookup '1'

config route6 'wan6_unreachable'
	option type 'unreachable'
	option table '1'
	option target '::/0'
	option metric '65535'
	option gateway '::'
	option interface 'loopback'

config interface 'mesh_radio0'
	option proto 'gluon_mesh'

config interface 'mesh_radio1'
	option proto 'gluon_mesh'

config interface 'mesh_wan'
	option ifname 'br-wan'
	option index '0'
	option proto 'gluon_wired'
	option disabled '1'
	option transitive '1'

config interface 'mesh_lan'
	option igmp_snooping '0'
	option ifname 'eth0.2'
	option index '4'
	option proto 'gluon_wired'
	option disabled '1'
	option transitive '1'
	option macaddr 'ea:4a:26:bf:fd:d4'

config interface 'client'
	option igmp_snooping '1'
	option type 'bridge'
	option auto '1'
	option ra_holdoff '30'
	option multicast_querier '1'
	option macaddr '78:8a:20:b2:e3:56'
	list ifname 'eth0.2'
	list ifname 'local-port'
	list ifname 'bat0'
	option ipv6 '1'
	option keep_ra_dnslifetime '1'
	option sourcefilter '0'
	option peerdns '0'
	option robustness '9'
	option reqprefix 'no'
	option query_interval '2000'
	option query_response_interval '500'
	option proto 'dhcpv6'

config device 'local_node_dev'
	option type 'veth'
	option name 'local-node'
	option peer_name 'local-port'
	option macaddr '36:f4:54:fc:e5:22'
	option peer_macaddr '78:8a:20:b2:e3:56'

config interface 'local_node'
	option ifname 'local-node'
	option ipaddr '10.71.63.254/18'
	option ip6addr 'fd8f:14c7:d318::ffff/128'
	option ip6deprecated '1'
	option proto 'static'

config interface 'gluon_bat0'
	option proto 'gluon_bat0'

config interface 'bat0'
	option multicast_router '2'
	option ifname 'bat0'
	option auto '1'
	option macaddr '78:8a:20:b2:e3:56'
	option learning '1'
	option proto 'none'

config route6 'local_node_route6'
	option target 'fd8f:14c7:d318::/64'
	option gateway '::'
	option interface 'client'

config interface 'mesh_vpn'
	option ifname 'mesh-vpn'
	option transitive '1'
	option mtu '1374'
	option macaddr 'ea:4a:26:bf:fd:d7'
	option fixed_mtu '1'
	option proto 'gluon_mesh'

h03e, 2019-01-03:
@blocktrron yes, it was incomplete. Sorry
The output of `etc/config/network` looks quite similar to the file from heini66.
```config interface 'loopback'
	option ifname 'lo'
	option proto 'static'
	option ipaddr '127.0.0.1'
	option netmask '255.0.0.0'

config globals 'globals'
	option ula_prefix 'fde1:26a1:335e::/48'

config interface 'wan6'
	option proto 'dhcpv6'
	option ifname 'br-wan'
	option ip6table '1'
	option sourcefilter '0'
	option peerdns '0'
	option reqprefix 'no'

config switch
	option name 'switch0'
	option reset '1'
	option enable_vlan '1'

config switch_vlan
	option device 'switch0'
	option vlan '1'
	option ports '2 0t'

config switch_vlan
	option device 'switch0'
	option vlan '2'
	option ports '3 0t'

config interface 'wan'
	option igmp_snooping '1'
	option ifname 'eth0.1'
	option multicast_querier '0'
	option peerdns '0'
	option auto '1'
	option type 'bridge'
	option proto 'dhcp'
	option macaddr 'de:cc:c7:c4:09:e8'

config rule6 'wan6_lookup'
	option mark '0x01/0x01'
	option lookup '1'

config route6 'wan6_unreachable'
	option type 'unreachable'
	option table '1'
	option target '::/0'
	option metric '65535'
	option gateway '::'
	option interface 'loopback'

config interface 'mesh_radio0'
	option proto 'gluon_mesh'

config interface 'mesh_radio1'
	option proto 'gluon_mesh'

config interface 'mesh_wan'
	option ifname 'br-wan'
	option index '0'
	option proto 'gluon_wired'
	option disabled '1'
	option transitive '1'

config interface 'mesh_lan'
	option igmp_snooping '0'
	option ifname 'eth0.2'
	option index '4'
	option proto 'gluon_wired'
	option disabled '1'
	option transitive '1'
	option macaddr 'de:cc:c7:c4:09:ec'

config interface 'client'
	option igmp_snooping '1'
	option type 'bridge'
	option auto '1'
	option ra_holdoff '30'
	option multicast_querier '1'
	option macaddr '78:8a:20:b2:ec:d9'
	list ifname 'eth0.2'
	list ifname 'local-port'
	list ifname 'bat0'
	option ipv6 '1'
	option keep_ra_dnslifetime '1'
	option sourcefilter '0'
	option peerdns '1'
	option robustness '9'
	option reqprefix 'no'
	option query_interval '2000'
	option query_response_interval '500'
	option proto 'dhcpv6'

config device 'local_node_dev'
	option type 'veth'
	option name 'local-node'
	option peer_name 'local-port'
	option macaddr '16:41:95:40:f7:dc'
	option peer_macaddr '78:8a:20:b2:ec:d9'

config interface 'local_node'
	option ifname 'local-node'
	option ipaddr '10.135.0.1/18'
	option ip6addr 'fd73:111:e824::1/128'
	option ip6deprecated '1'
	option proto 'static'

config interface 'gluon_bat0'
	option proto 'gluon_bat0'

config interface 'bat0'
	option multicast_router '2'
	option ifname 'bat0'
	option auto '1'
	option macaddr '78:8a:20:b2:ec:d9'
	option learning '1'
	option proto 'none'

config route6 'local_node_route6'
	option target 'fd73:111:e824::/64'
	option gateway '::'
	option interface 'client'

config interface 'mesh_vpn'
	option ifname 'mesh-vpn'
	option transitive '1'
	option mtu '1280'
	option macaddr 'de:cc:c7:c4:09:ef'
	option fixed_mtu '1'
	option proto 'gluon_mesh'
```

heini66, 2019-01-03:
we've got several ac-pro or ac-meh-pro's accessable via ssh.
if somebody needs more information, ping me ...

achterin, 2019-01-04:
@NeoRaider if @heini66 sees the interchanged interfaces on both ap-ac-pro & mesh-pro the solution might be to [remove this line](https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-core/luasrc/lib/gluon/upgrade/020-interfaces#L43), right?

@heini66 did all devices show the same behavior?

heini66, 2019-01-04:
yes, all ac-pro's and ac-mesh-pro's ishowing off same behavior.

heini66, 2019-01-05:
push
something more required?

NeoRaider, 2019-01-06:
I'm wondering if this is a new issue, or if we got this wrong from the beginning. Does the same issue occur on the AC Pro when downgrading (using sysupgrade -n) to an older release? Both LEDE 17.01-based releases (Gluon 2017.1 / 2018.1) and OpenWrt CC (Gluon 2016.2) have support for the AC Pro, can someone check when the issue was introduced?

rotanid, 2019-01-06:
@NeoRaider we don't have this issue, but we do have only 6 of them. started using them with Gluon v2016.2.x, currently running v2018.1.x. didn't try v2018.2.x, yet.

heini66, 2019-01-06:
2017.1.8 was our first ever flashed version. most of them are running with it until now

NeoRaider, 2019-01-06:
Hmm, while we were still using a downstream patch for the AC Pro in v2016.2.x, I fail to see when this change was introduced.

@rotanid Can you post `/etc/config/network` from one of your "working" devices? The `switch`, `switch_vlan` and `interface 'wan'` sections should be sufficient.

heini66, 2019-01-06:
my /etc/config/networks are from 2018.2
i don't think is an new problem ( but may unreported in older versions )

achterin, 2019-01-14:
Finally I've got time to test my device again. I can indeed confirm (at least on the device i have at hand) that the LAN ports are mixed up but i cannot confirm the MAC-mismatch.

heini66, 2019-01-14:
zitat:
but i cannot confirm the MAC-mismatch
ende:
hum?
did i miss something here?

achterin, 2019-01-14:
@heini66 the op said in his opening post:
> Also the detected MAC (78:8a:20:b0:ec:d9) is different by one digit compared to the lable (78:8a:20:b2:ec:d9).

Thats the MAC-mismatch i was referring to. And as I said, i cannot confirm this issue on my device. The MAC was set to the one printed on the label.

mweinelt, 2019-01-16:
I find it difficult to follow the conversation right now. Please mention for every post
- which model
- which primary mac? does it match the label?
- is the port assignment mixed up?

If the MAC address read out is indeed flawed we might need to collect a few EEPROM dumps to fix it up.

achterin, 2019-01-16:
@mweinelt 
- Model: UniFi AP-AC Mesh Pro
- Firmware: gluon-v2018.1-147-g276362e2
- Primary MAC: matches the label
- Port assignment: mixed up

h03e, 2019-01-17:
@mweinelt

* Model: UniFi AP-AC Mesh Pro
* Firmware: v2018.2
* Primary MAC: matches the label
In `/etc/config/network` the MAC matches lable
I just checked the Primary MAC on the Meshviewer, there it was different by one digit (gonna recheck this weekend). 
* Port assignment: mixed up

I concluded that these things were associated in some way.

mweinelt, 2019-01-17:
And now the people with the Unifi AP-AC Pro please .

heini66, 2019-01-17:
don't have a mesh-pro here in my hands. will report tomorow form our mesh-pro's when i can access them diretcly...

mmalte, 2019-01-18:
I've got remote access to one. I've just installed v2018.2, the model detections is working great now: Ubiquiti UniFi-AC-MESH-PRO

Autoupdate is working fine, too.

Primary Mac is matching the one on the label.

Mesh on WAN and Mesh on LAN is activated, the cable connected to the Main Port (with PoE in) results in mesh connection via eth0.2

```
root@Gemeinde-Roetgen-Wanderstation-Mesh-Pro:~# brctl show
bridge name	bridge id		STP enabled	interfaces
br-wan		7fff.426e7a2573f0	no		eth0.1
br-client		7fff.788a20b2ec5c	no		local-port
							bat0
							client1
							client0
```

I'm able to confirm that the Ports are mixed up.

rotanid, 2019-01-22:
i've opened a separate issue dealing with the wrong primary mac on the uap-ac-pro: #1629 
maybe this issue is somewhat influenced by the other issue...

blocktrron, 2019-01-23:
@h03e can you dump the eeprom partition of the device and send it over to me (david@darmstadt.freifunk.net)

rotanid, 2019-01-25:
further ideas for testing:

- make sure you use upstream gluon, not a patch you backported yourself
- try a new build with latest gluon master which includes a fix for the primary mac. make sure to flash the upgrade without keeping configuration, otherwise this fix and maybe also other fixes won't do anything
- compare two devices which have/dont have the issue, start with labeling, open them, dump eeprom, etc.

rotanid, 2019-03-09:
is the issue gone or why doesn't anyone provide the requested information?
@h03e @heini66 @achterin @mmalte  ?

mmalte, 2019-03-10:
The device I have access to is already mounted on a street light, we've just activated mesh on lan and mesh on wan, so we are safe on that one.

I try to get access to another one.

mmalte, 2019-03-11:
@rotanid is it necessary to build from the master or would v2018.2.x be sufficient?

rotanid, 2019-03-11:
current v2018.2.x would be better than the january stuff.
keep in mind to flash *without* keeping configuration, ideally also restore factory firmware and flash gluon again afterwards ( instructions: https://wiki.darmstadt.freifunk.net/Unifi_AC )
if the problem still exists afterwards, look at what blocktrron and i asked for in january.

rotanid, 2019-05-29:
closing because of missing feedback.

closed 2019-05-29 by rotanid