freifunk-gluon/gluon #1611: config-config-mode-geolocation-osm: i18n issue

mweinelt, 2019-01-03:
Something in the osm patchset is messing with i18n. The first sentence in the section is the literal string, then the translated string follows.

[image]

Maybe here? https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-config-mode-geo-location/luasrc/lib/gluon/config-mode/wizard/0400-geo-location.lua#L21

NeoRaider, 2019-01-03:
Hmm, the POT/PO files of gluon-config-mode-geo-location haven't been updated. I assume the issue will also occur without the OSM package.

NeoRaider, 2019-01-03:
Ah, it was the other way around. The POT/PO files were changed, but it seems the corresponding text change in the Lua sources got lost in a rebase.

commit 2aa324ec (2019-01-03): gluon-config-mode-geo-location: fix source/i18n mismatch

The text was changed in the i18n files, but the corresponding change in the
Lua sources seemingly got lost during a rebase.

Closes #1611
commit a152a6e0 (2019-01-03): gluon-config-mode-geo-location: fix source/i18n mismatch

The text was changed in the i18n files, but the corresponding change in the
Lua sources seemingly got lost during a rebase.

Closes #1611

(cherry picked from commit 2aa324ecf7f38ea2a59aee0eaf745c54c50858bb)
closed 2019-01-03 by NeoRaider