freifunk-gluon/gluon #1617: unbreak mikrotik target

wrosenauer, 2019-01-13:
Sorry, I'm not an openwrt expert but looking at the latest information here:
https://openwrt.org/toh/mikrotik/common
I understand that sysupgrade is meanwhile supported with current openwrt?

If so, would it be possible to unbreak the mikrotik target in gluon or are there other blockers?

rotanid, 2019-01-13:
marking the target ar71xx-mikrotik as "unbroken" needs more than just some new information from another wiki.

* the sysupgrade process has to be tested with devices of all types we specify (or only one but then only this type of device can be un-marked broken)
* the device integration has to be completely done: https://github.com/freifunk-gluon/gluon/wiki/Device-Integration-checklist

(i also posted this in the comments of the pull request #1613 )

txt-file, 2019-02-13:
I vote to close this issue and do what @rotanid and @mweinelt suggested in https://github.com/freifunk-gluon/gluon/pull/1613#issuecomment-454777370

closed 2020-03-03 by mweinelt