freifunk-gluon/gluon #1618: Auto-Updater Documentation regarding user consent

yanosz, 2019-01-13:
The current documentation on automatic updates and its implementation appears incomplete regarding user consent. Two shorcomings appear to exists:

- [ ] The config-mode wizard does not ask users regarding consent on automatic-updates. It's an expert feature not presented within the wizard
- [ ] The documentation does not outline, that enabling automatic updates in the site-configuration  may result in remote access to nodes, network and data without explicit user consent. An active configuration appears to be favorable for public installations managed by a group of people - But for community networks (e.g. Freifunk) this is rather problematic. It allows access backdoorish to private devices, networks and data. It appears reasonable to document a proper warning explaining the situation, explaining, that user consent should be gained out-of-band.

rotanid, 2019-01-13:
as long as the default stays enabled (or at least can be set to enabled via site.conf) i would be fine with some changes with your concerns in mind

rubo77, 2019-01-13:
I think every auto update feature contains the risk of an evil company rolling out a backdoor firmware. But the company has to be evil still. No third party could roll out an update.

Does for example avm show such a warning?

yanosz, 2019-01-13:
@rotanid: Default is fine - can stay like this.

@rubo77 - I don't really understand your point. Your argument sounds rather rather philosophical:  My fritzbox asks me for consent before installing an update, so does Google Play. I'm able to change this manually, still.

Please elaborate: Do you see a problem with asking users for consent during the config mode wizard? Do you see a problem with outlining remote access capabilities in the documentation?

yanosz, 2019-01-13:
Idea for modifying the docs.

rubo77, 2019-01-14:
https://static.giga.de/wp-content/uploads/2017/04/fritzbox-automatisches-update-deaktivieren.jpg

dasteht nur:

> Über neue FRITZ!OS-Versionen informieren und neue Versionen automatisch installieren
Die FRITZ!Box informiert Sie über neue FRITZ!OS-Versionen. Zusätzlich wird jede neue Version automatisch installiert. Die FRITZ!Box wählt dazu einen geeigneten Zeitpunkt aus, z.B. nachts. Während der Installation werden die Internet- und Telefonieverbindungen kurzzeitig unterbrochen. Im Sinne des technischen Fortschritts können in einer neuen Version einzelne Funktionen verändert und in seltenen Fällen inkompatibel zum Ursprung sein.

what do you mean with "asking for consent"?

yanosz, 2019-01-14:
> what do you mean with "asking for consent"? 

The system asks for consent regarding automatic updates. IMHO AVM does a rather good job asking the users.

Other examples:
- Windows asking for consent during installation http://www.subodh.com/Portals/2/assets/1/36/Windows-7-under-Hyper-V-First-Look/Windows-7-Setup-Wizard_21.png
- Google has asks for consent out of band (terms of usage) regarding critial updates and for non-ciritical updates indivudually https://support.google.com/googleplay/answer/113412

I'm still thinking about the UX, when asking for consent in gluon. Atm I'm in favor of a check box ("consent to automatic updates") durzing wizard ... but AVM's implementation also has come pros.

I'll sketch a design, when I start procastinating again ;-)

A-Kasper, 2019-01-14:
I like the Idea of consent autoupdater. 
Just a concept idea:

3 Options:
1. update branch
2. update mode
- via command line (manual update)
- inform and accept (half-automatic)
- community (full automatic)
3. Owners Mailadress

- communities can control available options via site.conf

manual update means: running autoupdate via ssh starts fw download und nodeupdate automatic (nothing new)

community means: if a signed fw is available the node will download and install (as already implemented)

new would be:
inform and accept: The node sends an mail (smtp setup needed in configmode) that a new fw is available. the owner can click a link in this mail with a token sending a request to the node. the node sets a flag to activate the classic autoupdate process. after update this flag is removed.

yanosz, 2019-01-15:
@A-Kasper Thanks for your input. Unfortunatly, I think that a token based mechanism is a lot of work. 

IMO gluon's security model is based on the idea to provide access on phyiscal presence. I.e. pressing a button you put a node into config mode. Disconnecting all Freifunk-networks, the device is accessible on its lan port with no credentials provided.

I guess, that a token based mechanism may require thread modelling, especially on the token workflow (e.g. is the token going to be cryptographically signed on confirmation? How to restrict the uci-session, when flashing via luci? Is it necessary, to prevent phising attacks: what's the implecation of catching a token maliciously?).

ATM I'd rule out your ideas, but I think its easier to start small.

Greetz, yanosz

yanosz, 2019-01-19:
well, I played a little bit using xhtml and designed some visual prototypes

Current behaviour. Consent is assumed / provided out of band
* https://yanosz.net/au-consent/wizard-given.xhtml 
* https://yanosz.net/au-consent/adv-given.xhtml 

The new functionaltiy, allowing users to consent in-band is illustrated here:
* https://yanosz.net/au-consent/wizard.xhtml 
* https://yanosz.net/au-consent/adv.xhtml

Please mind: 
* Afaik texts can be provided by communities nevertheless. Thus, I don't think it's important to agree on a specific wording. However, it is helpful to fix typos, etc.
* This is only a html prototype. There's nothing to merge.
* In the adv-screen, more options are presented after users consent to automatic updates

rotanid, 2019-01-19:
basically that's what i thought of after reading your initial text.

i'm absolutely against the wording you chose, even if you don't want to talk about this obviously.

yanosz, 2019-01-19:
@rotanid : Looking at your initial post  - Are you still fine with this? I've some doubts reading your comment, now.

> as long as the default stays enabled (or at least can be set to enabled via site.conf) i would be fine with some changes with your concerns in mind

[...]

> i'm absolutely against the wording you chose, even if you don't want to talk about this obviously.

... I think you got me wrong the wording.  I'd like have some feedback, but in the end it doesn't really matter to reach an agreement. I'm fine with almost any wording, that still transport the general concept of granting access for configuration changes by the administrators. IMHO communities will probably adapt it to their language (formal / informal) and their docs (i.e. refs to online ones) anyway.

For the concept / Looking from AVM's perspective:

- Unlike FritzOS, the branches are not 6.0, 7.0 per se. Here, automatic updates using the same branches are recommended (important security fixes) - installing new releases is not. I don't think, that one can expect a community network to have similar strategy regarding branches. ATM I see a lot of communities having "stable","beta", etc. instead of "v1", "v2". I guess, that the current prototype can be used for both strategies, if texts are adapted. It may be helpful to outline this in the documentation, though.

- The general information ("You have already given your consent for access by downloading and using this software.") is mostly meant for keeping the defaults. I don't think its wise to roll out a firmware like that, however, it reflects the policy as of today.

- I'm not that happy with the UX of the advanced mode. Using radio-buttons may be more concise, still. The one in 2018.2 also lacks a description. Looing at 2018.2 the UX perspective I don't understand, why users are able to select a branch even though updates are disabled. Also, I don't think the that term branch is somewhat misleading, because it usually refers to different similar versions in a SCM system, while "stable", "beta" refers to stages, IMHO.

yanosz, 2019-01-19:
> Hi @yanosz, thanks for the prototypes to illustrate what you have in mind.
> 
> Regarding this:
> 
> > "Do you allow administrators to access your private device and networks for automatic updates? If so, please mind protecting your data using a guest network or a firewall zone (DMZ). You can configure automatic updates in the Advanced settings section.
> > Allow administrators to access my node []"
> 
> Is this always true? Doesn't this depend on a communities policies?
> 
> For instance at Freifunk Lübeck I would not be able to access another person's node or network. If I were adding an SSH key or some backdoor to a firmware then our other potential signers would refrain from signing. Hence I would not be able to propagate that update.

Hmm ..  that's a difficult point. Do you have a suggestion? I had a few things in mind, when writing this:

- Access is not related to shell access. It's meant to be more general. Access as in: You're able to modify the configuration / software of a devices. It also not limited to malicious access.

- Different communities have different signing policies. E.g. 1 signature is enough for KBU (in some situations). Thus I'm trying to be neutral:  I'm trying to refer to "administrators" in general without describing the internal workings of this group. It's probably the best, if each community describe the process individually.

- Access is not restricted to malicious access. I'm thinking more about errors (e.g. a private network is exposed due to a configuration error, that didn't exists before an update)

yanosz, 2019-01-19:
hmm... I did another quoting the Memorandum of Understanding using the terms accordingly. I guess, its a little bit clumsy, but nevertheless communities are to change this. I hope, that this brings this discussion towards the actual UX:

Current behaviour. Consent is assumed / provided out of band

*    https://yanosz.net/au-consent/wizard-given-v2.xhtml
*    https://yanosz.net/au-consent/adv-given-v2.xhtml

The new functionaltiy, allowing users to consent in-band is illustrated here:

 *   https://yanosz.net/au-consent/wizard-v2.xhtml
 *   https://yanosz.net/au-consent/adv-v2.xhtml

rotanid, 2019-01-20:
I prefer your new wording over the old one.

Of course the the communities are able to change this, but the default should be good for the majority. Many Gluon users don't change the defaults.

yanosz, 2019-01-27:
Ok, well it looks like there no more complaints on the last screenshots. I'd happy to sort things out before starting to implement this - I'd like to avoid throwing work away, later.

From what I understood so far - please correct me if I'm wrong:

There are three distinct modes of operation / configuration:

1. autoupdater is not used. No configuration is done
2. autoupdater is configured, but not running in background. Key material is provided by the community.
3. autoupdater is configured and run peridically

If so, I guess that it'd helpful, to display to the user in advanced mode: 

- A note about command line usage and selecting a branch without enabling periodic updates
- When was the last automatic updates (timestamp)
- Which public keys are placed in the firmware (including fingerprints)

A button to check for updates would be nice, but I think its a next step. I think I'll get things done by the end of this - hopelly earlier :-), but it'll take some time. The last time I patched the config-mode was back in the old LFF days and I don't know, how much has changed since then - it may take to some to get used to it again :)

rubo77, 2019-01-28:
> Which public keys are placed in the firmware

Usually none.

It is not planned to place public keys in the firmware

rubo77, 2019-01-28:
>Node operators have the choice to allow remote maintenance. Interventions on the nodes via automatic firmware upgrades are done with explicit agreement of the respective operators. You can configure automatic updates in the Advanced settings section.
>
>Agree to remote maintenance via automatic updates

this is enough: 

```
Node operators have the choice to allow automatic firmware upgrades.
You can configure automatic updates in the advanced settings section.

Agree to automatic updates
```

the satatus right now: 
- there is a hint in the wizzard mode, that sais, that automatic updates are enabled or not:

     https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-config-mode-autoupdater/i18n/de.po#L14
  - "This node will automatically update its firmware when a new version is "
    or
  - "Automatic updates are disabled. They can be enabled in <em>Advanced "
"settings</em>."

Keep in mind:

- automatic updates have to be enabled by default, like before.
- I don't think we need the same button to agree in two places. 
- there is no access of anyone to your node possible (unless someone manages to serve a hacked updated firmware)

yanosz, 2019-01-28:
Thanks for your feedback. Regarding your comments

> This is not correct:
> 
> > Node operators have the choice to allow remote maintenance. Interventions on the nodes via automatic firmware upgrades are done with explicit agreement of the respective operators. You can configure automatic updates in the Advanced settings section.
> > Agree to remote maintenance via automatic updates

> this is enough:
I don't get your point - what is not correct? As stated - the text is copied from the Freifunk Memorandum of understanding and should work for many comunnities out there.  I just changed "must be done" to "is done", due to the different context of this statement.

Well, I don't see any substantial difference. To avoid bikeshedding, I'd like keep things as they are, while communities can change the text as they like.

>  there is a visible hint in the wizzard mode, that sais, that automatic updates are enabled or not:
>       https://github.com/freifunk-gluon/gluon/blob/master/package/gluon-config-mode-autoupdater/i18n/de.po#L14

This is interesting, because I overlooked this  text until now.  UX is complicated :)

> automatic updates have to be enabled by default, like before.

Have you followed this discussion so far? We already agreed no to change the defaults. The idea is to introduce additional options while keeping the defaults as before.
 
> I don't think we need the same button to agree in two places.

Hmm... don' think so. IMHO: The wizard should provide a clear & concise view, while the admin area presents all options.
 
> there is no access of anyone to your node possible (unless someone manages to serve a hacked updated firmware)

Well - this discussion leads nowhere. 
IMO the capability to install new software or configuration on a devce is remote management access - or short: access. I find this idea to be reflected in the MoU, but I respect, that your idea of access appears to be restricted to shell access. I guess, that we can agree to disagree.

> ## Suggestion:
> 
> let's just add an agree-button to the already existing hint in wizzard mode, that has to be checked to continue, unless you disable the autoupdater in advanced mode

Hmm.. interesting. Why do you prefer a button instead of check box? 
 
> add a hint in advanced mode, that if you disable the AU, the branch is only used on the ssh console, and only in the case, that you would actively allow access via a key (or password) in the ssh section.

Yep - good point.

> add an explanation somehow, that no admin access is possible through the AU. (take away the fear)

I guess, the MoU is explicit enough and serves as a good common ground. I'd like to stick to it.

rubo77, 2019-01-29:
With "button" I meant "checkbox", sorry.

It is most misleading if you say "Agree to remote maintenance" or such. Nearly half the users would not allow that, because they think admins would be on their node all the time via SSH then. And it is not the plan of any sane admin to access the node via SSH either, so maybe we can formulate the sentence, somehow, that it doesn't immediately initiate fear in the user:

Maybe a slightly chane to my suggested text then

```
Node operators have the choice to allow automatic firmware upgrade access.
You can configure automatic updates in the advanced settings section.

[x] Agree to automatic firmware upgrade access
```

kaechele, 2019-02-04:
I guess the point @yanosz is trying to make here is that with automatic updates the updates are usually supplied by whoever creates the firmware for a certain community. These people usually have signing power and can sign new firmware releases to be then automatically installed on the node.
Anyone with that power (explicitly or implicitly) can modify the firmware in a way that it contains SSH keys or any other type of backdoor.
While it is true that this is pretty much valid for any sort of software that updates itself through an AU mechanism, and people usually blindly accept that, I think @yanosz is trying to call upon our self-given moral obligation to be very transparent about what risks are involved. And to that point there is no doubt about the fact that SSH keys and backdoors could potentially be added in after the fact.

So, further unpacking this, @yanosz intention seems to be (and please correct me if I am wrong) to raise awareness of the potential implications of blindly enabling AU or enabling it by default for new nodes. This raised awareness and the need to opt-in will undoubtedly lead to drastically lower adoption rates of autoupdates, which I believe is not unintentional depending on which belief system you employ when running a Freifunk network (the both extremes being: DIY, decentralized, everyone responsible to keep their own nodes updated, maybe even with their own firmware builds vs. "full-service" Freifunk Network that updates itself).
Opt-in vs. Opt-out as well as clear stating of the implications of things, especially if they might be negative, will always lead to low adoption rates, because users are lazy and like to live with the defaults.

Then again: This is Gluon, a framework for building mesh firmwares based on OpenWrt. Everyone is free to decide which text they want to present to users in the firmwares they build (by either forking the package or providing overrides through translation). You don't have to be lazy and live with the default ;-)

So, all in all, I would welcome a more transparent and honest description for the AU function to be in place by default (for the Gluon framework) and leave it up to the individual communities as to how they would like to inform their users about the implications and how they present them.

rubo77, 2019-02-04:
You are right, so we could formulate it as it is then: a potential risk

yanosz, 2019-02-06:
@kaechele thanks for your feedback. I guess, that there are some subtle misunderstandings I'd like to outline.

1.  I'm not considering situations in which somebody tries to modify the firmware maliciously. I'm also not interested in speculation about attacks. IMHO it is necessary and polite to ask users for consent, before changing the software or configuration on their devices. E.g. 

 - Looking at Windows 7's onboarding, users are asked for consent *not regarding to malware using the windows update system or similar attacks*. (http://www.subodh.com/Portals/2/assets/1/36/Windows-7-under-Hyper-V-First-Look/Windows-7-Setup-Wizard_21.png)
 - The idea of "explicit" consent is also expressed in the MoU, but not regarding "evil" behavior or malware. It's stated regarding decentralization and empowering Freifunkas 
 - The cryptoparties' saying "the keyboard is lava" is very similar.

2. It's not Opt-In vs. Opt-Out, since this assumes a certain default. It's about asking without any default.

3. It's not about blindly enabling / or disabling automatic updates. I'd like to modify gluon to support Freifunk communities in actively asking the user for consent and implementing the MoU. Nowadays, the discussion is about Opt-In (and hope, the user is visiting advanced settings, that won't happen in most cases) or Opt-Out (and pretending that consent was given _explicitly_). IMHO both options are bad.  IMHO the best way as of today is to have two firmware downloads (with or without automatic updates), so that users have to decide before downloading. However, this creates more effort, since two images have to be maintained. I'd like to avoid this.

4. I guess, that many communities have a distinct opinion on what to promise (e.g. responsibility for providing updates) and transparency (log, release-notes, etc.). Thus modifying built-in text can hardly be avoided.

rubo77, 2019-02-06:
OK, so maybe just a checkbox:

    [x] I am aware, that this software has a built-in autoupdate mechanism, that can be en- or disabled in advanced settings

rotanid, 2019-02-07:
short note: please keep in mind that even if you both find a consenus, that doesn't mean it's acceptable for a Gluon maintainer who needs to merge this.
and looking at the list of open PRs, it is likely that feedback on this change may take time.

yanosz, 2019-06-25:
I'm not going to work on this anymore - see #1721 for details.
Closed - won't fix.

closed 2019-06-25 by yanosz